INTI-CMNB/KiBot #102: Show original filename on PCB sheet - not temporary filename

mdeweerd, 2021-11-13:
I noticed this on the PCB documentation scheet generated through yaml type "pdf_sch_print" .
[image]

The schematic sheet shows the correct name.

Expected: show original PCB design sheet name - possibly annoted by the variant (as the Fabrication layers are adjusted).

mdeweerd, 2021-11-13:
When the layers are exported to PDFs, the filename has an indication of the layers that are in the PDF, but the sheets do not mention what layer it is.

So maybe the temporary filename could be named after the original and extended with '-F.Cu' for the front copper layer.   
Which also makes me think about the fact that 'F.Cu' is not necessarily clear for everybody and in France we tend to write "Seriegraphie" for "Silk" for instance, and other labels for the other layers.  
So while doing anything about naming the sheets, one could think about:
- Customised texts;
- Using a comment field (chosen by the user) to write the layer name to or to append the layer name to (could be  a format string to increase freedom).

But the main thing is that it should be observable in a paper printed document which layer is shown and which design it is about.

As there is not really a place on github (AFAIK) to praise a project, I use this spot I must say that KiBot is a very good collective job to automating and adding functionnality to KiCAD.  Don't consider my "issues" as criticism!

closed 2021-11-16 by set-soft

mdeweerd, 2021-11-16:
@set-soft Thanks - I am afraid this introduced an extra popup stalling the output and aborting, tested with kibot_variants_arduprog (or something else did):

[image]

mdeweerd, 2021-11-16:
Regarding the feature: it's already better to have the original filename, 
What about the Layer name ('-F.Cu' , ...) and the variant name (_%v).  This can be kept open as a feature request ;-).

I think its in part KiBot (with regards to adding the variant) and KiAuto (with regards to adding the layer information).

set-soft, 2021-11-17:
> @set-soft Thanks - I am afraid this introduced an extra popup stalling the output and aborting, tested with kibot_variants_arduprog (or something else did):
> 
> [image]

I need a simple test case that produces it, I tried `kibot_variants_arduprog` and it works without problems:

```
salvador@galadriel:~/.../kibot_variants_arduprog$ make
kibot  -d Generated -s run_erc -i
Using config file: ardu_prog.kibot.yaml
WARNING:(W044) More than one SCH file found in current directory.
  Using t1.sch if you want to use another use -e option. (kibot - __main__.py:134)
- Running the DRC
Found 1 unique warning/s (1 total)
kibot  -d Generated -s run_drc -i
Using config file: ardu_prog.kibot.yaml
WARNING:(W044) More than one SCH file found in current directory.
  Using t1.sch if you want to use another use -e option. (kibot - __main__.py:134)
- Running the ERC
Found 1 unique warning/s (1 total)
kibot  -c ardu_prog.kibot.yaml -e t1.sch -d Generated/default -g variant=default -s all
- 'Spreadsheet for the Bill of Materials' (bom_xlsx) [bom]
WARNING:(W020) Malformed value: `CG0603MLC-05E` (no match) (kibot.kibot.bom.units - units.py:135)
- 'HTML for the Bill of Materials' (bom_html) [bom]
- 'Prototype mounting guide' (ibom) [ibom]
WARNING:(W065) iBoM needs information about user defined fields and no netlist provided (kibot.kibot.out_ibom - out_ibom.py:145)
WARNING:(W065) Create a BoM in XML format or use the `update_xml` preflight (kibot.kibot.out_ibom - out_ibom.py:147)
- 'Schematic PDF' (print_sch) [pdf_sch_print]
- 'Gerbers' (gerbers) [gerber]
- 'Drill files' (drill) [excellon]
- 'Pick & Place' (position) [position]
- 'PDF for PCB top layer' (print_pdf_l1) [pdf_pcb_print]
- 'PDF for PCB GND plane' (print_pdf_l2) [pdf_pcb_print]
- 'PDF for PCB power plane' (print_pdf_l3) [pdf_pcb_print]
- 'PDF for PCB bottom layer' (print_pdf_l4) [pdf_pcb_print]
- 'PDF to show fab layers' (print_fab) [pdf_pcb_print]
- 'Top layer view' (board_top) [pcbdraw]
WARNING:(W016) Ambiguous list of components to show `all` vs variant/filter (kibot.kibot.out_pcbdraw - out_pcbdraw.py:159)
- 'Top layer view with components' (board_top_filled) [pcbdraw]
- 'Bottom layer view' (board_bottom) [pcbdraw]
- 'Bottom layer view with components' (board_bottom_filled) [pcbdraw]
- 'STEP 3D model' (3D) [step]
Found 4 unique warning/s (8 total)
kibot  -c ardu_prog.kibot.yaml -e t1.sch -d Generated/USB -g variant=USB -s all
- 'Spreadsheet for the Bill of Materials' (bom_xlsx) [bom]
WARNING:(W020) Malformed value: `CG0603MLC-05E` (no match) (kibot.kibot.bom.units - units.py:135)
- 'HTML for the Bill of Materials' (bom_html) [bom]
- 'Prototype mounting guide' (ibom) [ibom]
WARNING:(W065) iBoM needs information about user defined fields and no netlist provided (kibot.kibot.out_ibom - out_ibom.py:145)
WARNING:(W065) Create a BoM in XML format or use the `update_xml` preflight (kibot.kibot.out_ibom - out_ibom.py:147)
- 'Schematic PDF' (print_sch) [pdf_sch_print]
- 'Gerbers' (gerbers) [gerber]
- 'Drill files' (drill) [excellon]
- 'Pick & Place' (position) [position]
- 'PDF for PCB top layer' (print_pdf_l1) [pdf_pcb_print]
- 'PDF for PCB GND plane' (print_pdf_l2) [pdf_pcb_print]
- 'PDF for PCB power plane' (print_pdf_l3) [pdf_pcb_print]
- 'PDF for PCB bottom layer' (print_pdf_l4) [pdf_pcb_print]
- 'PDF to show fab layers' (print_fab) [pdf_pcb_print]
- 'Top layer view' (board_top) [pcbdraw]
WARNING:(W016) Ambiguous list of components to show `all` vs variant/filter (kibot.kibot.out_pcbdraw - out_pcbdraw.py:159)
- 'Top layer view with components' (board_top_filled) [pcbdraw]
- 'Bottom layer view' (board_bottom) [pcbdraw]
- 'Bottom layer view with components' (board_bottom_filled) [pcbdraw]
- 'STEP 3D model' (3D) [step]
Found 4 unique warning/s (8 total)
kibot  -c ardu_prog.kibot.yaml -e t1.sch -d Generated/XTAL -g variant=XTAL -s all
- 'Spreadsheet for the Bill of Materials' (bom_xlsx) [bom]
WARNING:(W020) Malformed value: `CG0603MLC-05E` (no match) (kibot.kibot.bom.units - units.py:135)
- 'HTML for the Bill of Materials' (bom_html) [bom]
- 'Prototype mounting guide' (ibom) [ibom]
WARNING:(W065) iBoM needs information about user defined fields and no netlist provided (kibot.kibot.out_ibom - out_ibom.py:145)
WARNING:(W065) Create a BoM in XML format or use the `update_xml` preflight (kibot.kibot.out_ibom - out_ibom.py:147)
- 'Schematic PDF' (print_sch) [pdf_sch_print]
- 'Gerbers' (gerbers) [gerber]
- 'Drill files' (drill) [excellon]
- 'Pick & Place' (position) [position]
- 'PDF for PCB top layer' (print_pdf_l1) [pdf_pcb_print]
- 'PDF for PCB GND plane' (print_pdf_l2) [pdf_pcb_print]
- 'PDF for PCB power plane' (print_pdf_l3) [pdf_pcb_print]
- 'PDF for PCB bottom layer' (print_pdf_l4) [pdf_pcb_print]
- 'PDF to show fab layers' (print_fab) [pdf_pcb_print]
- 'Top layer view' (board_top) [pcbdraw]
WARNING:(W016) Ambiguous list of components to show `all` vs variant/filter (kibot.kibot.out_pcbdraw - out_pcbdraw.py:159)
- 'Top layer view with components' (board_top_filled) [pcbdraw]
- 'Bottom layer view' (board_bottom) [pcbdraw]
- 'Bottom layer view with components' (board_bottom_filled) [pcbdraw]
- 'STEP 3D model' (3D) [step]
Found 4 unique warning/s (8 total)
```

set-soft, 2021-11-17:
> What about the Layer name ('-F.Cu' , ...) and the variant name (_%v). This can be kept open as a feature request ;-).

The b4c1531e101575c0c1098a3b86b318070c7167c4 patch allows configuring the title ;-)

mdeweerd, 2021-11-17:
Thanks - I do think about test cases, but I am quite busy at the moment ;-).

commit cfcc9bea (2021-11-16): Fixed PCB Print: to show the real name of the PCB file.

Fixes #102