INTI-CMNB/KiBot #103: 3D Model variants (related or not to BOM variants)

mdeweerd, 2021-11-13:
I tend to make footprints that allow different types of components in the same spot.

Some developers will suggest to make two different footprints and put them in the same spot, but then we have to cope with overlapping footprints and footprint alignement, etc.

One good example is the choice between a surface mount push button or a through hole push button:
[image]

I also do Fuseholders, DC jacks, rectifying diode bridges, ethernet jacks, wire-to-Board connectors, etc. this way when possible.  Sometimes it's just one component type, but I add the connectors mating part, the 3D model for the case (my Edges are defined in a footprint where I add a 3D model of the casing when available).

In the 3D modeling setup, I add both the THT and SMD components (sometimes even 3), one on top of the other.   That allows me to check that they are all compatible for instance.

In the case of an ethernet connector, I'ld have a variant in the BOM setup to select either the THT type or SMD type, in other case I do not have a variant because I don't care about the type.

3D models are not managed by variants for the same component, and there is not really provisioning in KiCAD that helps us to manage that.

I see that the STEP output is generated by removing 3D models.

I suggest to a model filtering feature to the STEP output generation which allows us to remove models from the output.

Suggestion:
```yaml
- name: 'step with model variants'
  type: step
  variant_options:
      - variants: [ v1, v2 ]  # variant names defined in yaml configuration
        footprint: AV:SW-PHAP3301E_dual # Optional
        refs: "SW[12]" # Optional, filter applied only to these references
        keep: ".*B3F.*" # Keeps only the models matching the reference, full path match
        remove:  ".*430182070816.*" # removes the matching models, full path match, useless with keep.
      - variants: [ v3 ]
        footprint: ".*:OtherFootPrint" # Optional, regex, must also cope with rescue library
        refs: "^(U10|U11)$" # Optional, applies only to refs.
        remove:  ".*SMD.*" # removes the matching models, full path match, useless with keep.
```

set-soft, 2021-11-15:
Hi @mdeweerd !

We already have a mechanism to change the 3D model using variants, please take a look at #59

Isn't documented because @Freshdouble never confirmed it was working.

Please try it and let me know if it works of it needs more work. The `test_step_variant_1` regression test checks it.

mdeweerd, 2021-11-15:
Ok, this wouldn't work as the models do not have the same offsets.

I prefer to adjust and overlap the models in the footprint editor.

A technique is to add the variant in the model's path.  Sometime I add several possibility and then I disable them by adding an x to the path.
That 'x' could be '%variant1,variant2,variant3%" and if it matches, then the model is added (and the ones that do not match or do not have a variant specification are removed).
So there could be a default model "blabla.wrl"  or "blabla.stp" and a variant model "other%variant1,variant2%.wrl" .  If variant1 and/or variant2 matches, then "blabla.*" is removed/renamed, and  we get "other.wrl" with its positioning parameters.
If somebody wants to keep "blabla.wrl" in all cases, then this person will have to add "blabla%%" for instance on top of the "blabla.wrl".  (The goal is that KiCad still shows a model when using it interactively).

The method to place the models, is to place them without the %VARIANTSPEC% first, and then add the variant specification.

I do not have a public license available 3D model at hand (manufacturers subject them to a license agreement before downloading), but I could create one using KiCad models with a pin header that is originally 4 pins, but with 3 variants: 4 pin, left 2 pins, right 2 pins.

Where would you like me to put that?

set-soft, 2021-11-17:
> Ok, this wouldn't work as the models do not have the same offsets.

I found problems with the STEP export when using scales and offsets, then I saw that the FreeCAD plug-in says you should always avoid them.

> Where would you like me to put that?

If this is an example to show your workflow just use any GitHub/GitLab repo.

But let me know if I understand:

1. Your footprint has more than one 3D model
2. One or more models are enabled to produce the look for a variant, the rest are disabled
3. Changing which models are enabled/disabled produces different aspects
4. Each model has its own offset to compensate the "multi component footprint"

Is that what you say?
Then you can use the `VARIANT:_3D_model` because this mechanism allows to replace *all* the 3D models. You just need to provide a comma separated list of models to use for each variant. You can have a `dummy` 3D model to replace the ones you want to make invisible.

mdeweerd, 2021-11-17:
- I did not have problems with the exported steps when using offsets and scales.  The manufacturer model rarely have the same reference X,Y, Z axis and rotation needs to be adjusted to fit on the pads.

To help you understand:

1. Yes, theres more than one model for a footprint.
    - Different manufacturers, different looks;
    - Different types (e.g. fuse holder);
    - Different product casings
      - My board's edges are in a footprint (some drawbacks, but quite a few advantages);
      - I associate the casing with that "board footprint" - the same PCB can go in different shells.
2. By default, for interractive work, I generally keep all the models active:
  - Validate that they are all positioned correctly;
  - Always have the full picture about space requirements;
  - Remind about the alternatives.

  I disable models (not always) when producing 3D images & step outputs.  With KiBot/KiAuto I could be more selective.
3. Not sure what you mean exactly, but I'ld say : yes.
4. The models would have offsets (and scales) even in a single component footprint.

Supposing that the 3DVariant option in the schematic changes all models in they order that they appear in the footprint definition, I guess that this mechanism could be made to work.

However, its against the DNR principle (Do Not Repeat) and subject to maintaining the models order in the footprint and count.  (The same model names and paths needs to be kept both in the schematic, the PCB and the library footprint).
It also brings PCB model information to the schematic where it's normally contained in the Footprint Library and the PCB Cad File.

I'ld prefer avoiding the repetition and keeping the model information in the footprint library/kicad_pcb file.

mdeweerd, 2021-11-18:
I started a test case in https://github.com/mdeweerd/KiBot/tree/kibom-variant_4 .

I added example for 3DModel variants (using %% in model path), empty dnp field, not pre-existing component field definition by kicost variant.
I added the step generation to test_step.py without running that as I first tried to do a normal kibot run.

When I run
 ```
./runInKiAuto \
kibot -e tests/board_samples/kicad_5/kibom-variant_4.sch -b tests/board_samples/kicad_5/kibom-variant_4.kicad_pcb -c tests/yaml_samples/step_variant_2.kibot.yaml -g variant=topvariant schematic_color bom steptopvariant stepleftvariant &
```
where runInKiAuto runs in setsoft/kicad_auto:dev using my local kibot copy (volume mount), 

I get:
```
WARNING:(W042) Missing doc-lib entry for kibom-variant_4-rescue:WC1602A-Display_Character (kibot.kicad.v5_sch - log.py:88)
WARNING:(W062) Invalid column name `manf` (kibot.out_bom - log.py:106)
WARNING:(W062) Invalid column name `manf#` (kibot.out_bom - log.py:106)
WARNING:(W062) Invalid column name `sos#` (kibot.out_bom - log.py:106)
- '' (bom) [bom]
- '' (schematic_color) [pdf_sch_print]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 361, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 364, in generate_outputs
    run_output(out)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 340, in run_output
    out.run(get_output_dir(out_dir))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 92, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_pdf_sch_print.py", line 55, in run
    fname = GS.sch.save_variant(sch_dir)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1698, in save_variant
    self.save(fname, dest_dir)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1681, in save
    e.write(f)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1109, in write
    field.write(f)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 768, in write
    f.write(' {} {}{}{}'.format(self.hjustify, self.vjustify, ['N', 'I'][self.italic], ['N', 'B'][self.bold]))
TypeError: list indices must be integers or slices, not str
```

The generated BOM looks correct.

set-soft, 2021-11-18:
The above error is now fixed.

set-soft, 2021-11-18:
> However, its against the DNR principle (Do Not Repeat) and subject to maintaining the models order in the footprint and count.

Yes, I agree it isn't ideal. But having invalid file names in the PCB isn't better from my point of view.

> (The same model names and paths needs to be kept both in the schematic, the PCB and the library footprint).

This is not a real problem, I explain it below.

> It also brings PCB model information to the schematic where it's normally contained in the Footprint Library and the PCB Cad File.

Yes.

> I'ld prefer avoiding the repetition and keeping the model information in the footprint library/kicad_pcb file.

You can just put the information in the symbol library. There is not such a thing as a "Resistor" in real life, you will use an specific resistor. Things like the manf, manf#, distributor#, power, tolerance, voltage, etc. should be in the symbol library, so you can easily change all the components that refer to the same real life component.
I agree that having to maintain the sync between the footprint and the symbol is a real pitty, but is also a cleaner solution.
The fact that KiCad silently ignores errors about missing 3D models doesn't justify using unreal file names.

I'm reluctant to this solution.

One thing that I never explored is the fact that footprints can have an arbitrary number of "user defined texts". Perhaps we can use these fields to control the 3D aspects.

And again: currently the 3 variants mechanisms are in fact schematic centric, I don't see using schematic fields as strange for this purpose.

mdeweerd, 2021-11-18:
We're reluctant about the opposite things.

I will not be moving 3D Model paths to the schematic or repeat them - it's a headache to keep them in sync, and in the right order.  When you remove one and add one, the order changes - you can't add a model just in the middle of the list already there.  
Repeating is just looking for trouble, I've experienced the introduction of bug or error due to missed updates due to repeated information many times.  
In my view, the cleaner solution is the one that leads to less mistakes - even if it's "dirty".  
I agree that we can consider invalid names as "dirty", but I find repeating dirtier.

Invalid names in the PCB is not hindering anything in itself - we're working around KiCAD limitations - an invalid name in the model list is a simple trick and helps avoid spreading and (manually) repeating information all over the place.  
Having a single 'yaml' file for all the tools with the same variant mechanism is also better than having to specify the same variants in different ways for the different tools - the iBom author suggested to use kibot because he was not keen on managing variants differently from what iBom does now.

Putting it in the symbol library that is shared accross projects (using git, submodules) is not a solution IHMO either.  Putting the symbol in a local library limits the benefit from schematic symbol improvements and sharing thereof.  
A PCB footprint is more rarely updated - so I have no trouble there making it more project specific.

Anyway, it's up to you.

set-soft, 2021-11-18:
I'm looking for a better solution.

About your example: I see you use things like `Pin_Header_Straight_1x01_Pitch2.54mm.wrl`, but they aren't part of current stable release. 

Let me know when the example is ready for examination

mdeweerd, 2021-11-18:
Thank you.

I have updated the example.

set-soft, 2021-11-18:
> Sometimes I add several possibilities and then I disable them by adding an x to the path.

Why? Why not just use the check-box?
BTW: `steps/3Dcomponent.3dshapes/User Library-AC162A.STEP` and `steps/3Dcomponent.3dshapes/User Library-MT16S2J.STEP` aren't in the repo.

Looking at your example I see some repetition ... You have most models twice, one normal and the other for the variants.

What about using the following:

[image]

Corresponding to:

[image]

No invalid file name, less repetition and all contained in the footprint. I didn't investigate how to implement it, but looks cleaner.

And don't come and tell me "What if I have to remove a model?", just don't do it, disable it and add another at the end of the list.

mdeweerd, 2021-11-18:
> Why? Why not just use the check-box?

As far as I remember the checkbox state is not maintained and/or because it only influences the preview and not the step export - perhaps that changed, but once I have a working solution, I am not checking the original behavior all the time.

> BTW: steps/3Dcomponent.3dshapes/User Library-AC162A.STEP and steps/3Dcomponent.3dshapes/User Library-MT16S2J.STEP aren't in the repo.

Yes, these models are subject to some licence on 3Dcomponent so I did not upload them.  
I also tend to keep references to models to remind me of other (even bad) alternatives.  This is one of the extreme cases though: multipart, multiple main parts, even different colors.

> Looking at your example I see some repetition ... You have most models twice, one normal and the other for the variants.

Yes, to be consistent with my proposal.  When working interactively, KiBot & KiAuto will not intervene.
The visible componente are there to see the full potential physical space of the components on the footprint.  And also visually check that the positioning is ok.  
I proposed that when applying the variants, slightly modified here, that in when at least one of the model has a variant specification, only the models matching the variant will be ketp.  A wild card %% ("empty") would always match.  

The user is free to not add any visible model.  

Keeping the model definition in the same place makes it easier to compare them.  To build the test case, I edited the *.kicad_pcb file directly by copying the original definitions and adding the variant definitions.

> What about using the following: .. Corresponding to.

That's better than keeping it in the schematic.  It's still position dependant, but it is acceptable as it's still very close to where the models are set up.  

> No invalid file name, less repetition and all contained in the footprint. I didn't investigate how to implement it, but looks cleaner.

The clean up is ok for the test cases, but do "keep" at least a few missing models to ensure that it works also in those cases.

> And don't come and tell me "What if I have to remove a model?", just don't do it, disable it and add another at the end of the list.
It's a solution to add it at the list, when there are only a few and I really want to remove the model, I'll renumber .   But in my proposal I avoid renumbering the variant is directly verifiable with the model, the drawback is local duplication though (which I do not like either) which is not obliged if one accepts that the models are hidden during interactive work.  
I know I am challenging you, but I challenge myself more ;-) and I appreciate your consideration of my suggestions and your work a lot.  

I am not sure that you have looked at your *.org email today - I sent you one from mine.

set-soft, 2021-12-02:
Ok, I did a first implementation, using the following example:

[test.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7644876/test.tar.gz)

I get:

[image]

And:

[image]

The magic is done by the following texts in the module:

```
%leftvariant:2,4,5,9%
%topvariant:3,8%
```

The list of modules is:

```
${KISYS3DMOD}/Display.3dshapes/WC1602A.wrl
${KISYS3DMOD}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_2x07_P2.54mm_Vertical.wrl
${KISYS3DMOD}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x16_P2.54mm_Vertical.wrl
${KISYS3DMOD}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x01_P2.54mm_Vertical.wrl
${KISYS3DMOD}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x01_P2.54mm_Vertical.wrl
${KIPRJMOD}/steps/3Dcomponent.3dshapes/User Library-AC162A.STEP
${KIPRJMOD}/steps/3Dcomponent.3dshapes/User Library-MT16S2J.STEP
${KIPRJMOD}/steps/WH1602B-TMI-JT#.step
${KIPRJMOD}/steps/ERM1602DNS-2.1.step
```

Note that you don't need to add `x` to the names, they will be filtered by KiBot. Also note we don't have any duplicated entry with variant tags.

I'm closing this issue, but I think further refinement will be needed.

closed 2021-12-02 by set-soft

mdeweerd, 2021-12-02:
You've done quite a few things today (not only today).  

It isn't the kind of magic that you can bring to the "Penn & Teller" magic show, but kibot may need to be called KiMaGIC (KiCAD Magic Automation for Gerbers and Integral Customization ,  or so ...).

If you want I can now put a free text on these 3D screen models in case that can bring a nice touch for a presentation.

commit 84d6217d (2021-12-02): Another experimental mechanism to change 3D models according to the variant.

Closes #103