INTI-CMNB/KiBot #105: AttributeError: 'SchematicField' object has no attribute 'horizontal'

mdeweerd, 2021-11-14:
Using setsoft/kicad_auto:dev

Added debug informatoin to know what data is causing the failure:
> DEBUG:write field: {'name': 'comment', 'value': 'Requires snubber', 'number': 18} (kibot.kicad.v5_sch - v5_sch.py:761)

The failure:
> Traceback (most recent call last):
>   File "/usr/local/bin/kibot", line 16, in <module>
>     main()
>   File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 358, in main
>     generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 364, in generate_outputs
>     run_output(out)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 340, in run_output
>     out.run(get_output_dir(out_dir))
>   File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 88, in run
>     self.options.run(self.expand_filename(output_dir, self.options.output))
>   File "/usr/local/lib/python3.7/dist-packages/kibot/out_pdf_sch_print.py", line 55, in run
>     fname = GS.sch.save_variant(sch_dir)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1694, in save_variant
>     self.save(fname, dest_dir)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1677, in save
>     e.write(f)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1105, in write
>     field.write(f)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 762, in write
>     f.write('F {} "{}" {}'.format(self.number, self.value, ['V', 'H'][self.horizontal]))
> AttributeError: 'SchematicField' object has no attribute 'horizontal'
> 

The component is:
```
$Comp
L Triac_Thyristor:BT139-600 Q2
U 1 1 5D1D0879
P 13500 2250
F 0 "Q2" H 13350 2350 50  0000 L CNN
F 1 "T835T-8G" V 13650 2025 50  0000 L CNN
F 2 "YNAMICS:D2PAK" H 13700 2175 50  0001 L CIN
F 3 "https://www.mouser.fr/datasheet/2/389/en.CD00002266-1393627.pdf" H 13500 2250 50  0001 L CNN
F 4 "STMicroelectronics" H 13500 2250 50  0001 C CNN "manf"
F 5 "T835T-8G-TR" H 13500 2250 50  0001 C CNN "manf#"
F 6 "TRIAC Logic 8A 800V D2PAK Snubberless Triac" H 13500 2250 50  0001 C CNN "desc"
F 7 "mot2" H 13500 2250 50  0001 C CNN "variant"
F 8 "NXP" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf"
F 9 "BT139B-600\\,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf#"
F 10 "TRIAC Logic >=8A 800V D2PAK Snubberless Triac - requires snubber!" H 13500 2250 50  0001 C CNN "kicost.mot2pro:desc"
F 11 "Requires snubber" H 13500 2250 50  0001 C CNN "kicost.mot2pro:comment"
F 12 "BT139B-600,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:value"
F 13 "" H 13500 2250 50  0001 C CNN "kicost.mot2pro:"
F 14 "1740-1053-1-ND" H 13500 2250 50  0001 C CNN "kicost.mot2pro:disgikey#"
F 15 "2902357" H 13500 2250 50  0001 C CNN "kicost.mot2pro:farnel#"
F 16 "509522" H 13500 2250 50  0001 C CNN "kicost.mot2pro:rs#"
F 17 "771-BT139B-600118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:mouser#"
        1    13500 2250
        1    0    0    -1
$EndComp
```

This may have to do with the fact that the variant selects `mot2pro` :

```yaml
variants:
  - name: 'myvariant'
    file_id: _Pmyvariant
    variant: "^(mot2pro|snubber)$"
    type: kicost
    pre_transform: 'kicost_rename'
```

The issue is that there is no comment field initially and so the script seems to add a field without all the attributes.

I added a comment to the component:
```
$Comp
L Triac_Thyristor:BT139-600 Q2
U 1 1 5D1D0879
P 13500 2250
F 0 "Q2" H 13350 2350 50  0000 L CNN
F 1 "T835T-8G" V 13650 2025 50  0000 L CNN
F 2 "YNAMICS:D2PAK" H 13700 2175 50  0001 L CIN
F 3 "https://www.mouser.fr/datasheet/2/389/en.CD00002266-1393627.pdf" H 13500 2250 50  0001 L CNN
F 4 "STMicroelectronics" H 13500 2250 50  0001 C CNN "manf"
F 5 "T835T-8G-TR" H 13500 2250 50  0001 C CNN "manf#"
F 6 "TRIAC Logic 8A 800V D2PAK Snubberless Triac" H 13500 2250 50  0001 C CNN "desc"
F 7 "mot2" H 13500 2250 50  0001 C CNN "variant"
F 8 "NXP" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf"
F 9 "BT139B-600\\,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf#"
F 10 "TRIAC Logic >=8A 800V D2PAK Snubberless Triac - requires snubber!" H 13500 2250 50  0001 C CNN "kicost.mot2pro:desc"
F 11 "Requires snubber" H 13500 2250 50  0001 C CNN "kicost.mot2pro:comment"
F 12 "BT139B-600,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:value"
F 13 "" H 13500 2250 50  0001 C CNN "kicost.mot2pro:"
F 14 "1740-1053-1-ND" H 13500 2250 50  0001 C CNN "kicost.mot2pro:disgikey#"
F 15 "2902357" H 13500 2250 50  0001 C CNN "kicost.mot2pro:farnel#"
F 16 "509522" H 13500 2250 50  0001 C CNN "kicost.mot2pro:rs#"
F 17 "771-BT139B-600118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:mouser#"
F 18 "~" H 13500 2250 50  0001 C CNN "comment"
        1    13500 2250
        1    0    0    -1
$EndComp
```

The comment was fine after that, but I got an issue with an empty field:

> DEBUG:write field: {'number': 18, 'value': 'Requires snubber', 'horizontal': True, 'x': 13500, 'y': 2250, 'size': 50, 'flags': '0001', 'hjustify': 'C', 'vjustify': 'C', 'italic': False, 'bold': False, 'name': 'comment'} (kibot.kicad.v5_sch - v5_sch.py:761)[0m
> DEBUG:write field: {'name': '', 'value': '', 'number': 19} (kibot.kicad.v5_sch - v5_sch.py:761)[0m
> Traceback (most recent call last):
>   File "/usr/local/bin/kibot", line 16, in <module>
>     main()
>   File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 358, in main
>     generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 364, in generate_outputs
>     run_output(out)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 340, in run_output
>     out.run(get_output_dir(out_dir))
>   File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 88, in run
>     self.options.run(self.expand_filename(output_dir, self.options.output))
>   File "/usr/local/lib/python3.7/dist-packages/kibot/out_pdf_sch_print.py", line 55, in run
>     fname = GS.sch.save_variant(sch_dir)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1694, in save_variant
>     self.save(fname, dest_dir)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1677, in save
>     e.write(f)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 1105, in write
>     field.write(f)
>   File "/usr/local/lib/python3.7/dist-packages/kibot/kicad/v5_sch.py", line 762, in write
>     f.write('F {} "{}" {}'.format(self.number, self.value, ['V', 'H'][self.horizontal]))
> AttributeError: 'SchematicField' object has no attribute 'horizontal'
> [0m

Which is:
``` 
F 13 "" H 13500 2250 50  0001 C CNN "kicost.mot2pro:"
```

So the script could be improved to not create fields with no name (I fixed the schematic for that as well).
Next I got an issue with the "digikey#", "farnel#", etc. fields which did not exist when there is no variant.

When I got to the next component definition, the blocking issue was gone:
```
$Comp
L Triac_Thyristor:BT139-600 Q2
U 1 1 5D1D0879
P 13500 2250
F 0 "Q2" H 13350 2350 50  0000 L CNN
F 1 "T835T-8G" V 13650 2025 50  0000 L CNN
F 2 "YNAMICS:D2PAK" H 13700 2175 50  0001 L CIN
F 3 "https://www.mouser.fr/datasheet/2/389/en.CD00002266-1393627.pdf" H 13500 2250 50  0001 L CNN
F 4 "STMicroelectronics" H 13500 2250 50  0001 C CNN "manf"
F 5 "T835T-8G-TR" H 13500 2250 50  0001 C CNN "manf#"
F 6 "TRIAC Logic 8A 800V D2PAK Snubberless Triac" H 13500 2250 50  0001 C CNN "desc"
F 7 "mot2" H 13500 2250 50  0001 C CNN "variant"
F 8 "NXP" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf"
F 9 "BT139B-600\\,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:manf#"
F 10 "TRIAC Logic >=8A 800V D2PAK Snubberless Triac - requires snubber!" H 13500 2250 50  0001 C CNN "kicost.mot2pro:desc"
F 11 "Requires snubber" H 13500 2250 50  0001 C CNN "kicost.mot2pro:comment"
F 12 "BT139B-600,118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:value"
F 13 "1740-1053-1-ND" H 13500 2250 50  0001 C CNN "kicost.mot2pro:digikey#"
F 14 "2902357" H 13500 2250 50  0001 C CNN "kicost.mot2pro:farnell#"
F 15 "509522" H 13500 2250 50  0001 C CNN "kicost.mot2pro:rs#"
F 16 "771-BT139B-600118" H 13500 2250 50  0001 C CNN "kicost.mot2pro:mouser#"
F 17 "~" H 13500 2250 50  0001 C CNN "comment"
F 18 "" H 13500 2250 50  0001 C CNN "digikey#"
F 19 "" H 13500 2250 50  0001 C CNN "farnell#"
F 20 "" H 13500 2250 50  0001 C CNN "rs#"
F 21 "" H 13500 2250 50  0001 C CNN "mouser#"
        1    13500 2250
        1    0    0    -1
$EndComp
```

set-soft, 2021-11-15:
Hi @mdeweerd !

The patch attacks the root of the problem: the constructor didn't initialize misc. attributes. So when we tried to fake a field it was incomplete. This should fix the problem (again I need some real cases).

closed 2021-11-15 by set-soft

mdeweerd, 2021-11-18:
Real test case through https://github.com/INTI-CMNB/KiBot/issues/103#issuecomment-972372062 ;-).

set-soft, 2021-11-18:
Fixed in 2a09151a181833b4af7f9a636030bc494b8063e6 thanks

commit 10486af5 (2021-11-15): Fixed KiCost variants: when setting a field hat doesn't exist

Fixes #105