INTI-CMNB/KiBot #106: How to avoid crossed out test points in the schematic?

mdeweerd, 2021-11-14:
I mark my testpoints as "DNP" because they are not actual components and should not appear in the BOM - they exist on the board.

It's however "annoying" that they are crossed out in the schematic.

[image]

What's the best approach with regards to this?

set-soft, 2021-11-15:
I think you should mark them as virtual components, they aren't components, just pads in your PCB.

set-soft, 2021-11-15:
If this works for you please close the issue.

mdeweerd, 2021-11-15:
@set-soft 
The PADS are already virtual in the layout - they do not appear on the fabrication layers.  This concerns the schematic.
As far as I remember, I need to make them dnp in the schematic to not have them appear in the BOM and that makes them crossed out.

set-soft, 2021-11-15:
Hi @mdeweerd !

> This concerns the schematic.

Variants are in fact schematic centric, the PCB is "consequence".

> I need to make them dnp in the schematic to not have them appear in the BOM and that makes them crossed out.

Nope, you just need to exclude them, not make them DNP. You can exclude virtual components, you can also use other criteria for the filters. I usually like to use a distributor code. When defined as `none` means this isn't a component.

Take a look at the `exclude_filter` stuff. This is used to avoid problems with things like the mounting holes, test points, etc.

mdeweerd, 2021-11-15:
I'll see if these options can make a difference.

mdeweerd, 2021-12-24:
I made a test case for this:

[tc5.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7774859/tc5.tar.gz)

- Files are generated in tc5_01_20211224__header, log is tc5_header_build.log :

tc5_01_20211224__header/

- PCBA/tc5_01_20211224-bom.csv -> OK only has header component
- PCBA/tc5_01_20211224-kicost.xlsx
  - NOK
  - Has QR1, test points but not the header component.
  - => Add option for exclude filter?
- PDF/tc5_01_20211224-ibom.html
  - NOK
  - Has QR1 (Footprint "as is" from KiCAD footprint generator)
  - => Add options for exclude filter?
- tc5_01_20211224-both_pos.csv
  - OK for component placement
  - NOK for test pin location
  - => Generate separate file for TEST PIN Location ?  Best method?
  
  
Other observation: tc5.csv is generated at the output root - I do not think this is requested for one of the targets, I think it's related to the pre-flight BOM XML generation.

So in summary:
- the kicost output is incorrect (missing the selected component, adding unwanted components).  I'ld prefer to reuse the 'exclude_filter' here.
- IBOM: how to apply the exclude filter?
- Position file: ok for PCBA, but what is the recommended method to extract the test point locations.

set-soft, 2021-12-29:
> I made a test case for this:
> 
> [tc5.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7774859/tc5.tar.gz)

Hi @mdeweerd ! The tarball doesn't include the Schematic, PCB and KiBot configuration.

set-soft, 2021-12-29:
> * => Add option for exclude filter?

Most outputs support variants, and variants support exclude filters. So you can apply a dummy variant containing the filter. The internal BoM is a special case where you have both options. This is because this was the first place where variants and filters were implemented.

mdeweerd, 2021-12-29:
Aargh!   This should be better: [tc5.tar.gz](https://githuat https://github.com/mdeweerd/kibot-testcases .b.com/INTI-CMNB/KiBot/files/7790488/tc5.tar.gz) .  Also available at https://github.com/mdeweerd/kibot-testcases .

I haven't thought that completely through, but I think its not possible with a single run, and there are side-effects with multiple runs.

The limitation of creating a separate variant is that the outputs can not all be generated from the variant provided on the CLI as it requires one variant with the exclusion filter and another without the exclusion filter, and so for each global variant.

Running the generation twice, with different variants also creates different base directories.

set-soft, 2021-12-30:
> PCBA/tc5_01_20211224-kicost.xlsx
> * NOK
> * Has QR1, test points but not the header component.
> * => Add option for exclude filter?

This is documented, take a look at the KiCost output docs it says:

 - `dnf_filter`: [string|list(string)='_none'] Name of the filter to mark components as not fitted.
                        A short-cut to use for simple cases where a variant is an overkill.
                        Internal variants and filters are currently ignored.

- `variant`: [string=''] Board variant to apply.
                     Internal variants and filters are currently ignored.

If you wonder why: the mechanism used by KiCost is quite complex and supporting it means applying the variants to the netlist. Which isn't currently possible. But you have two options:

1. Simply specify your variant in the `kicost_variant` option.
2. Just migrate to the internal BoM, is far better than KiCost. And it supports KiCost variants (along with KiBoM and iBoM variants).

I'm adding some magic here that could help.

set-soft, 2021-12-30:
The above patch is passing the **header** variant to KiCost. This could help.

BTW: How do you exclude the QR* and TP* components in the KiCost output? (I mean using KiCost alone)

set-soft, 2021-12-30:
> tc5_01_20211224-both_pos.csv
> * OK for component placement

Is this `PCB/tc5_01_20211224-both_pos.csv`?

Here I got J1001 and QR1. I think QR1 shouldn't be there, but this is an error in the PCB. This component isn't marked as virtual. 

> * NOK for test pin location
> * => Generate separate file for TEST PIN Location ?  Best method?

Position for virtual components? This isn't the use of a position file, I guess this could be useful for testing.

If this is the case think the best is to create a special variant for it. Something like `test_points` variant.

mdeweerd, 2021-12-30:
> > tc5_01_20211224-both_pos.csv
> > 
> > * OK for component placement
> 
> Is this `PCB/tc5_01_20211224-both_pos.csv`?
I have PCBA/...

> 
> Here I got J1001 and QR1. I think QR1 shouldn't be there, but this is an error in the PCB. This component isn't marked as virtual.

I have QR1 as well (overlooked it).  So that is effectively too much too.

> Position for virtual components? This isn't the use of a position file, I guess this could be useful for testing.
>
> If this is the case think the best is to create a special variant for it. Something [...]
That is acceptable for the test pins.

> Just migrate to the internal BoM, is far better than KiCost
My intention is to use KiCost for the pricing & sourcing - even if pricing is disabled in the test case.  I am already using the internal BOM for the rest.  In practice I am not satisfied/haven't validated kicost fully yet in this kibot setup.

I will test the patch and add something for the testpoint generation.

set-soft, 2021-12-30:
I figured out that virtual components can't be part of the P&P file.

So now I added an option to include them.

I added the following filter:

```yaml
  - name: 'test_points'
    comment: 'Get the test points'
    type: generic
    invert: true
    exclude_refs:
      - TP*
```

And the following output:

```yaml
  - name: 'position_test_points'
    type: position
    dir: "+PCBA"
    extends: 'position'
    output_id: '_test_points'
    options:
      variant: 'test_points'
      include_virtual: true
```

This generated `PCBA\tc5_01_20211224-both_pos_test_points.csv` containing:

```csv
Designator,Val,Package,Mid X,Mid Y,Rotation,Layer
"TP1001","TP_IN","TestPoint_Pad_D1.5mm",161.2000,-90.2500,0.0000,bottom
"TP1002","TP_NEG","TestPoint_Pad_D1.5mm",157.1000,-104.1500,0.0000,bottom
"TP1003","TP_OUT","TestPoint_Pad_D1.5mm",164.1000,-103.8500,180.0000,bottom
"TP1004","TP_VCC","TestPoint_Pad_D1.5mm",153.8000,-90.6000,0.0000,bottom
"TP1005","TP_GND","TestPoint_Pad_D1.5mm",153.6500,-94.0750,0.0000,bottom
```

set-soft, 2021-12-30:
Ok, after looking at the examples:

- You should use `exclude_filter` only inside the variant. The need to include it for all the relevant variants isn't a real problem, is just a line indicating it.
- Standalone KiCost is currently limited to use its own mechanism. You can solve your problem by adding a field `kicost.header:DNP` to the components, but this is quite limited and annoying. I strongly suggest migrating to the internal BoM mechanism. I'll see how hard is to implement the netlist generator.

set-soft, 2021-12-30:
> Other observation: tc5.csv is generated at the output root - I do not think this is requested for one of the targets, I think it's related to the pre-flight BOM XML generation.

Yes, this is a side effect of the XML generation. Now is removed.

set-soft, 2021-12-30:
> * I'll see how hard is to implement the netlist generator.

Added some experimental code, KiCad 5 only. It generates an XLSX with just J1001 (after adding the `exclude_filter` to the `header` variant).

set-soft, 2021-12-30:
I think you need to create a "default" variant, perhaps using an empty `file_id`, but with a custom `exclude_filter` (`bom` in this case) and then set the `variant` field to `default` instead of an empty string. This should allow to generate both variants, the `default` and the one specified from the command line in the same run.

mdeweerd, 2021-12-30:
> I figured out that virtual components can't be part of the P&P file.
Right, I forgot that (I made them virtual at some point and then I wanted the positions in the P&P file and I did not get them).

> I added the following filter:
I added it as a filter to the "common.kibot.yaml" file, and got (after all targets were run) - this was done in [mdeweerd/kbot-testcases](https://github.com/mdeweerd/kibot-testcases/blob/193c79b4f3be5372842035cdf2d9b5dd697d5381/yaml/common.kibot.yaml#L674) and I imported the filter in the project yaml https://github.com/mdeweerd/kibot-testcases/blob/193c79b4f3be5372842035cdf2d9b5dd697d5381/designs/tc5/tc5.kibot.yaml#L6 :
```
- '' (kicost) [kicost]
ERROR:In section 'position_test_points' (position): Unknown variant name `test_points` (kibot - error.py:40)
```

Then I added it as a variant to try, but I got, so it's definitively a filter that you meant:
```raw
ERROR:Unknown variant type: `generic` (kibot - error.py:40)
```

I also noticed that the targets are not executed in the order they appear:
```bash
kibot -c designs/tc5/tc5.kibot.yaml -b designs/tc5/tc5.kicad_pcb -e designs/tc5/tc5.sch \
-g variant=$i   position_test_points schematic_color schematic_bw bom ibom position gerbers gerb_drill excellon pcbtop_img pcbbottom_img pdf_pcb pdf_pcb_fab step render_2d_top_black render_2d_bottom_black render_3d_top_black render_3d_bottom_black allpads_pcb allpads_drill allpads_position allpads_zip kicost 
```
I expected position_test_points to run first, but it didn't.

The log:
[tc5_header_build.log](https://github.com/INTI-CMNB/KiBot/files/7794875/tc5_header_build.log)

mdeweerd, 2021-12-30:
> Added some experimental code, KiCad 5 only. It generates an XLSX with just J1001 (after adding the `exclude_filter` to the `header` variant).
OK, I could repeat that.  
The idea is to keep as many "common" options in the "common.kibot.yaml" file,

set-soft, 2021-12-31:
> ```
> - '' (kicost) [kicost]
> ERROR:In section 'position_test_points' (position): Unknown variant name `test_points` (kibot - error.py:40)
> ```
> 
> Then I added it as a variant to try, but I got, so it's definitively a filter that you meant:
> 
> ```
> ERROR:Unknown variant type: `generic` (kibot - error.py:40)
> ```

You need the following variant:

```yaml
  - name: 'test_points'
    comment: 'Only the testpoints'
    type: kicost
    file_id: _test_points
    variant: ''
    exclude_filter: 'test_points'
```

> I also noticed that the targets are not executed in the order they appear:

They are executed in the order they are declared in the file. I'll add an optional mechanism for CLI order

set-soft, 2022-03-03:
Hi @mdeweerd : Is this question solved?

mdeweerd, 2022-03-03:
Same as some other "open issues": I will check it but not in the very near future.  The test points are a bit harder to setup & test.