INTI-CMNB/KiBot #107: Generate outputs when '-v 5' is specified

mdeweerd, 2021-11-14:
No outputs are generated when I add the '-v 5' option to kibot, and when removing it, I get the outputs which puzzled me a while.

I suggest that the '-v' option only changes verbosity, not the output.

set-soft, 2021-11-15:
`-v 5` is interpreted as `-v` and `5` as the help states:

```
kibot [-b BOARD] [-e SCHEMA] [-c CONFIG] [-d OUT_DIR] [-s PRE]
         [-q | -v...] [-i] [-m MKFILE] [-g DEF]... [TARGET...]
```

I guess you want to specify `-vvvvv`. Note the help doesn't say `-v LEVEL`

closed 2021-11-15 by set-soft

mdeweerd, 2021-11-15:
Ok, it's because I saw the following code from entrypoint.sh :
````
          -v | --verbose) shift
               if [ "$1" == "0" ]; then
                   VERBOSE=""
               elif [ "$1" == "1" ]; then
                   VERBOSE="-v"
               elif [ "$1" == "2" ]; then
                   VERBOSE="-vv"
               elif [ "$1" == "3" ]; then
                   VERBOSE="-vvv"
               else
                   VERBOSE="-vvvv"
               fi
               ;;
```

As I prefer -v <digit>, I opted for that, but the above code .  In the else case  it's always -vvvv, and I wanted maximum debug as well.  But on closer look that code looks flawed: the 0-3 digits are not shifted, and in the else case it's not verified if the parameters is a number to shift it out in that case only as well.

set-soft, 2021-11-15:
This script was kindly provided by @nerdyscout 

Is used for the dockerimage and translates the GitHub Action parameters into KiBot parameters. The native mechanism is `-vvvvv`, which is exactly the same as `-vvv` BTW. If you want to submit a patch to make it more general (i.e. work better when using values outside the defined range) I'll accept the PR.