INTI-CMNB/KiBot #108: Malformed Values - How to cope with these cases

mdeweerd, 2021-11-14:
As the "Value" of a component appears on the schematic, and in the BOM, I sometimes prefer mentioning specific attributes along with the nominal value.

Kibot changes the value and I do not know what the impact on the BOM, etc.  I'ld prefer that it just warns about it, and/or that this could be managed by an option.

For instance:
```
WARNING:(W020) Malformed value: `470uF 25V` (no match in C2) (kibot.bom.units - log.py:88)[0m
WARNING:(W020) Using `470uF` for C2 instead (kibot.bom.bom - log.py:106)
WARNING:(W020) Malformed value: `26K1 1/2W` (no match in R12) (kibot.bom.units - log.py:88)
WARNING:(W020) Using `26K1` for R12 instead (kibot.bom.bom - log.py:106)
```

set-soft, 2021-11-17:
> As the "Value" of a component appears on the schematic, and in the BOM, I sometimes prefer mentioning specific attributes along with the nominal value.

The preferred way is to use another fields for the other attributes. I.e. `voltage`, `power`, etc. You can join fields in the BoM, so you can put Value + voltage + power + tollerance + etc. in the `Value` column. 

> Kibot changes the value and I do not know what the impact on the BOM, etc. I'ld prefer that it just warns about it, and/or that this could be managed by an option.

Do you have `normalize_values` enabled?

mdeweerd, 2021-11-17:
I do not set "normalize_values".  I haven't moved to KiBot's bom yet.

Using KiBot allows to change the rules of the game.  

I am not sure that I can update the schematic this way though using KiBot unless I (can) make those extra fields visible.

Is there a "Guideline" file for this kind of stuff in KiBot as in "How to use KiCAD/adapt your methodology to optimize for KiBot"  (There's this, and the thing with test pads (that I asked about) which is the same thing for QR codes, Silk Symbols (like high voltage), etc).

mdeweerd, 2021-11-22:
Do not close this yet, I didn't completely get to this yet!

Just a note: I tried the join feature and I tried  [ 'desc', '/', 'value' ] to add a '/' between the description and the value.  The fields were joined, but without the '/'.  
I'll let you think about it, not a big thing.

set-soft, 2021-11-25:
These values are field names, you can't add arbitrary text. How can KiBot know if this is a misspelled field name or something you want to add on purpose?

This can't be done unless we define a proper mechanism. I don't know, may be use something like `'{arbitrary text}'`. And ask people to avoid using `{` and `}` at the beggining/end of field names.

mdeweerd, 2021-11-25:
I tested if there was automatic field recognition - I understood there was none.

Several methods are possible:
- What you propose: if the field has a special marker or markers, use it as text.  '{}' resembles variable substitution too much IMHO.
- If the field does not exist, use it as text
- If there is a special symbol, use it as text (but '#' and '.' are possible field name characters)
- If there is a space, assume it's text.  Field names should not contain spaces.  Regarding stripping: either not strip any space at all or only the one at the start of the string (if any).
- Use something rare: '§' ,

I'ld go with the space because it should not be in a field name.

set-soft, 2021-12-01:
Hi @mdeweerd !
I pushed a change, but missed the issue number ... anyways, the commit is 995219a8eaeeecfa644bcf69c05d9e57e388ee6c and the implementation is completely different.

Now the `join` option can be:
- Just a string, the name of a field to join.
- A list of strings, the name of fields to join.
- A list of `dicts`, these dicts can specify a fixed text (not sure if useful) or a field name. But also text to add before and after the field content. This text will be added only if the field isn't empty. So you could add **(Voltage)** and nothing will be added if `Voltage` is empty.
- A list of strings and dicts mixed.

We can add more options to the `dict` so I guess this is the most flexible option.

set-soft, 2021-12-01:
> I do not set "normalize_values". I haven't moved to KiBot's bom yet.

Can you post a small example of what are you using? Which output type?

mdeweerd, 2021-12-01:
Hi @set-soft 

The code update for the fields looks good - I'll check it out in a practical cas and share that.

Regarding the BOM: I use(d) KiCost and I have not investigated all the BOM options yet.  As my KiCost BOMs are the verified ones, I am sticking with those, but I did use the BOM output for a small board I ordered a week or two ago.

I see you made changes for this, and I will learn and apply these settings for normalization, etc.  So you can leave this set to "needs more info" and I'll provide feedback when I get to this.

mdeweerd, 2021-12-14:
I tested the field joining with the output defined in yaml (further below).  The goal is to show all component alternatives in the same cell - each alternative is on a seperate line - this is how one assembly house does it.

### Observations:
- A space is added after the main field even if there no space specified in yaml;
- There does not seem to be a method to define new lines.  I tried the [yaml literal block style](https://yaml.org/spec/1.2-old/spec.html#id2796118), but I only got spaces.
- This example also has "id_add", when put at the same level as type, it is not accepted.  When put inside "options" it has no effect.

```yaml
outputs:
  - name: 'PCBA_bom'
    comment: "BoM for PCBA"
    type: bom
    options:
      id_add: "_PCBA"
      #output: '%f_%r_%d-%i_PCBASTYLE.%x'
      format: XLSX
      ref_separator: ','
      columns:
        - field: Row
          name: "Pos."
        - field: References
          name: "Repère/Layout Ref"
        - field: "Quantity Per PCB"
          name: "Q"
        - field: desc
          name: "DESIGNATION"
        - field: "manf#"
          name: "Référence/Part number"
          join:
            - field: S1PN
            - field: S1PN
              text_before: |

            - field: S2PN
              text_before: '\012'
        - field: manf
          name: "FABRICANT/OEM"
          join:
            - field: S1MN
              text_before: '%0A'
            - field: S2MN
              text_before: '
'
        - field: Value
          name: "Value"
        - field: Footprint
          name: "Package"
        - field: comment
          name: "Observations"
        - field: 'LCSC#'
          name: 'LCSC Part #'
      xlsx:
        col_colors: true
        title: "BOM"
        logo: false
        style: modern-blue
        hide_pcb_info: false
        hide_stats_info: false
        generate_dnf: false
        datasheet_as_link: "datasheet"
        highlight_empty: false
```

set-soft, 2021-12-14:
> * This example also has "id_add", when put at the same level as type, it is not accepted.  When put inside "options" it has no effect.

Looking at the docs you'll see it isn't supported there. I added a new option to all outputs called `output_id` that can be used for all outputs and is the value for a new pattern named `%I`

set-soft, 2021-12-14:
> * A space is added after the main field even if there no space specified in yaml;

This was always like this, no surprise here ;-)
Ok, I understand that avoiding it when some `text*` is defined is a good idea. I removed the space in this case and documented that the user must add it manually.

set-soft, 2021-12-14:
> * There does not seem to be a method to define new lines.  I tried the [yaml literal block style](https://yaml.org/spec/1.2-old/spec.html#id2796118), but I only got spaces.

Not all YAML 1.2 stuff is implemented in the Python YAML parser.
I added some support for \n and \t in the `text` fields of `join` (not everywhere)

mdeweerd, 2021-12-14:
I pulled Kibot, tested and the \n replacement works perfectly.

set-soft, 2021-12-15:
If all the additions are working we could close this issue.

mdeweerd, 2021-12-15:
We digressed a bit from the initial message, I'll close this when I checked that out.

mdeweerd, 2022-03-03:
Not yet.  I have to do some PCB work again, but I have to finish other stuff first.

mdeweerd, 2022-03-14:
Back to the original issue.  Looking back on it I remembered a tool that could help with this and found [the kicost discussion](https://github.com/hildogjr/KiCost/issues/509#issuecomment-951387535)

https://kitspace.github.io/electro-grammar/

Parsing some of the cases that "BoM" warned about:
- "470uF 25V" -> decoded
- "26K1 1/2W" -> decoded
- "220nF/275V" -> not decoded, but I am ok to rewrite that to "220nF 275V"
- "275V/0.4W" and "275V 0.4W" not decoded -> X/Y protection component - could be added ;
- "275V 75J" -> protection component - could be added;
- "275V" -> protection component - could be specified better.

I tested withnormalize_values set to false and true - the warnings remained the same.

```yaml
  - name: bom
    comment: BOM
    dir: +PCBA
    type: bom
    options:
      normalize_values: false
      normalize_locale: false
   ...
```

Basically, I'ld like to keep the value specifications (mostly) as they are and not split them up in more columns.  More columns is going to be errorprone as well when changing the value or propagating it to other components.  
Manufacturers specify these values in the same text field as well, a solution like "electro-grammar" looks better.

mdeweerd, 2022-03-15:
# Values, renaming

It is confusing:

The in the BOM the value is not changed, but the log says it's using another value.  So the real issue seems to be with the warnings indeed.

The transcript indicates this:
```raw
- 'BOM' (bom) [bom]
WARNING:(W020) Malformed value: `Supercap` (no match in C11) (kibot - log.py:93)
WARNING:(W020) Malformed value: `470uF 25V` (no match in C2) (kibot - log.py:93)
WARNING:(W020) Using `470uF` for C2 instead (kibot - log.py:93)
```

And in the CSV I get this:
```
Value,"Designator","Description","Footprint","Manufacturer","Component code","Manufacturer2","Component code2","Manufacturer3","Component code4","Comment"
470uF 25V,"C2","","Panasonic_8x10.2L","Panasonic","EEEFT1E471AP","","","","",""
```

# elektro-grammar

This would help with regards to our old feedback "The preferred way is to use another fields for the other attributes" as all these parameters would be in a single field.

However, as such that does not seem to bring added value as BOM is not doing anything with these extra values.

There would be an added value to:
- Normalize the fields - even if with these extra values.  Put the values in a normalized order, uniform units, validate that every complementary value has a usefull unit.
- Normalizing the fields would help match fields that do not match because of spaces, slightly different writing, etc.

It's not critical for me in that way.  I am happy with getting rid of warnings that are given confusing information as it's not what is happening.

set-soft, 2022-03-15:
> The transcript indicates this:
> 
> ```
> - 'BOM' (bom) [bom]
> WARNING:(W020) Malformed value: `Supercap` (no match in C11) (kibot - log.py:93)
> WARNING:(W020) Malformed value: `470uF 25V` (no match in C2) (kibot - log.py:93)
> WARNING:(W020) Using `470uF` for C2 instead (kibot - log.py:93)
> ```

Ok, I added more text to clarify (when `normalize_values` is `false`)

set-soft, 2022-03-15:
> elektro-grammar

It could help in the future, but right now I don't see a real point in using it.
I'm closing this issue because the message is more clear now.

closed 2022-03-15 by set-soft