INTI-CMNB/KiBot #109: Custom output/custom command

mdeweerd, 2021-11-15:
The goal of custom output is to run a custom executable command.
This could allow making a copy of a file to the output directory, parse a template, run a script under development, etc.

```
 - name: '3d_view_export'
    comment: 'Exports the PCB to the most common exhange format. Suitable for printing.'
    type: 'custom'
    dir: "%f_%r_%d%v/PCBA"  # Passed to the script if the args_format is json
    args_format: json|string # when JSON all options to the target and several internal values are passed as JSON, when string, it's interpreted as a format.  For instance "{dir} {myarg} {otherarg}"
    command: NAMEOFTHEEXECUTABLE
    myarg: "Value for myarg"
    otherarg: "Value for otherarg"
    unusedarg: dummy # not used when args_format=string because its not in "{dir} {myarg} {otherarg}"
```

I thought of this while thinking of a way to execute the "3D Viewer" image export script that I created.

set-soft, 2021-11-22:
Is a good idea, but needs changes in some classes, so it can be applied to schematic, PCB and 3D variants. I'll look into it after finishing some KiCost plug-ins.

mdeweerd, 2021-11-22:
Ok, no hurry ;-).  I was thinking this could even be a base class with common options and build (refactor) the other tools on it.

richardbarlow, 2022-04-04:
I'd be very interested in this feature too. Since I am using KiBot to generate my final fabiration datapacks I need the ability to execute a custom script or two during the process to perform a little extra processing of generated files before they get compressed into a zip. Having the ability to have a 'custom command' output would work perfectly.

For reference, the extra processing I am looking to perform right now is to create a single assembly PDF by combining/merging multiple PDFs (not just concatenate pages like the 'pdfunite' output does).

set-soft, 2022-04-04:
> For reference, the extra processing I am looking to perform right now is to create a single assembly PDF by combining/merging multiple PDFs (not just concatenate pages like the 'pdfunite' output does).

Can you comment more about it? May be we can add more stuff to the `report` output so it can do what you want. It can currently nicely join the PCB/SCH print outputs.

richardbarlow, 2022-04-05:
> Can you comment more about it? May be we can add more stuff to the `report` output so it can do what you want. It can currently nicely join the PCB/SCH print outputs.

So the thing I am trying to achieve at the moment is to generate a single PDF that has two pages (one front and one back) on each page I would like to have the F.Fab layer + a couple of other layers visible (all overlayed). Unfortunately, the two mechanisms that KiCAD has at the moment (print and plot) have different features/limitations that means it is not possible to do this. The `pdfunite` output allows a front and back PDF to be combined into one - so KiBot helps with that, however the 'print' mechanism does not have as much flexibility as the 'plot' mechanism. For example you cannot suppress the visibility of the 'value' field on the fab layer with the 'print' mechanism.

Ultimately, this requires changes/improvements in KiCAD, which is no bad thing. However, this will not be quick and I must have a mechanism to enable me to work-around problems like this until the 'proper' solution is implemented in KiCAD/KiBot. My desire with the custom command functionality is to allow me to implement short-term workarounds and I would always strive to get the functionality implemented into KiCAD/KiBot in parallel.

set-soft, 2022-04-05:
Hi @richardbarlow !
Take a look at the last commit, is WIP, but I think you'll get the idea.
Is a new way to print the PCB using [Board2Pdf](https://gitlab.com/dennevi/Board2Pdf/), it generates the PDF using the plot functionality, but composing the output PDF using PyPDF2 (allows to combine and concatenate PDFs).

set-soft, 2022-04-08:
> For example you cannot suppress the visibility of the 'value' field on the fab layer with the 'print' mechanism.

With the new `pcb_print` output now you can.

matthiesenj, 2022-09-26:
I would like a "runner" output as well, for running a script/executable/whatever..
My use case is to manipulate the generated BOM and position csv files, because a footprint I use actually needs multiple connectors placed on the board during assembly.

set-soft, 2022-09-26:
Hi @matthiesenj !
The BoM currently supports multiple parts for the same symbol. Take a look at the `subparts` filter. It implements the concept found on KiCost.
About the position: Are you sure this isn't solved using more than one footprint? Please elaborate.

matthiesenj, 2022-09-26:
> Hi @matthiesenj ! The BoM currently supports multiple parts for the same symbol. Take a look at the `subparts` filter. It implements the concept found on KiCost. About the position: Are you sure this isn't solved using more than one footprint? Please elaborate.

The footprint in question is a Raspberry Pi Compute Module.. It is a single KiCad footprint, but has 4 mounting holes and two identical connectors. The reason it is designed as a single footprint is so it is easy to place and move on the pcb, I suppose.
I need to replace the footprint by 2 connectors and 4 SMD solderable spacers in the bom and position files. The position data for these 6 added parts will need to have their xy data generated from the parent footprint. Is something like this possible using subparts?

Footprint:
[image]

What I would like to place on the board (3D):
[image]

set-soft, 2022-09-26:
Hi @matthiesenj !
The BoM part is possible, the footprint isn't.
But I think your problem can be solved in the following way:

1. Create a virtual footprint with the Silkscreen part, just like a logo
2. Use separated footprints for each element, 6 of them.
3. Use the virtual component as an alignment guide, it will help to keep the distances 

I know that moving it in the PCB isn't simple, but the solution will be consistent with KiCad workflow.
The symbol for the virtual footprint could be something that helps to make the other symbols coherent.
This solution will keep all the information in the KiCad files, no external *magic*.
You'll most probably need some DRC exceptions to allow the integration because the virtual component will be over the others, but may be it isn't needed if the courtyard allows it.

set-soft, 2022-09-26:
> Your suggestion could work for the spacers, but not the connectors, I think - how would they get associated with the schematic symbol pins (the symbol has 200 pins, 100 for each connector) ?

Why not? Just use 2 connectors, lets name them A and B, both with pin numbers 1 to 100. 
Half of the connections goes to connA and half to connB.

matthiesenj, 2022-09-26:
> Why not? Just use 2 connectors, lets name them A and B, both with pin numbers 1 to 100. Half of the connections goes to connA and half to connB.

This would require using two distinct 100 pin schematic symbols, which would be a massive step backwards, considering the symbol I'm using has 10 functional units covering all 200 pins, and not something I would consider changing at this stage of the design.
[cm4_footprint.pdf](https://github.com/INTI-CMNB/KiBot/files/9647787/cm4_footprint.pdf)

> BTW: KiCad 6 can make footprint groups, so you should be able to align the parts, create a group and then move the whole group. Avoiding misalignments.

Interesting - might come in handy when I switch to KiCad 6 at some point.

set-soft, 2022-09-26:
> This would require using two distinct 100 pin schematic symbols, which would be a massive step backwards, considering the symbol I'm using has 10 functional units covering all 200 pins, and not something I would consider changing at this stage of the design.

We use a script to generate these symbols, is now outdated, but I'll add support to KiCad 6 next time I need to use it.

```perl
#!/usr/bin/perl
##############################################################################
#
# Copyright (c) 2014 David M. Caruso <daviud en inti gov ar>
# Copyright (c) 2015-2019 Salvador E. Tropea <salvador en inti gov ar>
# Copyright (c) 2014-2019 Instituto Nacional de Tecnología Industrial
#
##############################################################################
#
# Target:           Any
# Language:         Perl
# Interpreter used: v5.6.1/v5.8.4
# Text editor:      SETEdit 0.5.5
#
##############################################################################
#
# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; version 2.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program; if not, write to the Free Software
# Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA
# 02111-1307, USA
#
##############################################################################
#
# Description: Component Schematic Generator for Kicad
#
##############################################################################

#Usar una hash, para barrer los datos
#para el footprint barrer bolita por bolita.

use POSIX qw(strftime);
use Sort::Versions;

use Getopt::Long;
use Locale::TextDomain('kicad_tools');
$version='1.1.1';

ParseCommandLine();
#use constant AGE => 2;
$fecha = strftime "%e/%m/%Y-%H:%M:%S", localtime;

%kicadel = ('Input/Output','B',
            'Power Input','W',
            'Power Output','w',
            'NC','N',
            'Input','I',
            'Output','O',
            'Input Clock','I C',
            'Input Clock Negated','I IC',
            'Input Negated','I I',
            'Output Negated','O I',
            'Input/Output Negated','B I',
            'Output Clock','O C',
            'Output Clock Negated','O IC',
            'Tri-state Negated','T I',
            'Tri-state','T',
[... 429 lines cut ...]
141,IOT93/219,I/O 93/219 Bank0,Input/Output,PIO,Bank_0,VCCIO_0,0
142,IOT94/220,I/O 94/220 Bank0,Input/Output,PIO,Bank_0,VCCIO_0,0
143,IOT95/221,I/O 95/221 Bank0,Input/Output,PIO,Bank_0,VCCIO_0,0
144,IOT96/222,I/O 96/222 Bank0,Input/Output,PIO,Bank_0,VCCIO_0,0
```

Then you call it like this:

```shell
$ perl comp_gen.pl -input iCE40HX1-4K.csv -lib iCE40HX1-4K -dir . -part -wdcomp 1200 -withexfield -pinl 150
Component Schematic Generator for Kicad v1.1.1 Copyright (c) 2014-2015 David M. Caruso/SET/INTI
Footprint: TQFP-144
Descripcion: iCE40 High Speed FPGA
Fabricante: Lattice
Numero de parte: iCE40HX1K
Hoja de datos: -
Codigo: iCE40HX1K
Reading: 144 pins
There are: 6 groups of pins
```

And you get the 6 defined groups: 
[Esquema.pdf](https://github.com/INTI-CMNB/KiBot/files/9647886/Esquema.pdf)

Generating big components, and maintaining them, is much more easy this way.

matthiesenj, 2022-09-27:
That's very nice - I think many of the original KiCad symbols are created in a similar fashion.
Since you get one symbol with multiple units, it would still map to just a single footprint, though.
To use two footprints, it would have to be two distinct symbols, each mapping 1:1 to a footprint, but then you loose the ability to use functional units - at least that's my understanding of how KiCad 5 works. 

My schematic is finished by now, so I'm not going to rip it all up by changing it to two separate 100 pin symbols.
I have been aware of this problem since I started using the symbol, but have just been putting dealing with transforming it into two footprints off until now.

set-soft, 2022-09-27:
> Since you get one symbol with multiple units, it would still map to just a single footprint, though.

In this case, when you realize you need 2 footprints, you just split the CSV in 2, adjust the pin numbers (can be done with a script or just a spreadsheet) and generate 2 symbols. Each symbols with N sub-symbols. The number will be automatically computed according to the groups you defined.

set-soft, 2022-11-23:
Hi @Unichord !

> Executing custom scripts would be extremely useful for variable replacing, text manipulation outside of KiCad, file copy, etc.

Which of these operations can't be currently done?

> For example I'd like to automate KiKit to panelize PCBs, which can be easily done by custom scripting. I also plan to make a small database with all the boards we have, and external scripting would be handy to populate or extract data from it. Again, the possibilities would be endless.

Panelizing is in my TODO list. In general: why not writing a new output, and perhaps contribute it?

set-soft, 2022-12-01:
Hi @Unichord !

The `dev` branch now has support for KiKit panelization. You can find examples [here](https://github.com/INTI-CMNB/KiBot/tree/dev/tests/yaml_samples) (panelize* files)