INTI-CMNB/KiBot #110: render_3d output naming & orthogonal output

mdeweerd, 2021-11-18:
This is related to KiBot and KiAuto .  If you prefer that I split the issue, I can.

I generate 3D normal and 3D orthogonal output (=2D equivalent).

This is the configuration file:
[testyaml.zip](https://github.com/INTI-CMNB/KiBot/files/7563978/testyaml.zip)

Two things arise:
- The outputs of the default 3D rendering and the orthogonal renderings take the same filename - there several options, but maybe for this case we could add something like "orthogonal" so that the automatic naming rule works without fiddling.
- I increased the "window size" of the 3D output and this seems to impact the position of the "cube" to enable orthogonal rendering, so that's not active in that case.  This requires a trick to position the mouse (probably in respect to the top left corner of the window) to ensure that the offset to the "cube" is always the same.

Otherwise, I think that the implementation works well - its more than expected because you added the color changes, etc as well ;-) AND the orthogonal output.

(I have some kicost stuff in the configuration file to try to get a testcase for something else).

mdeweerd, 2021-11-18:
I editedt the initial messange and I add that the 3d rendering alsow leaves me with some tmp*.pro files after the process .

set-soft, 2021-11-18:
> Could the naming of the output slightly change for the orthogonal projection so that the general rule can be changed without having to update the "output" naming rule specifically for these targets?

I suggest using different target directories. I like to generate red, blue and green PCB views ... should I also add the color to the name? ;-)
But I think I have a solution.

> the 3d rendering alsow leaves me with some tmp*.pro files after the process .

Yes, KiCad creates project files if they don't exist. I committed a patch to remove them.

set-soft, 2021-11-18:
Try the `id_add` option, it will be concatenated with the %i value. Now we can also have `_red` ;-)

closed 2021-11-18 by set-soft

mdeweerd, 2021-11-18:
;-)  I hope that you'll experience the benefit from it ! - id_add is a better solution.  
Codewise, I often end up adding a common FilenameBuilder class or so.

commit 22b946ec (2021-11-18): Added `id_add` to `render_3d` to change the %i expansion

- Closes #110