INTI-CMNB/KiBot #112: Override "output" (& other) options from imported yaml

mdeweerd, 2021-11-21:
Importing Yaml configuration files is excellent for setting up common outputs/filters accross projects.

Most targes do not need to be specialised for a project, but there are cases where overriding some values would be helpful.

For example, for the 3D output the general configuration is ok, but the output size needs to be adjusted.

With this in the imported file ('common/common.kibot.yaml'):
```
kibot:
  version: 1

outputs:
  - name: "render_2d_top"
    type: render_3d
    dir: "+PCBA"
    options:
      id_add: "_2d"
      no_tht: false
      no_smd: false
      no_virtual: true
      background1: "#66667F"
      background2: "#CCCCE5"
      board: "#332B16"
      copper: "#B29C00"
      silk: "#E5E5E5"
      solder_mask: "#143324"
      solder_paste: "#808080"
      move_x: 0
      move_y: 1
      ray_tracing: false
      wait_ray_tracing: 5
      view: "z"
      zoom: 5
      #width: 2560
      #height: 720
      orthographic: true
```

The following in project configuration file would allow us to override some values.  In the example: solder mask color, zoom and output size.
The name is used as a unique key:

```yaml
kibot:
  version: 1

import:
  - common/common.kibot.yaml

outputs:
  - name: "render_2d_top"
      solder_mask: "#030303"
      zoom: 10
      width: 2560
      height: 720
```

set-soft, 2021-11-22:
The idea is good, but I think what you need is addressed by YAML anchors. So I think we should be sure this is the right way to implement it.

mdeweerd, 2021-11-22:
YAML anchros seem to be to avoid repeating (DRY), I am thinking of "override".
The common file could suggest a zoom of "5" dor the 3D model which seems to be ok for most cases, but the project specific zoom might need to be 10 (I actually have such a case).   

I have not found a document specifying how that works, but I think this is pretty much what I mean (done in docker-compose):
https://devilbox.readthedocs.io/en/latest/configuration-files/docker-compose-override-yml.html
https://github.com/internetarchive/openlibrary/blob/master/docker-compose.override.yml

set-soft, 2021-11-25:
> YAML anchros seem to be to avoid repeating (DRY), I am thinking of "override". The common file could suggest a zoom of "5" dor the 3D model which seems to be ok for most cases, but the project specific zoom might need to be 10 (I actually have such a case).

You can do it using anchors. The problem here is how the import works, but this is another story,

set-soft, 2021-12-02:
I added the `extends` attribute for outputs. It allows to take the `options` of another output as base.

Note that both outputs must have different names and that extending one output doesn't disable the other.

closed 2021-12-02 by set-soft

mdeweerd, 2021-12-02:
Thanks @set-soft .
This will already help out to reuse settings.

However, when you find a way to keep the same name, it's even better.  This option serves in particular when one has to tweak an output: zoom of a 3D view or colors for instance.  It would be nice to keep the "makefile" the same or to be able to just run "kibot" without having conflicting similar outputs (the output filename would typically be the original one).

I'll handle this in the project Makefile for now.

set-soft, 2021-12-02:
I understand your use, but in this way you can use it to generate various similar outputs, not just one.

mdeweerd, 2021-12-02:
I agree that it's also an interesting feature.  

I just went through the YAML specification again, and looked at some other tools or applications I use (Swagger, Home Assistant).

IMHO YAML does not define a way to combine several files or "override"/"change" values.
Therefore he tools do this "their way".
Aliases and tags allow reuse, but not overriding values.  

Methods to "include" other yaml files:
- 'import' in kibot,
-  '!include...' kind of includes in [Home Assistant](https://www.home-assistant.io/docs/configuration/splitting_configuration/) (with methods that allow to read all yaml files in a directory),
- Swagger has [$ref](https://swagger.io/docs/specification/using-ref/) to include other files or URLs.

Change/"merge" values:
- Kibot now has 'extends'
- Home Assistant introduced [customize](https://www.home-assistant.io/docs/configuration/customizing-devices/) which enable changing the values of "Entities" that were defined with other values elsewhere (or did not have those values);
- openapi (swagger) proposes the combination of definitions with [allOf, anyOf, oneOf, not](https://swagger.io/docs/specification/data-models/oneof-anyof-allof-not/), but that implies adding another name.

'extends' reminds me of the "Specman 'e' language" I used a long time ago : it's an aspect oriented language used in the verification of integrated circuits.   It pretty much lets you 'extend' and redefine "anything" "anywhere" - the inclusion order is important: the last "override" takes priority.

set-soft, 2021-12-02:
I added a mechanism to:

1. Create the concept of *run by default*. Outputs marked with it are implicitly selected when you don't specify a name in the command line.
2. An output can disable the status of another, so after extending one output you can disable the original.
3. An output can be created as a template that won't be run unless explicitly asked.

This will solve your problem without needing special care in the makefile.

mdeweerd, 2021-12-02:
I think that this will bring the flexibility needed.

I may test if I can create a template, extend it in the common file without any specialisation so that it's a default target, and then extend to the same name in the project, with specialisation.

So then the project's output with the same name might hide the common output.  
I know selective imports are possible already.

set-soft, 2021-12-03:
> So then the project's output with the same name might hide the common output.

Not the same name. You must choose another name.

commit b02ad007 (2021-12-02): Added reference to the issue in the changelog

Changes in a430916a4a8c8ad46aa221b6ffe2c415c77bbb7a
Closes #112