INTI-CMNB/KiBot #113: Add SMD/THT type information to BOM + Statistics

mdeweerd, 2021-11-21:
Several prototype manufacturers ask for:
- The number of unique components;
- The total number of SMD components (and sometimes also subclasses such as BGA);
- The total number of THT components.

At least one asks for it in the BOM (if possible).

In KiCAD this information is not in the schematic, it's in the PCB design file.

It could be a feature to gather information from the schematic and the PCB - I do not know if that's easily feasible because its not in the standard format.

Perhaps there should be an extra field in the schematic (ctype or so), which could be checked against the PCB value as a DRC/ERC check.  Then all information would be available in the schematic (for BOM generation) and it's ensured to be correct through these checks.

set-soft, 2021-11-22:
Hi @mdeweerd !

> Several prototype manufacturers ask for:
> 
> * The number of unique components;

I think this is the groups count.

> * The total number of SMD components (and sometimes also subclasses such as BGA);
> * The total number of THT components.

The new option should solve SMD/THT thing. About the BGA: I think this belongs to the potential tool that collects miscellaneous project information (the one you mention that collects things like min track width).

> At least one asks for it in the BOM (if possible).
> 
> In KiCAD this information is not in the schematic, it's in the PCB design file.

Yes, for this reason this option only works if the PCB is available. Otherwise you get a warning and the option is disabled.

> It could be a feature to gather information from the schematic and the PCB - I do not know if that's easily feasible because its not in the standard format.

We currently get this information from the PCB. Is used to filter components by SMD/THT/Virtual status.

> Perhaps there should be an extra field in the schematic (ctype or so), which could be checked against the PCB value as a DRC/ERC check. Then all information would be available in the schematic (for BOM generation) and it's ensured to be correct through these checks.

I don't think adding redundant information is a good idea.

closed 2021-11-22 by set-soft

mdeweerd, 2021-11-22:
Dear @set-soft 

I am surprised there is already something, but I suspected there was some "easy" possibility.

> > * The number of unique components;
> I think this is the groups count.
"groups count" is another designation I did not read yet (but we are not native english tongue speakers).
For me its the number of different manf/manf# combinations, I think this is what you mean as well. 

> > * The total number of SMD components (and sometimes also subclasses such as BGA);
> > * The total number of THT components.
>(SMD/THT/BGA) I think this belongs to the potential tool that collects miscellaneous project information (the one you mention that collects things like min track width).
Yes, I think this could be done in the same tool.  I know that that would be more effort, I am just proposing here - maybe something can be done with what is at hand, and other things later.  More than helpful for a manufacturer, it should also help improve the quality.

> Yes, for this reason this option only works if the PCB is available. Otherwise you get a warning and the option is disabled.

Thanks for adding an option already!

> > Perhaps there should be an extra field in the schematic (ctype or so), which could be checked against the PCB value as a DRC/ERC check. 
> I don't think adding redundant information is a good idea.
I fully agree, I was thinking about how to make it feasible ;-).

commit 49ea1bb6 (2021-11-22): Internal BoM: `count_smd_tht` option to compute SMD/THT stats.

Closes #113