INTI-CMNB/KiBot #114: Import does not import filters by default.

mdeweerd, 2021-11-21:
It seems that 'filters' are not imported as the  [documentation indicates](https://github.com/INTI-CMNB/KiBot/blob/27b26feb88f2b0dd4fbf5a9f50bbd9387aace7ab/README.md?plain=1#L1832)

The attached test case suggests that filters are not imported.
```
kibot -e tests/board_samples/kicad_5/kibom-variant_4.sch -b tests/board_samples/kicad_5/kibom-variant_4.kicad_pcb -c tests/yaml_samples/step_variant_2.kibot.yaml -g variant=topvariant JLCPCB_bom
```

[tc2.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7577183/tc2.tar.gz)

set-soft, 2021-11-22:
Thanks @mdeweerd !
The problem was when you tried to import filters from multiple files.

closed 2021-11-22 by set-soft

bob-swinkels, 2021-11-26:
Is there a way to circumvent this problem at the moment, or is it best to duplicate the filters to the "master" file?

mdeweerd, 2021-11-26:
@bob-swinkels This was fixed.  I had duplicated the filters to the master file, but removed them after the fix and this works.

bob-swinkels, 2021-11-29:
Thank you @mdeweerd. Should this now also work with the current setsoft/kicad_auto:dev because I still experience this problem when using this image.

bob-swinkels, 2021-11-29:
My bad, it does actually work :)

commit 4dcb289e (2021-11-22): Fixed problems when importing filter from multiple files.

- The 2nd load discarded alredy imported filters
Fixes #114