INTI-CMNB/KiBot #115: Generating pdf_pcb_print failing

mdeweerd, 2021-11-23:
The attached test case includes the logs using the kibot_variants_arduprog example.

I get the same result in another project.

```
- 'Exports the PCB to the most common exhange format. Suitable for printing.' (pdf_pcb) [pdf_pcb_print]
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot.kiplot - log.py:88)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot.kiplot - log.py:88)
ERROR:pcbnew_do returned 1 (kibot.out_pdf_pcb_print - out_pdf_pcb_print.py:131)
```

The logs are in 'Generated/' - output from the last kibot run is in kibot.log - the Makefile starts all kibot runs.  You can do 'make clean' to clean up the outputs.

[t1.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7588331/t1.tar.gz)

set-soft, 2021-11-25:
Next time please try to narrow the problem.
I found the root of the problem is differences in KiCad behavior when we fill zones and then try to print, sometimes the selected widget changes, must find when and how to avoid it.
As a workaround just skip the DRC (do it in a separated run) and you shouldn't have this problem.

set-soft, 2021-11-25:
I fixed a problem in KiAuto, v1.5.12. This should solve the problem

closed 2021-11-25 by set-soft

mdeweerd, 2021-11-25:
I understand that this was not easy to find, on my end I tried to understand what happened, I even checked frame by frame using
`ffmpeg -i pcbnew_export_screencast.ogv  out%03d.png` and made some interactive checks.

I could could not determine precisely what happened.
I could have tried many more combinations of options, etc, but it's not always easy to find the right 'minimum set' of combinations.

Then I verified using the example project, set up the test case, clean targets, output combination in Generated, and verified the package to ensure that the issue repeated after unpackaging.

In short, I appreciate your effort, I am doing some effort as well. 

I updated my local copies of kibot & KiAuto: I am still running into the issue using
```
kibot -c t1.kibot.yaml -b t1.kicad_pcb -e t1.sch -g variant=default   pdf_pcb 
```
(DRC/ERC are active).

mdeweerd, 2021-11-25:
Sorry, works fine for me too - It works (I pulled but not correctly).  Perfect!