INTI-CMNB/KiBot #116: Extend global '%.' variable expansion.

mdeweerd, 2021-11-25:
Continued from #104 

>> Currently the "script" can be run for one global variant at a time - this is the variant that I'ld expect in the "dir".
> I guess you mean: Currently I intend to ... because KiBot doesn't impose this limitation.

The global variant or default variant is only for one variant, correct?
So if I want to generate the outputs for 'default', 'USB' and 'XTAL' then I have to run kibot 3 times unless I repeat all targets three times in the YAML.

So when I run kibot 3 times, then I want to have 3 different output directories, each referring to the related global variant.

>> I'ld expect "dir" to be expanded before applying it, especially when the output dir is specified as '+SUBDIR'.
> This isn't the way things work. As I told you dir is an attribute of output (as the YAML describes).

Yes, I understood this is not the way things work, and I am expressing this is not the way I'ld expect.  In other words, yes, I'ld expect a tool to do as it's documentation states, but that does not mean that this is the behavior I'ld expect to be the actual and documented one.

Maybe we need a "%g" to refer to the global variant.

> And it doesn't. The global directory is the one you specify using -d command line option,..

I understand that this is the "global dir" to kibot, but it doesn't know about the date, the variant, etc.  Unless I can put the '%' expansions there as well (and I think they are not expanded for the -d option).  
But if the '-d' would have to serve for the purpose of combining the outputs per variant, I'ld prefer that it ('-d' value) can be set in the YAML file itself.

I think that you understand what I mean/what the goal is: have a single directory for each variant, generated from the parameters, with a good name including design, design version and date.

I understand that some items like '%i', '%x' are dependent on the target exclusively.  And should therefore not be mentioned in the directory for this use case.  
The case '%d' is manageable except when outputs do not depend on the PCB or(exclusive) SCH (like compress).  So to keep '%d' and '%D' as they are, and not changing their meaning according to context (globals' dir or specific output), there could be another '%.' value to use the latest SCH or PCB date (or just one of them, assuming that the user can make them equal).

Maybe a global option to specify what this new date expansion refers to ( "dateref: sch, pcb, FILE" ).

> But they dates are different! So you'll get one directory name when processing the schematic and another when processing the PCB.
Yes, for PCB and SCH that is manageable, but for compress and custom targets?

> This looks like magic, I can expect a user to understand it.
Yes, I think another '%' expansion can do, I proposed '%g', you as well ;-).

So in summary (subject to discussion/change/...):
- Add '-d' option to yaml, allow command line option to override it, and allow '%' expansion in the parameter (at least those that can be globally known, after getting information from SCH and PCB).
- Add '%g' expansion to refer to global variant;
- Add '%G' (or other) expansion to refer to global date.  
- Add some '(date)ref' global to define what 'global date' & sch/pcb specific tokens (%r,%f,%F,...)  mean in global context (to expand global dir).  Could default to schematic.

set-soft, 2021-11-30:
Hi @mdeweerd !

> Add '-d' option to yaml, allow command line option to override it, and allow '%' expansion in the parameter (at least those that can be globally known, after getting information from SCH and PCB).

Now the global `out_dir` option can be used to define the same as `-d`. 

> Add '%g' expansion to refer to global variant;

Now `%g` expands to the `file_id` of the global variant and `%G` to the name.

> Add '%G' (or other) expansion to refer to global date.
> Add some '(date)ref' global to define what 'global date' & sch/pcb specific tokens (%r,%f,%F,...) mean in global context (to expand global dir). Could default to schematic.

Now you can use `%sd` to get the date of the schematic and `%bd` for the date of the PCB. Same for the other patterns.

In the example you posted in #104 the global definition should be:

```yaml
global:
  variant: topvariant
  out_dir: "%sf_%sr_%sd%G"
  output: "%f_%r_%d-%i.%x"
```

This will expand the schematic data and the global variant. Using it you don't need to use `+...` in the included YAML.
Let me know if you can make it work. The patch is quite complex and was tested with a very naive case.

mdeweerd, 2021-11-30:
Thanks for the changes.

I will test it on a real (& "complex" with regards to variants) project.

mdeweerd, 2021-12-01:
@set-soft 
In my test this works as expected, I only have a comment about the dateformat.
With %sd it is 2020-11-09.  I prefer the date without the dashes in filenames - its shorter and I find it more readable in a file or directory name.

I have an NPTH drill gbr and pdf file appearing in my PDF directory - I'll check if that is my configuration.

I'll do some further testing, for example, get a date in the name of the schematic files (I suppose I have to improve my configuration).

mdeweerd, 2021-12-01:
Almost perfect in my tests.

Some minor things with regards to the naming:
- The gbrjob file did not contain the date while the PCB files did.
- The drill files get '-None' in their name (I think that '-None' should be omitted).

Tested core global setting, no -d specified to kibot:
```yaml
kibot:
  version: 1

import:
  - common.kibot.yaml

global:
  output: "%f_%r_%d-%i.%x"
  #dir: "%f_%r_%d%v/"
  out_dir: "%sf_%sr_%sd_%g/"
  # kiauto_time_out_scale: 2
  # kiauto_wait_start: 20
  variant: project_variant
```

And this imported common file:
[yaml.zip](https://github.com/INTI-CMNB/KiBot/files/7633723/yaml.zip)

set-soft, 2021-12-01:
> With %sd it is 2020-11-09. I prefer the date without the dashes in filenames - its shorter and I find it more readable in a file or directory name.

The a503c5fdc6ad6fc989f40f2cf3b51ace40e6162f commit allows configuration for this.

set-soft, 2021-12-01:
Hi @mdeweerd !

> * The gbrjob file did not contain the date while the PCB files did.

Fixed, it was using the default global pattern, not the one defined by the user.

> * The drill files get '-None' in their name (I think that '-None' should be omitted).

Well ... you explicitly asked for `None`, take a look at your file ;-)
Ok, I realize this was a missleading example, now it says `null`, which in YAML means the same as `None` in Python.

closed 2021-12-01 by set-soft

mdeweerd, 2021-12-01:
😄 Yes, my file based on one of the examples ...    I understand: at the end of the day we forget what language we are speaking - more than many translaters !