INTI-CMNB/KiBot #117: Timeout while waiting for pcbnew during run DRC

hkleen, 2021-11-26:
When I try to run kibot I get the following error:

`- Running the DRC
ERROR:Time-out waiting for pcbnew, giving up (pcbnew_do - pcbnew_do:165) (kibot.kibot.kiplot - kiplot.py:145)
ERROR:DRC returned 9 (kibot.kibot.pre_drc - pre_drc.py:63)
`
Looking at the pcbnew_run_drc_screencast.ogv I can see, that pcbnew is actually started, but the program just hangs in wait_for_window.

### System Information
- Ubuntu 20.04
- KiBot 0.11.0
- Kicad 5.1.12

mdeweerd, 2021-11-26:
@hkleen Can you provide a testcase, the ogv?

Which KiAuto are you using?

There was a change in kiauto (1.5.12-1) with related to zone filling, did you use that version.  If not, can you check if you have the same problem in kiauto (1.5.11-1) ?

hkleen, 2021-11-27:
I'm using version 1.5.12 of KiAuto.

I get this timeout for every Kicad project I tried, so I guess the reason might be some incompatibility with my SW setup. I attached a very simple testcase, that also shows this behavior for me.
[simple-testcase.zip](https://github.com/INTI-CMNB/KiBot/files/7611553/simple-testcase.zip)
[image]

mdeweerd, 2021-11-27:
I tested using the Docker container setsoft/kicad_auto:dev where it works normally.

You could do:
```
kibot -vvvv  |& grep "'pcbnew_do"
#or
./runInKiAuto kibot -vvvv  |& perl -n -e 'if(/\[.pcbnew_do/){s/^.*\[(.*)].*/$1/;s/,//g;print;}'
```
to find the command executed for pcbnew_do.
In my case:
````
'pcbnew_do' '-r' '-vvvv' 'run_drc' '-o' '/root/workdir/simple-drc.txt' '/root/workdir/simple.kicad_pcb' '/root/workdir/.'
```

Run that, and share the log.

And then do:
```
'pcbnew_do' '-r' '-vvvv' 'run_drc' '-o', '/root/workdir/simple-drc.txt' '/root/workdir/simple.kicad_pcb' '/root/workdir/.
```

Kicad 5.1.9
> pcbnew_do --version
pcbnew_do 1.5.12 
> kibot -- version
KiBot 0.11.0

hkleen, 2021-11-29:
I did as you asked, this resulted in the following log:
[error.log](https://github.com/INTI-CMNB/KiBot/files/7616403/error.log)
I also did some debugging on my own and it seems to me that the call to "xdotool search" looking for the main pcbnew window doesn't find the window. Which seems strange, since the recording shows that the window does indeed open.

mdeweerd, 2021-11-29:
Yes, it's strange.

@set-soft For debugging purposes you may want to add the equivalent of the next bash line in debug mode when a window is not found.  It lists the names of all the windows found:
````
for i in  $(xdotool search --name '.*') ; do echo $i ; xdotool getwindowname $i ; done
```

And possibly even capture the root window at the moment the window can not be found using the installed Imagemagick toolset (complementary to recordmydesktop because the screenshot can be related directly to the moment of failure with relevant naming):
```
import -window root pctnew-notfound-N.png
```

hkleen, 2021-11-29:
... and it keeps getting stranger. So for debugging I added the following code to debug_window in ui_automation.py:
```
logger.debug("Visible windows:")
window_id = xdotool(['search', '--onlyvisible', '--name', 'FOOBAR']).splitlines()
for i in window_id:
    logger.debug("window ID: " + str(i) + "; name: " + str(xdotool(['getwindowname', i])))
logger.debug("All windows:")
window_id = xdotool(['search', '--name', '.*']).splitlines()
for i in window_id:
    logger.debug("window ID: " + str(i) + "; name: " + str(xdotool(['getwindowname', i])))
```
which gives me the following output:
```
INFO:Waiting for "Main pcbnew window" ...
DEBUG:Visible windows: (pcbnew_do.kiauto.ui_automation - ui_automation.py:254)
DEBUG:window ID: b'1294'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:257)
DEBUG:window ID: b'4194412'; name: b'Pcbnew \xe2\x80\x94 /home/henningk/simple/simple.kicad_pcb\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:257)
DEBUG:window ID: b'4194413'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:257)
DEBUG:All windows: (pcbnew_do.kiauto.ui_automation - ui_automation.py:258)
DEBUG:window ID: b'1294'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194305'; name: b'pcbnew\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194307'; name: b'pcbnew\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194311'; name: b'pcbnew\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194320'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194412'; name: b'Pcbnew \xe2\x80\x94 /home/henningk/simple/simple.kicad_pcb\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194308'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194312'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
DEBUG:window ID: b'4194413'; name: b'\n' (pcbnew_do.kiauto.ui_automation - ui_automation.py:261)
...
```
OK, so there seems to be window with the name we are looking for, weird...

If I now change the search pattern from ".*" to some name that does not exist, say "FOOBAR" it kind of starts to work...
```
INFO:Waiting for "Main pcbnew window" ...
DEBUG:Visible windows: (pcbnew_do.kiauto.ui_automation - ui_automation.py:254)
DEBUG:Pcbnew is no longer running (returned None) (pcbnew_do - pcbnew_do:150)
INFO:Open Inspect->DRC
INFO:Waiting for "DRC modal window" ...
DEBUG:Found DRC modal window window (1) (pcbnew_do.kiauto.ui_automation - ui_automation.py:323)
DEBUG:Window id: b'4195074' (pcbnew_do.kiauto.ui_automation - ui_automation.py:328)
DEBUG:Waiting for b'4195074' window to get focus... (pcbnew_do.kiauto.ui_automation - ui_automation.py:281)
DEBUG:Currently focused id: b'4195074' (pcbnew_do.kiauto.ui_automation - ui_automation.py:284)
INFO:Enable reporting all errors for tracks
INFO:Pasting output dir
INFO:Waiting for "Report completed dialog" ...
DEBUG:Found Report completed dialog window (1) (pcbnew_do.kiauto.ui_automation - ui_automation.py:323)
DEBUG:Window id: b'4201071' (pcbnew_do.kiauto.ui_automation - ui_automation.py:328)
DEBUG:Waiting for b'4201071' window to get focus... (pcbnew_do.kiauto.ui_automation - ui_automation.py:281)
DEBUG:Currently focused id: b'4201071' (pcbnew_do.kiauto.ui_automation - ui_automation.py:284)
INFO:Waiting for "DRC modal window" ...
DEBUG:Found DRC modal window window (1) (pcbnew_do.kiauto.ui_automation - ui_automation.py:323)
DEBUG:Window id: b'4195074' (pcbnew_do.kiauto.ui_automation - ui_automation.py:328)
DEBUG:Waiting for b'4195074' window to get focus... (pcbnew_do.kiauto.ui_automation - ui_automation.py:281)
DEBUG:Currently focused id: b'4195074' (pcbnew_do.kiauto.ui_automation - ui_automation.py:284)
INFO:Closing the DRC dialog
INFO:Waiting for "Main pcbnew window" ...
```
The second time waiting for the pcbnew window still fails and the first time it spends the whole timeout period (60s) on the line "Waiting for...", but then for some reason kind of works. Although the debug_window function should not be called in this case so changes there should not change anything.

Another observation: If I catch the CalledProcessError that is caused by searching for the "FOOBAR" window it stops working again.

In case this is somehow dependent on the python version
```
python3 --version
Python 3.8.10
```

mdeweerd, 2021-11-29:
What if your add a space after "Pcbnew" in pcbnew_do

```python
    return wait_for_window('Main pcbnew window', r'Pcbnew ', time, others=others, popen_obj=popen_obj)
```

That still works for me and may help with the multiple windows having the same name.
When there are multiple windows that match a name, the method takes the second one in the list, but it seems that the actual window is the 4th one matching 'Pcbnew' (case insensitive).

hkleen, 2021-11-29:
Adding the space after "Pcbnew" did not change the behavior for me.
Also in the list of visible windows there is only one window named 'Pcbnew'. For some reason the call to "xdotool" returns with returncode "1" even if there is a window matching the search pattern.
```
xdotool --version
xdotool version 3.20160805.1
```

ferdinandkeil, 2021-11-29:
Can confirm this issue. System setup is identical. Base is a fresh Ubuntu install.

`kibot` just freezes and times out after some time. There's no PcbNew window.

### System Information

- Ubuntu 20.04
- KiBot 0.11.0
- Kicad 5.1.12

mdeweerd, 2021-11-29:
To improve my docker skills, I created a Ubuntu20.04 setup.

Adding
```
RUN  apt-get update && apt-get install -yq locales && rm -rf /var/lib/apt/lists
ENV _TARGET_LANG=en_AG.UTF-8
RUN sed -i "/${_TARGET_LANG}/s/^# //g" /etc/locale.gen && \
    locale-gen
ENV LANG $_TARGET_LANG
ENV LANGUAGE en_AG:en
ENV LC_ALL en_AG.UTF-8
```

Make the DRC fail.

hkleen, 2021-11-30:
> One recent change that could be generating problems (even when shouldn't) is the locale detection. Can you try changing line 265 of misc.py from:
> 
> ```python
>     return res
> ```
> 
> To:
> 
> ```python
>     return 'C.UTF-8'
> ```
That seems to do the trick!! It's working for me now. I would not have guessed that the unicode characters in the window name cause xdotool search to fail. 

For reference my locale config:
```
$locale
LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=
```

set-soft, 2021-11-30:
> To improve my docker skills, I created a Ubuntu20.04 setup.

Thanks  @mdeweerd for confirming it.

closed 2021-11-30 by set-soft

set-soft, 2021-11-30:
Hi @hkleen and @ferdinandkeil !

The 1.5.13 version of KiAuto should fix it. I also updated the docker image.

Please let me know if it works for you. I added an english locale to my system and verified the problem (some bug in xdotool). With this patch the problem went away. Now KiAuto isn't relying on xdotool to do the regex match, is doing it by itself.

ferdinandkeil, 2021-11-30:
Wow, that was quick! Thank you @set-soft :+1:

Seems to work now.

hkleen, 2021-12-01:
Awesome! Works for me too now. Thanks for the quick response @set-soft