INTI-CMNB/KiBot #118: Generate production data without schematic file

ferdinandkeil, 2021-11-30:
Hi,

I want to use KiBot to streamline the process of creating PCB production data (Gerber, Excellon) from Pcbnew files (.kicad_pcb). Unfortunately, KiBot is unhappy that I'm missing a schematic file when trying to generate the Gerber data. Is there some switch that I didn't find to get this working?

I have attached my configuration (it's basically the example config): [gerber_out.kibot.yaml.txt](https://github.com/INTI-CMNB/KiBot/files/7627505/gerber_out.kibot.yaml.txt)

set-soft, 2021-11-30:
Hi @ferdinandkeil !
You have to avoid specifying a variant or filter. In this case just remove the `dnf_filter: ''` line and the problem will be fixed.
I'm making a better example to avoid this problem.

closed 2021-11-30 by set-soft

ferdinandkeil, 2021-11-30:
Hi @set-soft !
Thank you once more for the quick reply 👍 That did the trick!

commit fadb2557 (2021-11-30): Changed examples to avoid schematic dependencies.

Closes #118
Now the examples use `_none` as empty filter.