INTI-CMNB/KiBot #120: Missing tool option(s) - kicost "split_extra_fields"

mdeweerd, 2021-12-14:
KiBot 0.11.0 

I was looking to add 'split_extra_fields' to the kicost parameters, but this was not available.  The 'board_qty' seems to be missing as well.

I suppose that the goal is that tool options are made available as yaml options, but keeping up with that may be a burden.  Maybe there should be a more generic options that allows the user to add any option to the tool's execution.

set-soft, 2021-12-15:
I added them, but didn't test if they work. Please let me know it.

About the generic option: not sure ... this will kill the report of issues like this one ... if you take a look at the patch you'll see it adds interesting stuff.

closed 2021-12-15 by set-soft

mdeweerd, 2021-12-15:
I did not get a complaint from the tool for the split.. option, but I haven't verified its effect yet.

"board_qty" failed:
```
WARNING:(W015) Unknown option `board_qty` (kibot.optionable - log.py:88)
```

And the `fields` option just adds a single column with the header "comment,S1MN,S1PN,S2MN,S2PN":
```yaml
  - name: 'kicost'
    type: kicost
    dir: "+PCBA"
    options:
      currency: [ EUR ]
      no_collapse: true
      no_price: true
      fields:
        - comment
        - S1MN
        - S1PN
        - S2MN
        - S2PN
      board_qty: 1
      show_cat_url: true
```

I yet have to look for the generic option (it's a 'bit' late here).

set-soft, 2021-12-16:
> "board_qty" failed:

The name is `number`, I used it to make it coherent with KiBoM and the internal BoM. Is documented.

> I yet have to look for the generic option (it's a 'bit' late here).

I didn't add a generic option.

> I did not get a complaint from the tool for the split.. option, but I haven't verified its effect yet.

You can use the debug mode to see the command used to invoke KiCost.

mdeweerd, 2021-12-16:
> Generic option
I read too fast: "About the generic option ... you'll see it adds interesting stuff" ;-).  
I agree that having the option will likely limit reports ("issues") about missing options.

> debug
This is from the version I tested yesterday:
```
l default variant field renamer filter (KiCost style)'} (kibot.fil_base - fil_base.py:232)
DEBUG:Running: ['kicost', '-w', '-o', '/root/workdir/PROJECT_02_20211214__5V_3btn/PCBA/PROJECT_02_20211214-kicost.xlsx', '-i', '/root/workdir/PROJECT.xml', '--debug=4', '--no_price', '--no_collapse', '--show_cat_url', '--currency=EUR', '--split_extra_fields=h,comment,desc,lcsc#,S1MN,S1PN,S2MN,S2PN', '--fields=comment,S1MN,S1PN,S2MN,S2PN'] (kibot.out_kicost - out_kicost.py:188)
```

In my original CLI command, my kicost call uses "-f comment S1MN ..." (multiple parameters, no comma separation) for the fields which is what is documented for kicost, including for ignore_fields, split_extra_fields, etc.
At least for the "-f" or "--fields" parameters the comma separator is not effective.
So either kicost needs to change or kibot.

set-soft, 2021-12-16:
Thanks for testing, should be solved now

commit 96a5f3da (2021-12-15): Added support for new KiCost options `split_extra_fields` and `board_qty`.

Closes #120