INTI-CMNB/KiBot #121: "Extends" & "date_format" feedback following "test"

mdeweerd, 2021-12-14:
Hi @set-soft
This is related to #116 and #112 .

I just put the "extends" option into practice.  Which allowed met to adjust the 3D rendering outputs to a black solder mask as well as adjusting zoom and offset:
```yamloutputs:
  - name: render_2d_top_black
    type: render_3d
    extends: render_2d_top
    dir: "+PCBA"
    disable_run_by_default: render_2d_top
    options:
      zoom: 10
      width: 4096
      solder_mask: '#1A1A1A'
      copper: '#2A2A2A'
      move_y: 1
      height: 1024
  - name: render_2d_bottom_black
    type: render_3d
    extends: render_2d_bottom
    dir: "+PCBA"
    disable_run_by_default: render_2d_bottom
    options:
      zoom: 10
      width: 4096
      solder_mask: '#1A1A1A'
      copper: '#2A2A2A'
      move_y: 1
      height: 1024
  - name: render_3d_top_black
    type: render_3d
    extends: render_3d_top
    dir: "+PCB"
    disable_run_by_default: render_3d_top
    options:
      zoom: 8
      width: 4096
      solder_mask: '#1A1A1A'
```

Some observations:
- The option 'type' needs to be specified, but we could assume that this option is provided due to "extends".
- disable_run_by_default: it's ok, but I expected this to be a boolean (disable the parent rule or not).  This is more flexible though as one can disable another rule that extends the same "template".
- The dir option is not inherited (I had '+PCBA' and '+PCB' but the output was in the root of the global output directory), so I set them again (see example above).
- The output type 'bom' does not support the 'id_add' option
- I did not get the expected dateformats with the following configuration - I removed time from the date_time_format as my "out_dir" has two underscores ('_') between %d and %g:
```yaml
global:
  output: "%f_%r_%d-%i.%x"
  #out_dir: "%f_%br_%bd_%g/"
  out_dir: "%bf_%r_%d_%g/"
  date_time_format: '%Y%m%d_%H%M%S'
  date_format: '%Y%m%d'
  time_format: '%H%M%S'
  variant: '5V_3btn'
  # variant: 'proto'
  #date_time_format: '%Y%m%d_%H%M%S'
  date_time_format: '%Y%m%d'
  date_format: '%Y%m%d'
  time_format: '%H%M%S'
```

set-soft, 2021-12-14:
Hi @mdeweerd !

> * The option 'type' needs to be specified, but we could assume that this option is provided due to "extends".
> * The dir option is not inherited (I had '+PCBA' and '+PCB' but the output was in the root of the global output directory), so I set them again (see example above).

This is logic because what you inherit is the `options`, not the whole output.

> * disable_run_by_default: it's ok, but I expected this to be a boolean (disable the parent rule or not).  This is more flexible though as one can disable another rule that extends the same "template".

I think it can be extended to support just `true`

> * The output type 'bom' does not support the 'id_add' option

Ok, it can be added, isn't a generic option.

> * I did not get the expected dateformats with the following configuration - I removed time from the date_time_format as my "out_dir" has two underscores ('_') between %d and %g:

Please tell me what did you expect, how did you configured it to get it and what did you get. So I don't need to guess.

mdeweerd, 2021-12-14:
@set-soft 
1. Regarding 'extends', maybe the documentation can be improved a bit.  When I read "options", I think of all the options.  I suggest to write: "extends: [string=''] Copy the `options` field from the indicated output.`

2. Regarding the date format: the configuration is provided at the end of the initial message.  
  What I got is: "PROJECTNAME_02_2021-12-14__5V_3btn" as the output director, but similar in output filenames.  
  What I expected is "PROJECTNAME_02_20211214_5V_3btn".  
  '02' is the revision, 2021-12-14 is the date I have in the schematic and the PCB, `5V_3btn` is the variant.
  So basically, I expect the date and time format specification to applicable to all dates and times or datetimes.

set-soft, 2021-12-14:
> * Regarding 'extends', maybe the documentation can be improved a bit.  When I read "options", I think of all the options.  I suggest to write: "extends: [string=''] Copy the `options` field from the indicated output.`

I added some clarification.

> * Regarding the date format: the configuration is provided at the end of the initial message.
>   What I got is: "PROJECTNAME_02_2021-12-14__5V_3btn" as the output director, but similar in output [...]

Not sure if even posible. If you have an **string** in your schematic with "2021-12-14" as content ... this is not a date, is just an string. 
I did a couple of tests and looks like the `<<<` button in KiCad's dialog always uses **YYYY-MM-DD** as format, so I could add some magic for this case, but this is just a string, you can put anything you want there.

mdeweerd, 2021-12-15:
Technically I suppose that the field is a free-format field, but in practice (confirmed by your tests), it will have the same format for (almost) 100% of the users that fill this field in the suggested way.
When empty, this field already generates a date-time based on the file timestamps as far as I remember.  
If the field does not match the date format, KiBot could issue a warning and either use the field "as-is" or use the file's timestamp.  
I think it's fair to assume that users use this field the "suggested" way.

set-soft, 2021-12-15:
> Technically I suppose that the field is a free-format field, but in practice (confirmed by your tests), it will have the same format for (almost) 100% of the users that fill this field in the suggested way.

But will most probably fail for files created before the button was added.

> When empty, this field already generates a date-time based on the file timestamps as far as I remember.

As explained in the documentation.

> If the field does not match the date format, KiBot could issue a warning and either use the field "as-is" or use the file's timestamp.
> I think it's fair to assume that users use this field the "suggested" way.

The above commit implements it. Can be disabled, I documented it. The `test_date_format_1` and `test_date_format_2` regression test are related.

mdeweerd, 2021-12-15:
@set-soft I tested with my specification, the generated directory date format is as I expected.  Thanks.

I have one underscore too many, but I'll check that later.

set-soft, 2021-12-16:
@mdeweerd, nice. Be aware that some characters are converted to underscore to avoid annoying characters in filenames.

set-soft, 2022-03-03:
Hi @mdeweerd : What's the current status of this issue. I think things are working.

mdeweerd, 2022-03-03:
I am not currently doing PCB stuff - if the underscore can't be explained/fixed by me, I'll open a new issue for that.

closed 2022-03-03 by mdeweerd