INTI-CMNB/KiBot #122: grammar for symbol and footprint library from digikey

ganghuang, 2021-12-17:
I am new on using this KiBot.
After install the ver 0.11.0, I made an example yaml and change the name to myproject.kibot.yaml
Then I run 
kibot -b myproject.kicad_pcb -c myproject.kibot.yaml
I got error on the 
ERROR:At line 7 of `digikey-kicad-library/digikey-symbols/dk_Ferrite-Beads-and-Chips.lib`: Malformed component field (kibot.kiplot - kiplot.py:234)

The library is come from digikey provided great library for lots of components at https://github.com/Digi-Key/digikey-kicad-library.
They work in my kicad 5.1.9. 

So I hack in the 
/home/myname/.local/lib/python3.9/site-packages/kibot/kicad/v5_sch.py
and find the line 115 is 
                          r'([LRCBT][IN][BN])\s*'       # 8 VJustify+Italic+Bold
which does not allow space between the 'C N N'.
I added \s*  in between 
                          r'([LRCBT]\s*[IN]\s*[BN])\s*'       # 8 VJustify+Italic+Bold
And that pass that check. 

Then the problem shows up at 
DEBUG:Loading doc-lib `digikey-kicad-library/digikey-symbols/dk_Ferrite-Beads-and-Chips.dcm` (kibot.kicad.v5_sch - v5_sch.py:703)

with this error message

```
Traceback (most recent call last):
  File "/home/myname/.local/bin/kibot", line 16, in <module>
    main()
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/__main__.py", line 362, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kiplot.py", line 374, in generate_outputs
    config_output(out)
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kiplot.py", line 340, in config_output
    load_sch()
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kiplot.py", line 260, in load_sch
    load_any_sch(GS.sch, GS.sch_file, GS.sch_basename)
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kiplot.py", line 229, in load_any_sch
    sch.load_libs(file)
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kicad/v5_sch.py", line 1632, in load_libs
    o.load(file)
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kicad/v5_sch.py", line 706, in load
    line = f.get_line()
  File "/home/myname/.local/lib/python3.9/site-packages/kibot/kicad/v5_sch.py", line 86, in get_line
    res = self.f.readline()
  File "/usr/lib/python3.9/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xae in position 455: invalid start byte
```

This time it goes beyond what I can do by myself. 
I opened the dcm file mentioned above, and search for the 0xae or even any non ascii char, I didn't find it.

This time the error message does not tell me what happened enough for me to figure it out. 
Can you help me on this?

Or even better try to get some more sample libraries from the digikey library and test if that works?

Thanks

set-soft, 2021-12-17:
Hi @ganghuang !

These files are really wrong and you should report it to the project maintainers.

1. The *.lib file uses `C N N` instead of `CNN`. KiCad never writes it as `C N N` and, even when you don't get an error, it doesn't work. I tried `C I B` and it didn't work, the italic and bold were ignored.
2. The *.dcm file is also wrong, the encoding for these files is UTF-8 and the file you mention contains invalid UTF-8. KiCad silently ignores these lines, just load these components in the library editor and you'll see the faulty components doesn't have keywords, even when you can see them in the file, but the problem is a stray 0xAE character (ISO-8859-1 registered mark symbol).

I'll add some tolerance to these errors, but the files are really corrupted.

closed 2021-12-17 by set-soft

set-soft, 2021-12-17:
I reported it to the original project, issues Digi-Key/digikey-kicad-library#162 and Digi-Key/digikey-kicad-library#163

commit 95ddda09 (2021-12-17): Made the v5 schematic loader more robust

- Allow DMC files with wrongly encoded chars
- Allow libs with wrong justify+italic+bold fields
- Both from https://github.com/Digi-Key/digikey-kicad-library
Closes #122