INTI-CMNB/KiBot #123: "NoneType' object is not subscriptable"

mdeweerd, 2021-12-17:
A fatal script error - I checked it's not a "None" specification in the yaml.
It also happened before updating KiBot.
If needed I can try to make a test case, hopefully it will be clear from the code.

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 362, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 374, in generate_outputs
    config_output(out)
  File "/usr/local/lib/python3.7/dist-packages/kibot/kiplot.py", line 342, in config_output
    out.config(None)
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_base.py", line 98, in config
    super().config(parent)
  File "/usr/local/lib/python3.7/dist-packages/kibot/optionable.py", line 191, in config
    self._perform_config_mapping()
  File "/usr/local/lib/python3.7/dist-packages/kibot/optionable.py", line 166, in _perform_config_mapping
    v.config(self)
  File "/usr/local/lib/python3.7/dist-packages/kibot/out_any_drill.py", line 77, in config
    super().config(parent)
  File "/usr/local/lib/python3.7/dist-packages/kibot/optionable.py", line 191, in config
    self._perform_config_mapping()
  File "/usr/local/lib/python3.7/dist-packages/kibot/optionable.py", line 124, in _perform_config_mapping
    if cur_doc[0] == '[':
TypeError: 'NoneType' object is not subscriptable
```
This is the log file (some information has been obfuscated):
[kibot.log](https://github.com/INTI-CMNB/KiBot/files/7734543/kibot.log)

This happens on the JCLPCB_drill target from this included yaml:
```
  - name: JLCPCB_drill
    comment: Drill files compatible with JLCPCB
    type: excellon
    dir: "+JLCPCB"
    options:
      pth_and_npth_single_file: false
      pth_id: '-PTH'
      npth_id: '-NPTH'
      metric_units: true
      map: gerber
      route_mode_for_oval_holes: false
      output: "%f%i.%x"

```

set-soft, 2021-12-17:
Thanks! for some reason I forgot (or lost) the documentation for `route_mode_for_oval_holes`

The code was tolerant to this, now emmits an error when generating the docs.

closed 2021-12-17 by set-soft

mdeweerd, 2021-12-17:
Works now with your fix ;-).

commit 267cc384 (2021-12-17): Fixed missing documentation for `route_mode_for_oval_holes` option

Closes #123