INTI-CMNB/KiBot #125: Gerber outputs with solder paste for all pads regardless of default variant

mdeweerd, 2021-12-18:
The goal is to add a full set of gerber production files with solder paste for all components alongside the variant specific outputs (controlled through the global variant option provided on the CLI).

This is the yaml file that is imported in the project yaml file.
I have set the variant to the empty string '', but that does not override the global.
I figure that I could add another variant that I call 'all' that would select all components (including dnp components).

@set-soft What do you suggest?

```yaml
# Gerber and drill files with paste for all SMD PADS
kibot:
  version: 1

outputs:
  - name: allpads_gerbers
    comment: Gerbers with all the pads
    type: gerber
    dir: "+ALLPADS"
    options: &gerber_options
      variant: ''
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: true
      #output: "%f.%x"
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      disable_aperture_macros: true
      line_width: 0.1
      subtract_mask_from_silk: false
      #inner_extension_pattern: '.gl%N'
    layers:
      - copper
      - F.SilkS
      - B.SilkS
      - F.Mask
      - B.Mask
      - F.Paste
      - B.Paste
      - Edge.Cuts

  - name: allpads_drill
    comment: Drill files compatible with PCBWay
    type: excellon
    dir: "+ALLPADS"
    options:
      variant: ''
      metric_units: false
      minimal_header: true
      zeros_format: SUPPRESS_LEADING
      # left_digits: 3
      # right_digits: 3
      # See https://github.com/INTI-CMNB/kicad-ci-test-spora/issues/1
      # and https://docs.oshpark.com/design-tools/gerbv/fix-drill-format/
      left_digits: 2
      right_digits: 4
      pth_and_npth_single_file: false
      pth_id: ''
      npth_id: '-NPTH'
      #output: "%f%i.drl"

  - name: allpads_zip
    comment: ZIP file for PCBWay
    type: compress
    #dir: "+ALLPADS"
    options:
      format: ZIP
      files:
        - from_output: allpads_gerbers
          dest: /
        - from_output: allpads_drill
          dest: /
```

set-soft, 2021-12-18:
Please provide a small example, I don't know how to reproduce the situation.

mdeweerd, 2021-12-18:
@set-soft Ok, if its not obvious enough, you can ask me faster - I thought this was a matter of configuration.  Let me some time to make a test case.

mdeweerd, 2021-12-24:
Hi @set-soft  - Seasons greetings!

I made a test case for this an and another "issue".

The test case shows that I can generate the Gerbers with all pads with a global variant active.  I set the variant to '' in 'yaml/pcbfull.kibot.yaml" and the files generated in the ALLPADS subdirectory are ok - there is solder paste for all pads, while in the "PCB" directory, the gerber files have solder paste only for the selected components.

[tc5.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7774859/tc5.tar.gz)

You can close this.

closed 2021-12-29 by set-soft