INTI-CMNB/KiBot #126: Support KiCad 6.0

neilenns, 2021-12-26:
I have no idea if there's any work required for this but... KiCad 6.0 is officially out. Can KiBot be updated (does it even need updates?) to work with 6.0?

set-soft, 2021-12-27:
Hi @neilenns !
Support for 6.0 is in progress, more of the stuff was working with 6.0.0 RC1.
I reported a couple of bugs affecting KiBot:

1. The STEP output, somebody thought it was a good idea to make `kicad2step` a graphical tool without any kind of option to make it work from command line.
2. The 3D Viewer window doesn't check if the OpenGL supports vertical retrace sync, if it isn't supported KiCad just crashes. This is the case for the virtual X server.

The first can be workaround, the second ... I don't think so.
I'm testing that other tools also work with KiCad 6. This is why I updated the iBoM Debian package and added v2.4.1 to the Docker image. Not fully tested yet.

BTW, supporting KiCad 6 was a nightmare, the developers changed a lot in the PCBNew Python API, mostly for cosmetic purpose, and didn't add any compatibility wrapper. They also put in the roadmap the EEschema Python API for months (years) and then decided to let it for 7. So I had to implement the new schematic format, which isn't really well designed ... But again, most stuff should be working or need small adjusts.

closed 2021-12-27 by set-soft

bugadani, 2022-01-12:
Is there a way to try the Kicad 6 support? If I change my github workflow to `uses: INTI-CMNB/KiBot@master` I still get the following error:

```
ERROR:Error loading PCB file. Corrupted? (kibot - kiplot.py:219)
ERROR:KiCad was unable to open this file, as it was created with
a more recent version than the one you are running.
To open it, you'll need  to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 10/14/21
```

set-soft, 2022-01-13:
Hi @bugadani !
I'm adding a new image tagged `dev_k6`. This image will be available using the `v1_k6` tag for the GitHub action. Please let me know if you can use it.

bugadani, 2022-01-13:
Hi @set-soft !

Thanks for the tag! Do I need to do anything in my github action to get the docker image to build? If I update my workflow to `uses: INTI-CMNB/KiBot@v1_k6` it fails with the following:

```
  ...
  Reading package lists...
  E: The value 'buster-backports' is invalid for APT::Default-Release as such a release is not available in the sources
  The command '/bin/sh -c apt-get update  && 	apt-get -y install -t buster-backports make wget curl && 	curl -s https://api.github.com/repos/INTI-CMNB/kicad-git-filters/releases/latest | grep "browser_download_url.*deb" | cut -d : -f 2,3 | tr -d \" | wget -i - && 	apt -y install --no-install-recommends ./*.deb && 	apt-get -y remove curl wget && 	apt-get -y autoremove && 	rm /*.deb && 	rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

I'm using the `ubuntu-latest` github image, should I choose a different one?

set-soft, 2022-01-14:
Hi @bugadani !

My fault, I forgot to remove the Debian Back-ports redference. I removed it and moved the tag. Try now

bugadani, 2022-01-18:
@set-soft sorry for the slow response, I was able to use the new action, thank you very much! One thing that should probably be noted for others, is that if someone ports a project from kicad 5 to 6, they also have to point kibot at the new schematic files, instead of the old ones. Otherwise, like me, they can waste some time :)