INTI-CMNB/KiBot #127: ERROR:Unable to determine kicad2step_do version:

mdeweerd, 2021-12-30:
Hi,

I got
```raw
ERROR:Unable to determine kicad2step_do version:
 (kibot - kiplot.py:102)
```
Using the current kicad_auto:dev image with or without up-to-date kibot scripts:
```
$ docker pull setsoft/kicad_auto:dev
dev: Pulling from setsoft/kicad_auto
Digest: sha256:fb3daefe81dbb2b35bbc17a4442b4a1aaae68fc49135b27bad25cc0b8b3c00cc
Status: Image is up to date for setsoft/kicad_auto:dev
docker.io/setsoft/kicad_auto:dev
```

and same test case as before  [tc5.tar.gz](https://github.com/INTI-CMNB/KiBot/files/7790488/tc5.tar.gz) .  Also available at https://github.com/mdeweerd/kibot-testcases .

```
DEBUG:Expanding `/root/workdir/%bf_%br_%bd_%g/./STEP` in PCB context for <kibot.out_step.STEPOptions object at 0x7faf37010358> parent: 'Exports the PCB as a 3D model.' (step) [step] (kibot - optionable.py:270)
DEBUG:Expanded `/root/workdir/tc5_01_20211224__header/./STEP` (kibot - optionable.py:314)
DEBUG:Output destination: /root/workdir/tc5_01_20211224__header/STEP (kibot - kiplot.py:328)
DEBUG:Expanding `%f_%r_%d-%i%I.%x` in PCB context for <kibot.out_step.STEPOptions object at 0x7faf37010358> parent: 'Exports the PCB as a 3D model.' (step) [step] (kibot - optionable.py:270)
DEBUG:Expanded `tc5_01_20211224-3D.step` (kibot - optionable.py:314)
DEBUG:Filters reset (kibot - fil_base.py:161)
DEBUG:Applying transform filter `_var_rename_kicost,_kicost_rename` (kibot - fil_base.py:140)
DEBUG:Applying filter `_kicost_dnp` to fitted (kibot - fil_base.py:171)
DEBUG:Applying KiCost style variant `header` (kibot - var_kicost.py:58)
DEBUG:J1001 True True (kibot - var_kicost.py:66)
DEBUG:QR1 True True (kibot - var_kicost.py:66)
DEBUG:R1001 True True (kibot - var_kicost.py:66)
DEBUG:ref: R1001 value: 10k -> False (kibot - var_kicost.py:81)
DEBUG:TP1001 True True (kibot - var_kicost.py:66)
DEBUG:TP1002 True True (kibot - var_kicost.py:66)
DEBUG:TP1003 True True (kibot - var_kicost.py:66)
DEBUG:TP1004 True True (kibot - var_kicost.py:66)
DEBUG:TP1005 True True (kibot - var_kicost.py:66)
DEBUG:U1001 True True (kibot - var_kicost.py:66)
DEBUG:ref: U1001 value: MCP6001-OT -> False (kibot - var_kicost.py:81)
DEBUG:Running: ['kicad2step_do', '--version'] (kibot - kiplot.py:95)
ERROR:Unable to determine kicad2step_do version:
 (kibot - kiplot.py:102)
```
[tc5_header_build.log](https://github.com/INTI-CMNB/KiBot/files/7793795/tc5_header_build.log)

set-soft, 2021-12-30:
Try again. I pulled the last one and `kicad2step_do` is there.
`docker images` says:

```
REPOSITORY                TAG                          IMAGE ID       CREATED         SIZE
setsoft/kicad_auto        dev                          6e23adb95eac   3 hours ago     982MB
```

Also make sure you don't have a KiAuto installed in your HOME and you are mounting it in the docker image.

mdeweerd, 2021-12-30:
OK, it had to do with my setup - I thought I excluded the kiauto replacement but I didn't do that properly.

Interactively I gott:
```
root@ce938c0487bb:~/workdir# kicad2step_do --version
Traceback (most recent call last):
  File "/usr/bin/kicad2step_do", line 32, in <module>
    from kiauto.ui_automation import (PopenContext, xdotool, wait_not_focused, wait_for_window, recorded_xvfb,
ImportError: cannot import name 'clipboard_retrieve' from 'kiauto.ui_automation' (/usr/lib/python3/dist-packages/kiauto/ui_automation.py)
```

closed 2021-12-30 by mdeweerd

set-soft, 2021-12-31:
Also note that `xclip` is needed again. We need it for KiCad 6, to read the `kicad2step` messages.