INTI-CMNB/KiBot #128: KiBot Crash under Python 3.10.1

brathis, 2022-01-03:
Hi

After upgrading to **Python 3.10.1**, we can no longer use KiBot:

```
$ kibot
Traceback (most recent call last):
  File "/home/user/.local/bin/kibot", line 16, in <module>
    main()
  File "/home/user/.local/lib/python3.10/site-packages/kibot/__main__.py", line 300, in main
    load_actions()
  File "/home/user/.local/lib/python3.10/site-packages/kibot/kiplot.py", line 72, in load_actions
    _load_actions(os.path.abspath(os.path.dirname(__file__)), True)
  File "/home/user/.local/lib/python3.10/site-packages/kibot/kiplot.py", line 61, in _load_actions
    _import(name, p)
  File "/home/user/.local/lib/python3.10/site-packages/kibot/kiplot.py", line 44, in _import
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 879, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1017, in get_code
  File "/home/user/.local/lib/python3.10/site-packages/kibot/mcpyrate/importer.py", line 28, in source_to_xcode
    return compiler.compile(data, filename=path, self_module=self.name)
  File "/home/user/.local/lib/python3.10/site-packages/kibot/mcpyrate/compiler.py", line 221, in compile
    code, _ignored_docstring = _compile(source, filename, optimize, self_module)
  File "/home/user/.local/lib/python3.10/site-packages/kibot/mcpyrate/compiler.py", line 274, in _compile
    code = builtins.compile(expansion, filename, mode="exec", dont_inherit=True, optimize=optimize)
TypeError: required field "lineno" missing from alias
```
KiBot was installed using
```
pip install --no-compile kibot
```
However, we observe the same behavior when trying to run the latest version from the master branch.
The error appears to stem from `mcpyrate` trying to parse the file `out_position.py`. On line 18, there is an `import` statement which results in an `ast.Import` instance that has no `lineno` attribute, which causes the crash. I unfortunately do not have enough knowledge about compilers and syntax trees to further troubleshoot this issue. Has anyone else experienced this crash?

Thanks

liborw, 2022-01-03:
Same here,  Archlinux installed from aur (kibot-git), Python 3.10.1

set-soft, 2022-01-03:
I can't look it right now, I'm in the middle of a complex addition.
Things to try: take a look at mcpyrate repo to see if a new version supports Python 3.10 and copy it to kibot/mcpyrate

brathis, 2022-01-04:
Unfortunately, the most recent version of mcpyrate appears to be incompatible with Python 3.10, attempting to run the test results in exactly the same error.

brathis, 2022-01-05:
Here's what I've found out so far:

- In [this commit](https://github.com/python/cpython/commit/75a06f067bd0a2687312e5f8e78f9075be76ad3a#diff-0258ad568d1bc0a5fa1c949351df0cc72a0a63441a1ed78c77d59f5695704857), additional attributes were added to the Python AST ABI.
- The error originates in `expander.py` in mcpyrate (line 588 in mcpyrate included with KiBot), where an `alias` instance is created without the `lineno` and `col_offset` attributes that are now required in Python 3.10.
- Replacing the statement with the following snippet makes the mcpyrate tests pass (in the latest version of the upstream mcpyrate):
  ```
  tree.body[index] = copy_location(Import(names=[
                        alias(name=module_absname,
                              asname=None,
                              lineno=getattr(statement, 'lineno', 0),
                              col_offset=getattr(statement, 'col_offset', 0))]),
                        statement)
  ```
- KiBot still does not work with this modification, unfortunately. Instead, we now get the following error:
  ```
  ERROR:Unable to import plug-ins: cannot import name 'macros' from 'kibot.macros' (/home/user/KiBot/kibot/macros.py) (kibot - kiplot.py:50)
  ```

brathis, 2022-01-05:
I've found a fix:

- Update to most recent version of mcpyrate
- Fix call to `alias` constructor in `mcpyrate/expander.py`
- Fix call to `alias` constructor in `macros.py`

There is a (somewhat dirty) implementation of this on my fork: https://github.com/brathis/KiBot/tree/hotfix-python-3.10

set-soft, 2022-01-06:
Hi @brathis !
The a5052b49e4a99babcad8f55c7aa73ec140c44c28 commit includes your patch.
I tried it using Python 3.9 and it seems to be ok (no errors about unknown extra options).
Can you confirm that this fixes the Python 3.10 issues?

brathis, 2022-01-06:
Hey,
At least for the tasks that we've been using KiBot for, this restores all of the functionality under Python 3.10.
Thanks for addressing this so swiftly!

set-soft, 2022-01-06:
Thanks @brathis for your patch!

closed 2022-01-06 by set-soft

Technologicat, 2022-01-10:
Author of `mcpyrate` here. Thanks @brathis and @set-soft for the investigation!

I'll look into adding Python 3.10 support in the next release.

Technologicat, 2022-01-10:
As of [1e57a77](https://github.com/Technologicat/mcpyrate/commit/1e57a77cf72a52ee87bcb45c76c756463de9ccc1), `mcpyrate` should now support Python 3.10.

set-soft, 2022-01-10:
Hi @Technologicat and @brathis !
I don't have 3.10 so @brathis should try it.

brathis, 2022-01-10:
I'll take a look at it tomorrow 👍🏻

brathis, 2022-01-11:
I've tested kibot@master with mcpyrate@master. No issues whatsoever, macro expansion seems to work just fine :)

Technologicat, 2022-01-11:
Thank you for testing! This closes https://github.com/Technologicat/mcpyrate/issues/29.

I have one more bug to fix before the next `mcpyrate` release. In the meantime, there's mcpyrate@master :)

Technologicat, 2022-01-28:
`mcpyrate` 3.6.0 released and available on PyPI. Enjoy!