INTI-CMNB/KiBot #130: Kicad 6 ERC Error

Tcub, 2022-01-04:
Hi,

Just installed Kicad v6 and I thought I would also take KiBot for a spin at the same time but I get an error when running ERC using the example template:

```
tony@nworkstation:~/HW/newTest$ kibot -b newTest.kicad_pcb --verbose
DEBUG:Detected KiCad v6.0.0 (6.0.0-d3dd2cf0fa~116~ubuntu20.04.1 6000000) (kibot - __main__.py:237)
DEBUG:Importing from /usr/lib/python3/dist-packages/kibot (kibot.kibot.kiplot - kiplot.py:53)
DEBUG:- Importing out_hpgl (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_pdf_sch_print (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_dxf (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_gerb_drill (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_step (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_svg (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_kibom (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_svg_sch_print (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_any_layer (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_pdf (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_bom (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_ps (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_ibom (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_any_drill (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_gerber (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_kicost (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_base (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_pdf_pcb_print (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_position (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_pcbdraw (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_sch_variant (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_compress (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_excellon (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing out_boardview (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_ignore_unconnected (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_filters (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_check_zone_fills (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_base (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_erc_warnings (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_update_xml (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_erc (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing pre_drc (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing var_kicost (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing var_ibom (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing var_base (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing var_kibom (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_rot_footprint (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_field_rename (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_base (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_var_rename (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_var_rename_kicost (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_subparts (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing fil_generic (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:- Importing globals (kibot.kibot.kiplot - kiplot.py:60)
DEBUG:Deactivating macros (kibot.kibot.kiplot - kiplot.py:83)
Using config file: example_template.kibot.yaml
DEBUG:Using configuration file: `example_template.kibot.yaml` (kibot - __main__.py:164)
DEBUG:Parsing preflight options: {'check_zone_fills': True, 'erc_warnings': False, 'filters': [{'filter': 'Filter description', 'error': '10', 'regex': 'Regular expression to match'}], 'ignore_unconnected': True, 'run_drc': True, 'run_erc': True, 'update_xml': True} (kibot.kibot.config_reader - config_reader.py:147)
DEBUG:Parsing preflight check_zone_fills (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Parsing preflight erc_warnings (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Parsing preflight filters (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Adding DRC/ERC errors filter 'Filter description','10','Regular expression to match' (kibot.kibot.pre_filters - pre_filters.py:66)
DEBUG:Parsing preflight ignore_unconnected (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Parsing preflight run_drc (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Parsing preflight run_erc (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Parsing preflight update_xml (kibot.kibot.config_reader - config_reader.py:155)
DEBUG:Pre-parsing output options for `boardview_example` (boardview) (kibot.kibot.config_reader - config_reader.py:79)
DEBUG:Pre-parsing output options for `bom_example` (bom) (kibot.kibot.config_reader - config_reader.py:79)
DEBUG:Pre-parsing output options for `compress_example` (compress) (kibot.kibot.config_reader - config_reader.py:79)
DEBUG:Pre-parsing output options for `dxf_example` (dxf) (kibot.kibot.config_reader - config_reader.py:79)
DEBUG:Pre-parsing output options for `excellon_example` (excellon) (kibot.kibot.config_reader - config_reader.py:79)
DEBUG:Pre-parsing output options for `gerb_drill_example` (gerb_drill) (kibot.kibot.config_reader - config_reader.py:79)
[... 44 lines cut ...]
DEBUG:Running: ['eeschema_do', '--version'] (kibot.kibot.kiplot - kiplot.py:92)
Traceback (most recent call last):
  File "/usr/bin/kibot", line 16, in <module>
    main()
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 351, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 350, in generate_outputs
    preflight_checks(skip_pre)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 308, in preflight_checks
    BasePreFlight.run_enabled()
  File "/usr/lib/python3/dist-packages/kibot/pre_base.py", line 59, in run_enabled
    v.run()
  File "/usr/lib/python3/dist-packages/kibot/pre_erc.py", line 43, in run
    output = self.get_targets()[0]
  File "/usr/lib/python3/dist-packages/kibot/pre_erc.py", line 36, in get_targets
    name = Optionable.expand_filename_sch(self, out_pattern)
  File "/usr/lib/python3/dist-packages/kibot/optionable.py", line 222, in expand_filename_sch
    GS.load_sch_title_block()
  File "/usr/lib/python3/dist-packages/kibot/gs.py", line 102, in load_sch_title_block
    assert GS.sch is not None
AssertionError

```

Just thought I would ask if there if there is an obvious config issue on my part. (I know it will be something like that I just don't know what)

set-soft, 2022-01-04:
Is this KiBot from the git repo? I think it isn't (line numbers doesn't match).
Support for KiCad 6 still experimental and only in current git code.

Tcub, 2022-01-04:
Ah no it is the v0.11.0 release as I used the .deb file as I was following along with the install instructions.

Okay so miss built on my end then. Apologies

set-soft, 2022-01-04:
No problems, let me know if the git code works for you. Could need adjusts because I use KiCad 6 as a "nightly" variant (No debian package for 6.0.0 yet)

closed 2022-01-04 by set-soft