INTI-CMNB/KiBot #131: bom only generates xlsx files

jnavila, 2022-01-04:
When specifying "csv" and "xlsx" in bom options, I only get the xslx file, no CSV file is generated.

set-soft, 2022-01-04:
Could you post the configuration file?
Note that you can find plenty of examples [here](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples)

jnavila, 2022-01-04:
```yaml
---
kibot:
  version: 1
variants:
  - name: original
    comment: 'Original ACH1 board'
    type: kibom
    variant: "original"
  - name: alt_instrum_amp
    type: kibom
    variant: "alt_instrum_amp"
    file_id: _instrum
  - name: routage
    type: kibom
    variant: ""
    file_id: routage
outputs:
  - name: bom_original
    comment: "BoM for original"
    type: bom
    options:
      columns:
        - field: Row
          name: "#"
        - field: References
          name: Ref
        - field: Quantity Per PCB
          name: Qnty
        - Value
        - Description
        - characteristics
        - manufacturer
        - field: manf#
          name: "Manufacturer #"
        - Equivalent
        - Critical
        - Notes
      output: '%f_%i_original_rev_%r.%x'
      ref_separator: ','
      csv:
        hide_pcb_info: true
        hide_stats_info: true
        quote_all: true
      no_conflict: ["cfc", "ti", "config"]
      variant: original
      xlsx:
        # [boolean=true] Use colors to show the field type
        col_colors: true
        # [string=''] Column with links to the datasheet
        datasheet_as_link: ''
        # [string|list(string)=''] Column/s containing Digi-Key part numbers, will be linked to web page
        digikey_link: ''
        # [boolean=true] Generate a separated section for DNF (Do Not Fit) components
        generate_dnf: true
        # [boolean=false] Hide project information
        hide_pcb_info: false
        # [boolean=false] Hide statistics information
        hide_stats_info: false
        # [boolean=true] Use a color for empty cells. Applies only when `col_colors` is `true`
        highlight_empty: true
        # [boolean=false] Enable KiCost worksheet creation
        kicost: false
        # [string|boolean=''] PNG file to use as logo, use false to remove
        logo: 'Images/Scantech.png'
        # [number=60] [20,999] Maximum column width (characters)
        max_col_width: 60
        # [string='modern-blue'] Head style: modern-blue, modern-green, modern-red and classic
        style: 'modern-blue'
        # [string='KiBot Bill of Materials'] BoM title
[... 88 lines cut ...]
        digikey_link: ''
        # [boolean=true] Generate a separated section for DNF (Do Not Fit) components
        generate_dnf: true
        # [boolean=false] Hide project information
        hide_pcb_info: false
        # [boolean=false] Hide statistics information
        hide_stats_info: false
        # [boolean=true] Use a color for empty cells. Applies only when `col_colors` is `true`
        highlight_empty: true
        # [boolean=false] Enable KiCost worksheet creation
        kicost: false
        # [string|boolean=''] PNG file to use as logo, use false to remove
        logo: 'Images/Scantech.png'
        # [number=60] [20,999] Maximum column width (characters)
        max_col_width: 60
        # [string='modern-blue'] Head style: modern-blue, modern-green, modern-red and classic
        style: 'modern-blue'
        title: 'ACH1 CFC Alternative'
  - name: 'alt_instrum'
    comment: "Alternate instrum"
    type: sch_variant
    dir: alt_instrum
    options:
      variant: alt_instrum_amp
```

set-soft, 2022-01-04:
You must split the CSV and XLSX outputs.
To avoid repetition use YAML anchors, here is an [example](https://github.com/INTI-CMNB/KiBot/blob/master/tests/yaml_samples/int_bom_subparts_1.kibot.yaml)

closed 2022-01-04 by set-soft