INTI-CMNB/KiBot #132: sch_replace and pcb_replace throw an error?

Faeranne, 2022-01-05:
Attempting to use sch_replace or pcb_replace throws the error `ERROR:Unknown preflight: 'pcb_replace'`.
I'm using version `0.11.0` of KiBot, and `6.0.0` of KiCAD and installed KiBot using pip.
I've tested by using the contents from `tests/yaml_samples/pcb_replace_1.kibot.yaml` as the only thing in my *.kibot.yaml, and can be replicated by simply using a blank kicad project.

set-soft, 2022-01-06:
Hi @Faeranne !
These are new options, also KiCad 6 support, you must install the code from GitHub. Clone the repo, change the directory to it and run `pip install -e .` The `-e` option will allow you to update the repo and get the new code usable without installing again.

closed 2022-01-06 by set-soft

ZOUDIO, 2022-03-11:
@set-soft I am having some difficulty with getting pcb_replace to work on the setsoft/kicad_auto:dev_k6 container. Is there something I am doing wrong?

```
jobs:
  generate-artifacts:
    runs-on: ubuntu-latest
    container: setsoft/kicad_auto:dev_k6
      
    steps:
    - name: Update system repositories and install required libraries
      run: |
        apt update
        apt -y install git zip
        pip install -e .

    - name: Checkout Repository
      uses: actions/checkout@v2     

    - name: Run KiBot
      run: |
        cd pnp/pcb/mobo
        rm -rf export/
        kibot -c ../../../config-4layer.kibot.yaml -e mobo.kicad_sch -b mobo.kicad_pcb -d export
```

I get "pip: not found"

set-soft, 2022-03-12:
Hi @ZOUDIO !
The `dev_k6` image doesn't need to install KiBot, is already installed. The image does the following:

```
	curl https://codeload.github.com/INTI-CMNB/KiBot/zip/refs/heads/master --output pp.zip && \
	unzip pp.zip && \
	pip3 install --no-compile KiBot-master/ && \
```

So it always contains the last code that passed all regression tests.

Oh!, BTW the correct command is `pip3`

ZOUDIO, 2022-03-15:
Just found the new 'set_text_variables'. That is even better, works like a treat! Thanks for your continued efforts @set-soft