INTI-CMNB/KiBot #135: Export step - Unable to expand `KICAD6_3DMODEL_DIR` in `${KICAD6_3DMODEL_DIR}/...`

bzgec, 2022-01-07:
When running `step` I get the following error:
```
- '3D model in STEP format' (3D_STEP) [step]
WARNING:(W010) Unable to find KiCad libraries (kibot - config.py:237)
WARNING:(W010) Unable to find KiCad 3D modules (kibot - config.py:252)
ERROR:Unable to expand `KICAD6_SYMBOL_DIR` in `${KICAD6_SYMBOL_DIR}/4xxx.kicad_sym` (kibot - config.py:64)
ERROR:Unable to expand `KICAD6_SYMBOL_DIR` in `${KICAD6_SYMBOL_DIR}/4xxx_IEEE.kicad_sym` (kibot - config.py:64)
ERROR:Unable to expand `KICAD6_SYMBOL_DIR` in `${KICAD6_SYMBOL_DIR}/74xGxx.kicad_sym` (kibot - config.py:64)
ERROR:Unable to expand `KICAD6_SYMBOL_DIR` in `${KICAD6_SYMBOL_DIR}/74xx.kicad_sym` (kibot - config.py:64)
ERROR:Unable to expand `KICAD6_SYMBOL_DIR` in `${KICAD6_SYMBOL_DIR}/74xx_IEEE.kicad_sym` (kibot - config.py:64)
...
```

Am I missing something?

You can try to recreate the problem from my [repo/commit](https://github.com/bzgec/esp32_display/commit/04b78afc3025f6fd021f7977da7de7473db49a5d).
Running Arch Linux with `kibot-git` package.

KiCad version: 6.0.0
KiBot version: 0.11.0
Python version: 3.10.1

set-soft, 2022-01-10:
May be the 3D models.
I see KiCad 6 is using some "magic definitions", I'm adding them, but I need much more information. (See 256876a1bd7888594a651ee29157f8406f98bc46 the patch should solve the `KICAD6_SYMBOL_DIR` issue)
You must provide the output with debug, otherwise is hard to guess what's going on.
Where is defined `MYLIB`?, I see various references in your project, but no definition.

bzgec, 2022-01-11:
[256876a](https://github.com/INTI-CMNB/KiBot/commit/256876a1bd7888594a651ee29157f8406f98bc46) solved `KICAD6_SYMBOL_DIR` issue.

`MYLIB` is included as a submodule into the project, in KiCad it is an absolute path. 

[Here](https://pastebin.com/tYBXVL0M) is `kibot -vvv` output (looks like `MYLIB` is found):
```
...
- '3D model in STEP format' (3D_STEP) [step]
DEBUG:Output destination: /home/bzgec/Projects/homeAutomation/esp32_display/HW/Outputs/v0.1 (kibot - kiplot.py:330)
DEBUG:Running: ['kicad2step_do', '--version'] (kibot - kiplot.py:96)
DEBUG:Reading KiCad config from `/home/bzgec/.config/kicad/6.0/kicad_common.json` (kibot - config.py:251)
DEBUG:- KiCad var: MYLIB="/home/bzgec/Projects/homeAutomation/esp32_display/HW/kicadLib" (kibot - config.py:242)
DEBUG:Detected KICAD_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:264)
DEBUG:Detected KICAD_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:274)
DEBUG:Exporting KISYS3DMOD="/usr/share/kicad/3dmodels" (kibot - config.py:289)
DEBUG:Exporting KICAD6_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:300)
DEBUG:Exporting KICAD6_3DMODEL_DIR="/usr/share/kicad/3dmodels" (kibot - config.py:308)
DEBUG:Exporting KICAD6_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:316)
DEBUG:Exporting KICAD6_FOOTPRINT_DIR="/usr/share/kicad/footprints" (kibot - config.py:324)
DEBUG:Loading symbols lib table `/home/bzgec/.config/kicad/6.0/sym-lib-table` (kibot - config.py:330)
DEBUG:- Adding lib alias `4xxx` -> `/usr/share/kicad/symbols/4xxx.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `4xxx_IEEE` -> `/usr/share/kicad/symbols/4xxx_IEEE.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `74xGxx` -> `/usr/share/kicad/symbols/74xGxx.kicad_sym` (kibot - config.py:342)

...

DEBUG:- Adding lib alias `Transistor_FET` -> `/usr/share/kicad/symbols/Transistor_FET.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `Transistor_IGBT` -> `/usr/share/kicad/symbols/Transistor_IGBT.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `Triac_Thyristor` -> `/usr/share/kicad/symbols/Triac_Thyristor.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `Valve` -> `/usr/share/kicad/symbols/Valve.kicad_sym` (kibot - config.py:342)
DEBUG:- Adding lib alias `Video` -> `/usr/share/kicad/symbols/Video.kicad_sym` (kibot - config.py:342)
DEBUG:Loading symbols lib table `/home/bzgec/Projects/homeAutomation/esp32_display/HW/Board/v0.1/sym-lib-table` (kibot - config.py:330)
DEBUG:- Adding lib alias `myLib` -> `/home/bzgec/Projects/homeAutomation/esp32_display/HW/kicadLib/myLib.kicad_sym` (kibot - config.py:342)
DEBUG:Executing: ['kicad2step_do', '-o', '/home/bzgec/Projects/homeAutomation/esp32_display/HW/Outputs/v0.1/esp32_display.step', '-f', '-vv', '--grid-origin', '/home/bzgec/Projects/homeAutomation/esp32_display/HW/Board/v0.1/esp32_display.kicad_pcb'] (kibot - out_step.py:80)
ERROR:Failed to create Step file, error 1 (kibot - out_step.py:84)
DEBUG:Output from command: DEBUG:Detected KiCad v6.0.0 (6.0.0 6000000) (kicad2step_do - misc.py:138)
DEBUG:Config path /home/bzgec/.config/kicad/6.0 (kicad2step_do - misc.py:153)
DEBUG:KiCad environment: {'MYLIB': '/home/bzgec/Projects/homeAutomation/esp32_display/HW/kicadLib'} (kicad2step_do - misc.py:233)
INFO:KiCad to STEP wrapper
DEBUG:Command: ['kicad2step', '-o', '/home/bzgec/Projects/homeAutomation/esp32_display/HW/Outputs/v0.1/esp32_display.step', '-f', '--grid-origin', '/home/bzgec/Projects/homeAutomation/esp32_display/HW/Board/v0.1/esp32_display.kicad_pcb'] (kicad2step_do - kicad2step_do:133)
DEBUG:Waiting for virtual X server ... (kicad2step_do.kiauto.ui_automation - ui_automation.py:87)
DEBUG:Current DISPLAY is :1787452035 (kicad2step_do.kiauto.ui_automation - ui_automation.py:89)
DEBUG:Checking using ['setxkbmap', '-query'] (kicad2step_do.kiauto.ui_automation - ui_automation.py:100)
INFO:Waiting for "KiCad to STEP" ...
DEBUG:get_windows 1293  (kicad2step_do.kiauto.ui_automation - ui_automation.py:272)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:274)
DEBUG:get_windows 1293  (kicad2step_do.kiauto.ui_automation - ui_automation.py:272)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:274)
DEBUG:Closing pipe with 3918 (kicad2step_do.kiauto.ui_automation - ui_automation.py:45)
DEBUG:Terminating 3918 (kicad2step_do.kiauto.ui_automation - ui_automation.py:57)
Traceback (most recent call last):
  File "/usr/bin/kicad2step_do", line 140, in <module>
    res = run_gui_version(cmd, cfg, flog_out, flog_err)
  File "/usr/bin/kicad2step_do", line 41, in run_gui_version
    wait_for_window('KiCad to STEP', 'Kicad2step', popen_obj=proc)
  File "/usr/lib/python3.10/site-packages/kiauto/ui_automation.py", line 396, in wait_for_window
    raise CalledProcessError(ret_code, 'KiCad')
subprocess.CalledProcessError: Command 'KiCad' returned non-zero exit status 127.
 (kibot - out_step.py:86)
```

set-soft, 2022-01-11:
> `MYLIB` is included as a submodule into the project, in KiCad it is an absolute path.
> 
> [Here](https://pastebin.com/tYBXVL0M) is `kibot -vvv` output (looks like `MYLIB` is found):

Ok! This variable is defined in your `kicad_common.json` (KiCad general config). But note that this won't work for other people or for GitHub Actions. I suggest avoiding it and using `sym-lib-table` instead, with a relative path.

Now about the STEP fail: Can you try to manually run:

```
kicad2step -o  /home/bzgec/Projects/homeAutomation/esp32_display/HW/Outputs/v0.1/esp32_display.step  -f --grid-origin /home/bzgec/Projects/homeAutomation/esp32_display/HW/Board/v0.1/esp32_display.kicad_pcb
```

According to the log this command exited with error code 127, looks like a crash or fatal error.

bzgec, 2022-01-11:
> Ok! This variable is defined in your kicad_common.json (KiCad general config). But note that this won't work for other people or for GitHub Actions. I suggest avoiding it and using sym-lib-table instead, with a relative path.

Thanks for the idea!

> Now about the STEP fail: Can you try to manually run:

```
$ kicad2step -o  /home/bzgec/Projects/homeAutomation/esp32_display/HW/Outputs/v0.1/esp32_display.step  -f --grid-origin /home/bzgec/Projects/homeAutomation/esp32_display/HW/Board/v0.1/esp32_display.kicad_pcb
------------
kicad2step: symbol lookup error: kicad2step: undefined symbol: _ZN24BRepBuilderAPI_MakeShape5BuildEv
```

set-soft, 2022-01-11:
BTW, the KiCad 6 I have is installed as a nightly package and reports:

```
$ ldd /usr/lib/kicad-nightly/bin/kicad2step
	linux-vdso.so.1 (0x00007fff47be8000)
	libTKBO.so.7 => /lib/x86_64-linux-gnu/libTKBO.so.7 (0x00007fdd3870c000)
	libTKBRep.so.7 => /lib/x86_64-linux-gnu/libTKBRep.so.7 (0x00007fdd385f7000)
	libTKCDF.so.7 => /lib/x86_64-linux-gnu/libTKCDF.so.7 (0x00007fdd3859c000)
	libTKernel.so.7 => /lib/x86_64-linux-gnu/libTKernel.so.7 (0x00007fdd383b8000)
	libTKG3d.so.7 => /lib/x86_64-linux-gnu/libTKG3d.so.7 (0x00007fdd38270000)
	libTKIGES.so.7 => /lib/x86_64-linux-gnu/libTKIGES.so.7 (0x00007fdd37e3c000)
	libTKLCAF.so.7 => /lib/x86_64-linux-gnu/libTKLCAF.so.7 (0x00007fdd37d3f000)
	libTKMath.so.7 => /lib/x86_64-linux-gnu/libTKMath.so.7 (0x00007fdd37a95000)
	libTKPrim.so.7 => /lib/x86_64-linux-gnu/libTKPrim.so.7 (0x00007fdd37a21000)
	libTKSTEP.so.7 => /lib/x86_64-linux-gnu/libTKSTEP.so.7 (0x00007fdd376cf000)
	libTKTopAlgo.so.7 => /lib/x86_64-linux-gnu/libTKTopAlgo.so.7 (0x00007fdd373b8000)
	libTKXCAF.so.7 => /lib/x86_64-linux-gnu/libTKXCAF.so.7 (0x00007fdd372ce000)
	libTKXDEIGES.so.7 => /lib/x86_64-linux-gnu/libTKXDEIGES.so.7 (0x00007fdd372b1000)
	libTKXDESTEP.so.7 => /lib/x86_64-linux-gnu/libTKXDESTEP.so.7 (0x00007fdd371e6000)
	libTKXSBase.so.7 => /lib/x86_64-linux-gnu/libTKXSBase.so.7 (0x00007fdd36f1c000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fdd36eff000)
	libwx_gtk3u_core-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0 (0x00007fdd36875000)
	libwx_baseu-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0 (0x00007fdd365b1000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fdd363e2000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fdd3629e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fdd36284000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdd36262000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdd3609d000)
	libTKShHealing.so.7 => /lib/x86_64-linux-gnu/libTKShHealing.so.7 (0x00007fdd35ddb000)
	libTKGeomAlgo.so.7 => /lib/x86_64-linux-gnu/libTKGeomAlgo.so.7 (0x00007fdd357ea000)
	libTKGeomBase.so.7 => /lib/x86_64-linux-gnu/libTKGeomBase.so.7 (0x00007fdd35294000)
	libTKG2d.so.7 => /lib/x86_64-linux-gnu/libTKG2d.so.7 (0x00007fdd35229000)
	libtbb.so.2 => /lib/x86_64-linux-gnu/libtbb.so.2 (0x00007fdd351e4000)
	libtbbmalloc.so.2 => /lib/x86_64-linux-gnu/libtbbmalloc.so.2 (0x00007fdd351a1000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fdd35199000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdd38a75000)
	libTKBool.so.7 => /lib/x86_64-linux-gnu/libTKBool.so.7 (0x00007fdd34d4e000)
	libTKSTEPAttr.so.7 => /lib/x86_64-linux-gnu/libTKSTEPAttr.so.7 (0x00007fdd34c33000)
	libTKSTEP209.so.7 => /lib/x86_64-linux-gnu/libTKSTEP209.so.7 (0x00007fdd34b83000)
	libTKSTEPBase.so.7 => /lib/x86_64-linux-gnu/libTKSTEPBase.so.7 (0x00007fdd348e0000)
	libTKVCAF.so.7 => /lib/x86_64-linux-gnu/libTKVCAF.so.7 (0x00007fdd3489c000)
	libTKV3d.so.7 => /lib/x86_64-linux-gnu/libTKV3d.so.7 (0x00007fdd34514000)
	libTKService.so.7 => /lib/x86_64-linux-gnu/libTKService.so.7 (0x00007fdd343de000)
	libTKCAF.so.7 => /lib/x86_64-linux-gnu/libTKCAF.so.7 (0x00007fdd34337000)
	libgtk-3.so.0 => /lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007fdd33b85000)
	libgdk-3.so.0 => /lib/x86_64-linux-gnu/libgdk-3.so.0 (0x00007fdd33a7e000)
	libpangocairo-1.0.so.0 => /lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007fdd33a6d000)
	libpango-1.0.so.0 => /lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007fdd33a1b000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007fdd338f6000)
	libgdk_pixbuf-2.0.so.0 => /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007fdd338d0000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fdd33876000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fdd33745000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fdd33602000)
	libSM.so.6 => /lib/x86_64-linux-gnu/libSM.so.6 (0x00007fdd335f7000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fdd335bd000)
	libjpeg.so.62 => /lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007fdd33539000)
	libtiff.so.5 => /lib/x86_64-linux-gnu/libtiff.so.5 (0x00007fdd334b3000)
	libTKMesh.so.7 => /lib/x86_64-linux-gnu/libTKMesh.so.7 (0x00007fdd3339e000)
	libTKHLR.so.7 => /lib/x86_64-linux-gnu/libTKHLR.so.7 (0x00007fdd33252000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fdd331cb000)
	libfontconfig.so.1 => /lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007fdd33185000)
	libfreeimage.so.3 => /lib/x86_64-linux-gnu/libfreeimage.so.3 (0x00007fdd330c8000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fdd33005000)
	libgmodule-2.0.so.0 => /lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007fdd32fff000)
	libXi.so.6 => /lib/x86_64-linux-gnu/libXi.so.6 (0x00007fdd32fed000)
	libXfixes.so.3 => /lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fdd32fe5000)
	libcairo-gobject.so.2 => /lib/x86_64-linux-gnu/libcairo-gobject.so.2 (0x00007fdd32fd7000)
	libatk-1.0.so.0 => /lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007fdd32fad000)
	libatk-bridge-2.0.so.0 => /lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0 (0x00007fdd32f76000)
	libepoxy.so.0 => /lib/x86_64-linux-gnu/libepoxy.so.0 (0x00007fdd32e47000)
[... 40 lines cut ...]
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007fdd318fa000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fdd318a3000)
[... cut ...]
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fdd31198000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007fdd31147000)
	libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007fdd310af000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007fdd310a2000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fdd3107d000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fdd30f5d000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fdd30f37000)
```

bzgec, 2022-01-11:
Hmmm KiCad 5 should not be installed.

Here is my output:
```
ldd `which kicad2step`
	linux-vdso.so.1 (0x00007fff02908000)
	libTKBO.so.7 => /usr/lib/libTKBO.so.7 (0x00007fb6ddf21000)
	libTKBRep.so.7 => /usr/lib/libTKBRep.so.7 (0x00007fb6dde2e000)
	libTKCDF.so.7 => /usr/lib/libTKCDF.so.7 (0x00007fb6dddd7000)
	libTKernel.so.7 => /usr/lib/libTKernel.so.7 (0x00007fb6ddbfa000)
	libTKG3d.so.7 => /usr/lib/libTKG3d.so.7 (0x00007fb6ddad5000)
	libTKIGES.so.7 => /usr/lib/libTKIGES.so.7 (0x00007fb6dd756000)
	libTKLCAF.so.7 => /usr/lib/libTKLCAF.so.7 (0x00007fb6dd680000)
	libTKMath.so.7 => /usr/lib/libTKMath.so.7 (0x00007fb6dd3f3000)
	libTKPrim.so.7 => /usr/lib/libTKPrim.so.7 (0x00007fb6dd399000)
	libTKSTEP.so.7 => /usr/lib/libTKSTEP.so.7 (0x00007fb6dd091000)
	libTKTopAlgo.so.7 => /usr/lib/libTKTopAlgo.so.7 (0x00007fb6dce2b000)
	libTKXCAF.so.7 => /usr/lib/libTKXCAF.so.7 (0x00007fb6dcd5d000)
	libTKXDEIGES.so.7 => /usr/lib/libTKXDEIGES.so.7 (0x00007fb6dcd40000)
	libTKXDESTEP.so.7 => /usr/lib/libTKXDESTEP.so.7 (0x00007fb6dcc8c000)
	libTKXSBase.so.7 => /usr/lib/libTKXSBase.so.7 (0x00007fb6dca3a000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007fb6dca20000)
	libwx_gtk3u_core-3.0.so.0 => /usr/lib/libwx_gtk3u_core-3.0.so.0 (0x00007fb6dc432000)
	libwx_baseu-3.0.so.0 => /usr/lib/libwx_baseu-3.0.so.0 (0x00007fb6dc206000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007fb6dbfee000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007fb6dbeaa000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007fb6dbe8f000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007fb6dbcc3000)
	libTKShHealing.so.7 => /usr/lib/libTKShHealing.so.7 (0x00007fb6dbaa0000)
	libTKGeomAlgo.so.7 => /usr/lib/libTKGeomAlgo.so.7 (0x00007fb6db5de000)
	libTKGeomBase.so.7 => /usr/lib/libTKGeomBase.so.7 (0x00007fb6db138000)
	libTKG2d.so.7 => /usr/lib/libTKG2d.so.7 (0x00007fb6db0d6000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007fb6db0b5000)
	libtbb.so.12 => /usr/lib/libtbb.so.12 (0x00007fb6db064000)
	libtbbmalloc.so.2 => /usr/lib/libtbbmalloc.so.2 (0x00007fb6db024000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fb6db01b000)
	libTKBool.so.7 => /usr/lib/libTKBool.so.7 (0x00007fb6dad23000)
	libTKSTEPAttr.so.7 => /usr/lib/libTKSTEPAttr.so.7 (0x00007fb6dab96000)
	libTKSTEP209.so.7 => /usr/lib/libTKSTEP209.so.7 (0x00007fb6daafd000)
	libTKSTEPBase.so.7 => /usr/lib/libTKSTEPBase.so.7 (0x00007fb6da8b0000)
	libTKVCAF.so.7 => /usr/lib/libTKVCAF.so.7 (0x00007fb6da879000)
	libTKV3d.so.7 => /usr/lib/libTKV3d.so.7 (0x00007fb6da58e000)
	libTKService.so.7 => /usr/lib/libTKService.so.7 (0x00007fb6da449000)
	libTKCAF.so.7 => /usr/lib/libTKCAF.so.7 (0x00007fb6da3c9000)
	libgtk-3.so.0 => /usr/lib/libgtk-3.so.0 (0x00007fb6d9c1e000)
	libgdk-3.so.0 => /usr/lib/libgdk-3.so.0 (0x00007fb6d9b28000)
	libpangocairo-1.0.so.0 => /usr/lib/libpangocairo-1.0.so.0 (0x00007fb6d9b15000)
	libpango-1.0.so.0 => /usr/lib/libpango-1.0.so.0 (0x00007fb6d9aaf000)
	libcairo.so.2 => /usr/lib/libcairo.so.2 (0x00007fb6d9990000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0 (0x00007fb6d994a000)
	libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0 (0x00007fb6d98ef000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007fb6d97b3000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007fb6d966f000)
	libXxf86vm.so.1 => /usr/lib/libXxf86vm.so.1 (0x00007fb6d9668000)
	libSM.so.6 => /usr/lib/libSM.so.6 (0x00007fb6d965e000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007fb6d9627000)
	libjpeg.so.8 => /usr/lib/libjpeg.so.8 (0x00007fb6d9595000)
	libtiff.so.5 => /usr/lib/libtiff.so.5 (0x00007fb6d9501000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007fb6de1b0000)
	libTKMesh.so.7 => /usr/lib/libTKMesh.so.7 (0x00007fb6d942c000)
	libTKHLR.so.7 => /usr/lib/libTKHLR.so.7 (0x00007fb6d931b000)
	libfontconfig.so.1 => /usr/lib/libfontconfig.so.1 (0x00007fb6d92cc000)
	libfreeimage.so.3 => /usr/lib/libfreeimage.so.3 (0x00007fb6d9209000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007fb6d913c000)
	libavcodec.so.58 => /usr/lib/libavcodec.so.58 (0x00007fb6d7d03000)
	libavformat.so.58 => /usr/lib/libavformat.so.58 (0x00007fb6d7a69000)
	libswscale.so.5 => /usr/lib/libswscale.so.5 (0x00007fb6d79d7000)
	libavutil.so.56 => /usr/lib/libavutil.so.56 (0x00007fb6d772c000)
	libgmodule-2.0.so.0 => /usr/lib/libgmodule-2.0.so.0 (0x00007fb6d7725000)
	libgio-2.0.so.0 => /usr/lib/libgio-2.0.so.0 (0x00007fb6d7559000)
[... 90 lines cut ...]
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007fb6c9901000)
	libjpegxr.so.0 => /usr/lib/libjpegxr.so.0 (0x00007fb6c98bd000)
	libIlmThread-3_1.so.30 => /usr/lib/libIlmThread-3_1.so.30 (0x00007fb6c98b4000)
	libjasper.so.4 => /usr/lib/libjasper.so.4 (0x00007fb6c9850000)
	liblcms2.so.2 => /usr/lib/liblcms2.so.2 (0x00007fb6c97ee000)
	libgomp.so.1 => /usr/lib/libgomp.so.1 (0x00007fb6c97aa000)
	libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x00007fb6c9787000)
	libsoxr.so.0 => /usr/lib/libsoxr.so.0 (0x00007fb6c9705000)
	libogg.so.0 => /usr/lib/libogg.so.0 (0x00007fb6c96f8000)
	libmvec.so.1 => /usr/lib/libmvec.so.1 (0x00007fb6c96cc000)
	libp11-kit.so.0 => /usr/lib/libp11-kit.so.0 (0x00007fb6c9597000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007fb6c9575000)
	libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007fb6c93f3000)
	libtasn1.so.6 => /usr/lib/libtasn1.so.6 (0x00007fb6c93db000)
	libnettle.so.8 => /usr/lib/libnettle.so.8 (0x00007fb6c9394000)
	libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x00007fb6c934b000)
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x00007fb6c906a000)
	libblkid.so.1 => /usr/lib/libblkid.so.1 (0x00007fb6c9032000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007fb6c8f54000)
	libicudata.so.70 => /usr/lib/libicudata.so.70 (0x00007fb6c7338000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007fb6c7315000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007fb6c7309000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007fb6c71cb000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007fb6c71a3000)
```

set-soft, 2022-01-11:
Strange all dependencies seems to be there.
Looking at my binary I find:
```
_ZN24BRepBuilderAPI_MakeShape5ShapeEv
_ZN24BRepBuilderAPI_MakeShapecv12TopoDS_ShapeEv
```
But no 
```
_ZN24BRepBuilderAPI_MakeShape5BuildEv
```
Both members are part of the current [API](https://dev.opencascade.org/doc/refman/html/class_b_rep_builder_a_p_i___make_shape.html)

My KiCad is linked to version 7.5.1 i.e `/lib/x86_64-linux-gnu/libTKTopAlgo.so.7.5.1`

Looks like your KiCad doesn't match with the installed libocct

bzgec, 2022-01-11:
Now I things are getting weird. I fully removed KiCad and deleted all files/folders named `kicad`, and reinstalled everything.
Now my 3D models are missing, can't even do manual export to `.step`..
I am going to come back here when this is fixed.

set-soft, 2022-01-11:
KiCad 3D models are usually in a separated package, in Debian/Ubuntu they are in `kicad-packages3d`

bzgec, 2022-01-12:
The problem was `opencascade` v7.6. For now they have [reverted](https://github.com/archlinux/svntogit-community/commit/1273177dd98ab4a12e46f66c1a36a0229d7da1c0#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a) it back to v7.5 and it works.

closed 2022-01-12 by bzgec