INTI-CMNB/KiBot #138: hidden config files / in hidden folder

nerdyscout, 2022-01-12:
seems like kibot freezes if a config file like `.config.kibot.yaml`, or inside a hidden folter `.kibot/config.kibot.yaml`, is used.

set-soft, 2022-01-13:
Hi @nerdyscout !

I need more information, I tried this:

```
$ src/kibot -c .config.kibot.yaml -b tests/board_samples/kicad_5/bom.kicad_pcb 
- 'Example of board view export' (Board View Test) [boardview]
user@host:~/kibot$ cat .config.kibot.yaml 
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'Board View Test'
    comment: "Example of board view export"
    type: boardview
user@host:~/kibot$ ls -la bom*
-rw-r--r-- 1 user user 289 ene 13 19:55 bom-boardview.brd
```

And worked. Of course it can't detect the file by itself:

```
$ src/kibot  -b tests/board_samples/kicad_5/bom.kicad_pcb 
ERROR:No config file found (*.kibot.yaml), use -c to specify one. (kibot - __main__.py:163)
```

But this seems to be ok since the file is hidden.

nerdyscout, 2022-01-14:
I do use kibot from inside docker
```docker run --volume="${PWD}:${HOME}/workdir:rw" --volume="/opt/kicad/etc/group.docker:/etc/group:ro" --volume="/opt/kicad/etc/passwd.docker:/etc/passwd:ro" --volume="/etc/shadow:/etc/shadow:ro" --volume="${HOME}/.config/kicad:${HOME}/.config/kicad:rw" --volume="${HOME}/.cache/kicad:${HOME}/.cache/kicad:rw" --entrypoint /bin/bash -ti --workdir="${HOME}/workdir" --user $(id -u):$(id -g) setsoft/kicad_auto:dev $@```
inside this docker container I am running `kibot -c .hidden.kibot.yaml -v` which freezes but can be terminated by ctrl+c

```
DEBUG:- Importing fil_rot_footprint (kibot - kiplot.py:64)
DEBUG:- Importing fil_field_rename (kibot - kiplot.py:64)
DEBUG:- Importing fil_generic (kibot - kiplot.py:64)
DEBUG:- Importing globals (kibot - kiplot.py:64)
DEBUG:Deactivating macros (kibot - kiplot.py:87)
DEBUG:Using configuration file: `.hidden.kibot.yaml` (kibot - __main__.py:168)
DEBUG:Parsing preflight options: {'run_erc': True, 'run_drc': False, 'check_zone_fills': False, 'ignore_unconnected': True, 'update_xml': True} (kibot - config_reader.py:165)
DEBUG:Parsing preflight run_erc (kibot - config_reader.py:173)
DEBUG:Parsing preflight run_drc (kibot - config_reader.py:173)
DEBUG:Parsing preflight check_zone_fills (kibot - config_reader.py:173)
DEBUG:Parsing preflight ignore_unconnected (kibot - config_reader.py:173)
DEBUG:Parsing preflight update_xml (kibot - config_reader.py:173)
DEBUG:Parsing global options: {} (kibot - config_reader.py:181)
^CTraceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 363, in main
    GS.set_sch(solve_schematic(args.schematic, args.board_file, plot_config))
  File "/usr/local/lib/python3.7/dist-packages/kibot/__main__.py", line 120, in solve_schematic
    config = os.path.splitext(config)[0]
  File "/usr/lib/python3.7/posixpath.py", line 129, in splitext
    return genericpath._splitext(p, sep, None, extsep)
  File "/usr/lib/python3.7/genericpath.py", line 129, in _splitext
    dotIndex = p.rfind(extsep)
KeyboardInterrupt
```
the config file is running only some erc 
```
kibot:
  version: 1

preflight:
  run_erc: true
  run_drc: false
  check_zone_fills: false
  ignore_unconnected: true
  update_xml: true
```

But I can confirm I have seen one setup where it is working like expected, so far I was not able to figure out the differences.

commit 8676f28f (2022-01-14): Fixed problems when using a hidden config file

- Found when using an output that needs the SCH, not specifying the
  SCH and more than one SCH was found.

Closes #138
closed 2022-01-14 by set-soft