INTI-CMNB/KiBot #140: AttributeError: 'NoneType' object has no attribute 'group'

bzgec, 2022-01-17:
I am not sure if this is `kiauto` or `kibot` problem, but since you are the owner of both (and this is a parent project) I am posting here.

Everything I run I get the `AttributeError: 'NoneType' object has no attribute 'group'` error. 
This was introduced with some update, but sadly I don't know with which update/package.

Running ERC:
```
- Running the ERC
DEBUG:Executing: ['eeschema_do', '-r', '-vvv', 'run_erc', '-o', '/home/bzgec/Projects/VoltLink/Outputs/voltLink-erc.txt', '-f', '/home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter', '/home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch', '/home/bzgec/Projects/VoltLink/Board/../Outputs'] (kibot - kiplot.py:168)
DEBUG:Command line: eeschema_do -r -vvv run_erc -o /home/bzgec/Projects/VoltLink/Outputs/voltLink-erc.txt -f /home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter /home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch /home/bzgec/Projects/VoltLink/Board/../Outputs (kibot - kiplot.py:170)
DEBUG:Failed with error 1, retrying ... (kibot - kiplot.py:177)
DEBUG:Output from command:
> Traceback (most recent call last):
>   File "/usr/bin/eeschema_do", line 542, in <module>
>     cfg = Config(logger, args.schematic, args)
>   File "/usr/lib/python3.10/site-packages/kiauto/misc.py", line 134, in __init__
>     self.kicad_version_major = int(m.group(1))
> AttributeError: 'NoneType' object has no attribute 'group'
>  (kibot - kiplot.py:181)
ERROR:ERC returned 1 (kibot.pre_erc - pre_erc.py:65)
```

Running DRC:
```
- Running the DRC
DEBUG:Executing: ['pcbnew_do', '-r', '-vvv', 'run_drc', '-o', '/home/bzgec/Projects/VoltLink/Outputs/voltLink-drc.txt', '-f', '/home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter', '/home/bzgec/Projects/VoltLink/Board/voltLink.kicad_pcb', '/home/bzgec/Projects/VoltLink/Board/../Outputs'] (kibot - kiplot.py:168)
DEBUG:Command line: pcbnew_do -r -vvv run_drc -o /home/bzgec/Projects/VoltLink/Outputs/voltLink-drc.txt -f /home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter /home/bzgec/Projects/VoltLink/Board/voltLink.kicad_pcb /home/bzgec/Projects/VoltLink/Board/../Outputs (kibot - kiplot.py:170)
DEBUG:Failed with error 1, retrying ... (kibot - kiplot.py:177)
DEBUG:Output from command:
> Traceback (most recent call last):
>   File "/usr/bin/pcbnew_do", line 823, in <module>
>     cfg = Config(logger, args.kicad_pcb_file, args)
>   File "/usr/lib/python3.10/site-packages/kiauto/misc.py", line 134, in __init__
>     self.kicad_version_major = int(m.group(1))
> AttributeError: 'NoneType' object has no attribute 'group'
>  (kibot - kiplot.py:181)
ERROR:DRC returned 1 (kibot.pre_drc - pre_drc.py:63)
```

Running schematics to PDF:
```
- 'Print schematics to PDF' (PDF_SCH) [pdf_sch_print]
DEBUG:Output destination: /home/bzgec/Projects/VoltLink/Outputs (kibot - kiplot.py:330)
DEBUG:Running: ['eeschema_do', '--version'] (kibot - kiplot.py:96)
DEBUG:Executing: ['eeschema_do', '-r', '-vvv', 'export', '--all_pages', '--file_format', 'pdf', '/home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch', '/home/bzgec/Projects/VoltLink/Outputs'] (kibot - kiplot.py:168)
DEBUG:Command line: eeschema_do -r -vvv export --all_pages --file_format pdf /home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch /home/bzgec/Projects/VoltLink/Outputs (kibot - kiplot.py:170)
DEBUG:Failed with error 1, retrying ... (kibot - kiplot.py:177)
DEBUG:Output from command:
> Traceback (most recent call last):
>   File "/usr/bin/eeschema_do", line 542, in <module>
>     cfg = Config(logger, args.schematic, args)
>   File "/usr/lib/python3.10/site-packages/kiauto/misc.py", line 134, in __init__
>     self.kicad_version_major = int(m.group(1))
> AttributeError: 'NoneType' object has no attribute 'group'
>  (kibot - kiplot.py:181)
ERROR:eeschema_do returned 1 (kibot - out_pdf_sch_print.py:73)
```

KiCad version: 6.0.1
KiBot version: 0.11.0
Python version: 3.10.1

set-soft, 2022-01-17:
Try this:
1. Start `pcbnew`
2. Go to `Tools`|`Scripting console` menu option, you'll get the python console
3. Enter the following commands:
```
import pcbnew
pcbnew.GetBuildVersion()
```
Tell me what you get. From what I see you are not getting a version number, not at least one complete version. The code is using the following regex: (\d+)\.(\d+)\.(\d+)

Also post the whole output from KiBot, because KiBot uses the same and isn't failing.

bzgec, 2022-01-17:
> Tell me what you get. From what I see you are not getting a version number, not at least one complete version. The code is using the following regex: (\d+).(\d+).(\d+)

Well I get an error as soon as I open Scripting console.
```
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/usr/share/kicad/scripting/kicad_pyshell/__init__.py", line 253, in makePcbnewShellWindow
    return KiCadPyShell(parent)
  File "/usr/share/kicad/scripting/kicad_pyshell/__init__.py", line 39, in __init__
    KiCadEditorNotebookFrame.__init__(self, parent)
  File "/usr/share/kicad/scripting/kicad_pyshell/kicad_pyeditor.py", line 939, in __init__
    KiCadEditorFrame.__init__(self, parent)
  File "/usr/share/kicad/scripting/kicad_pyshell/kicad_pyeditor.py", line 732, in __init__
    self._setup()
  File "/usr/share/kicad/scripting/kicad_pyshell/__init__.py", line 115, in _setup
    self.crust = crust.Crust(parent=self.notebook,
  File "/usr/lib/python3.10/site-packages/wx/py/crust.py", line 43, in __init__
    self.shell = Shell(parent=self, introText=intro,
  File "/usr/lib/python3.10/site-packages/wx/py/shell.py", line 332, in __init__
    self.Bind(wx.EVT_MENU, lambda evt: self.CopyWithPrompts(), id=frame.ID_COPY_PLUS)
  File "/usr/lib/python3.10/site-packages/wx/core.py", line 1406, in _EvtHandler_Bind
    event.Bind(self, id, id2, handler)
  File "/usr/lib/python3.10/site-packages/wx/core.py", line 1477, in Bind
    target.Connect(id1, id2, et, function)
TypeError: EvtHandler.Connect(): argument 1 has unexpected type 'WindowIDRef'
```
[image]

> Also post the whole output from KiBot, because KiBot uses the same and isn't failing.

[whole output from KiBot](https://pastebin.com/nNyDcaUB)

bzgec, 2022-01-17:
It seems that it is an Arch problem - [link](https://aur.archlinux.org/packages/kicad-nightly/#pinned-807958)
Let me check this.

bzgec, 2022-01-17:
Looks like that Scripting console problem was not just on Arch - [link](https://gitlab.com/kicad/code/kicad/-/issues/10077)

set-soft, 2022-01-17:
This is all strange, KiBot can parse the version because you get:
```
DEBUG:Detected KiCad v6.0.1 ((6.0.1) 6000001) (kibot - __main__.py:257)
```
But KiAuto can't.

The console problem seems to be a wrong build, some missmatch between wxWidgets versions.

I pushed to the repo a change in KiAuto that will print more information when we can't parse the KiCad version. Can you try it?

bzgec, 2022-01-17:
> I pushed to the repo a change in KiAuto that will print more information when we can't parse the KiCad version. Can you try it?

```
- Running the ERC
DEBUG:Executing: ['eeschema_do', '-r', '-vvv', 'run_erc', '-o', '/home/bzgec/Projects/VoltLink/Outputs/voltLink-erc.txt', '-f', '/home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter', '/home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch', '/home/bzgec/Projects/VoltLink/Board/../Outputs'] (kibot - kiplot.py:168)
DEBUG:Command line: eeschema_do -r -vvv run_erc -o /home/bzgec/Projects/VoltLink/Outputs/voltLink-erc.txt -f /home/bzgec/Projects/VoltLink/Outputs/kibot_errors.filter /home/bzgec/Projects/VoltLink/Board/voltLink.kicad_sch /home/bzgec/Projects/VoltLink/Board/../Outputs (kibot - kiplot.py:170)
DEBUG:Failed with error 11, retrying ... (kibot - kiplot.py:177)
ERROR:Unable to detect KiCad version, got: `(6.0.1)` (eeschema_do - misc.py:135) (kibot - kiplot.py:149)
DEBUG:Output from command:
> ERROR:Unable to detect KiCad version, got: `(6.0.1)` (eeschema_do - misc.py:135)
>  (kibot - kiplot.py:181)
ERROR:ERC returned 11 (kibot.pre_erc - pre_erc.py:65)
```

set-soft, 2022-01-17:
I see what's the problem, thanks!
I pushed a modification that should fix the problem, try now.

bzgec, 2022-01-18:
Now everything works except ERC - [link](https://pastebin.com/B2T0jN8R).
Should I open a new issue?

set-soft, 2022-01-18:
According to the log the `import` command isn't installed. This is part of ImageMagick. 
Most probably it changed its name, will look it in the next days

set-soft, 2022-01-18:
Do you have `imagemagick` on your system?

bzgec, 2022-01-18:
You were correct, I was missing `imagemagick`.

Now it is working, thanks!

closed 2022-01-18 by bzgec