INTI-CMNB/KiBot #141: `pdf_pcb_print` output format times out

bugadani, 2022-01-18:
I have a job that generates an assembly diagram based on the fabrication layers. This works well enough when I use the `pcb` output type, but fails with the following using `pdf_pcb_print`:

```
- '' (Assembly diagram) [pdf_pcb_print]
WARNING:(W058) Missing KiCad main config file /github/home/.config/kicad/6.0/kicad_common.json (pcbnew_do - misc.py:172) (kibot - kiplot.py:152)
ERROR:Time-out waiting for pcbnew, giving up (pcbnew_do - pcbnew_do:173) (kibot - kiplot.py:149)
ERROR:pcbnew_do returned 9 (kibot - out_pdf_pcb_print.py:136)
```

Unfortunately, I can't share my project, but all 5 of my different PCBs fail with this timeout error.

set-soft, 2022-01-18:
Try running it using `-vvv` and post the debug output, the video capture could also help.

bugadani, 2022-01-18:
I'm running things in github actions, so I'm not sure if I can do either.

set-soft, 2022-01-18:
You can use the `verbose` option for the action. Add `verbose: 3`
The GitHub workflow should capture the debug info and the artifacts should contain the video.
To force artifacts collection even during a fail you must add `if: ${{ always() }}` like this:
```
    - name: Store results
      if: ${{ always() }}
      uses: actions/upload-artifact@v1
      with:
        name: Test_Output_${{ matrix.ki_release }}
        path: output
```

bugadani, 2022-01-18:
Thanks, I'll do it and report back

bugadani, 2022-01-18:
That explains things :)
[pcbnew_export_screencast.zip](https://github.com/INTI-CMNB/KiBot/files/7890064/pcbnew_export_screencast.zip)

set-soft, 2022-01-18:
Ok, I found the problem. KiCad 6's pcbnew can't start without a `kicad_common.json`. I don't have this problem in my tests because the `eeschema_do` tool creates one.
I'm working on a fix.
You can just run an output that uses `eeschema_do` (like printing the schematic before the PCB).

set-soft, 2022-01-18:
I updated the docker images, now this should work.

closed 2022-01-18 by set-soft