INTI-CMNB/KiBot #142: Clearance violations - possible rounding error?  (KiCad 6)

jameswilddev, 2022-01-22:
Hello!

I'm currently tinkering with an [electronics project](https://github.com/jameswilddev/quantum-monster) and decided to try out KiBot!

I seem to be hitting a lot of DRC errors in CI, but I can't replicate them locally in KiCad 6:

```
ERROR:Found 50 DRC errors and 0 unconnected pads (pcbnew_do - pcbnew_do:724) (kibot - kiplot.py:149)
ERROR:(clearance) Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm)
    Rule: netclass 'Default'
    @(124.9630 mm, 95.7260 mm): Track [DISPLAY_DC] on F.Cu, length 4.6685 mm
    @(124.4390 mm, 95.2500 mm): Track [GND] on F.Cu, length 0.1054 mm (pcbnew_do.kiauto.file_util - file_util.py:132) (kibot - kiplot.py:149)
ERROR:(clearance) Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm)
    Rule: netclass 'Default'
    @(124.9630 mm, 95.7260 mm): Track [DISPLAY_DC] on F.Cu, length 4.6685 mm
    @(124.5135 mm, 95.1755 mm): Track [GND] on F.Cu, length 1.5675 mm (pcbnew_do.kiauto.file_util - file_util.py:132) (kibot - kiplot.py:149)
ERROR:(clearance) Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm)
    Rule: netclass 'Default'
    @(124.9630 mm, 95.7260 mm): Track [DISPLAY_DC] on F.Cu, length 4.6685 mm
    @(125.4126 mm, 96.2764 mm): Track [DISPLAY_SDIN] on F.Cu, length 10.8690 mm (pcbnew_do.kiauto.file_util - file_util.py:132) (kibot - kiplot.py:149)
```

[image]

It looks like it might be rounding error given they're all within 0.0005mm.  Is this a known KiCad 6 problem (aware it's new).

Thanks!

theterg, 2022-01-23:
+1 to this issue. I have encountered this as well, very recently migrated a project from KiCad 5 to 6. Received several of these DRC failures in KiBot that were not present in the KiCad DRC report:

[image]

However when I zoomed way, way in on this particular failure, I found that the traces were indeed off by 0.5um!

[image]

This was running a PPA build of Kicad (6.0.1-1-202201152026+79c1e3a40b\~116\~ubuntu20.04.1) and a locally built version of KiBot (fbfee8771dfcbf8bef55c05177646e3f6e9fa51f)

set-soft, 2022-01-29:
Hi @jameswilddev and @theterg !

First thing to check is if this a real violation, like @theterg spotted in his design.
Second thing to check are KiCad versions. Is the one used in CI environment the same used locally? 
KiBot prints the KiCad version when using verbose output.

theterg, 2022-01-30:
Hi @set-soft -

I've checked the versions and I believe they agree (is this what you mean?):
[image]

I know that there's some complexity to whether or not this constitutes a "real violation" (what real world-downside would be associated with a 0.5 micron offset??). But my preference would be for KiBot to mirror the KiCad Design Rules Checker GUI if at all possible. It takes a fair amount of time to find each of these clearance violations without a graphical marker. Worse yet for users that lean heavily on Shove routing it may take a lot of manual work to resolve these sorts of clearance errors (if you have several parallel nets all stacked up with this 0.5um error).

In any event, I have a worry this could be a sneakier / more difficult issue than I suspect. Thank you for your help!

set-soft, 2022-02-19:
Thanks @axemag !
I'm closing it as we can't fix it from KiBot side, and will be fixed in the next release.

closed 2022-02-19 by set-soft