INTI-CMNB/KiBot #146: Gerber Export of User.Drawings layer

frfa, 2022-01-30:
I noticed that when I export the "User.Drawings" (internally "Dwgs.User") layer in Gerber format with units set to "mm", `pcbnew` will export and show measurement units correctly as "mm", while `kibot` will show measurements in inches "in".

This is how a dimension measurement  in the "Dwgs.User" layer is encoded in the `.kicad_pcb` s-expression syntax:
```scheme
  (dimension (type aligned) (layer "Dwgs.User") (tstamp 337fb8fb-ac6f-4139-8183-5a562491ca89)
    (pts (xy 123.9774 122.791) (xy 120.4739 122.791))
    (height -5.5298)
    (gr_text "3,5035 mm" (at 122.22565 127.1708) (layer "Dwgs.User") (tstamp b283c3f8-09ac-4669-8299-5730e409a671)
      (effects (font (size 1 1) (thickness 0.15)) (justify left))
    )
    (format (units 3) (units_format 1) (precision 4))
    (style (thickness 0.1) (arrow_length 1.27) (text_position_mode 0) (extension_height 0.58642) (extension_offset 0.5) keep_text_aligned)
  )
```

Notice that `gr_text` correctly describes the unit as "mm", and it seems the `units` and/or `units_format` expressions describe which unit and format to use.

User.Drawings as exported from `pcbnew` File -> Fabrication Outputs -> Gerbers:
[image]

Same detail from User.Drawings when exported with `kibot`:
[image]

Is there an option that I miss that needs to be set in the `kibot` YAML config file in order to honour the units setting in the `.kicad_pcb` file?

frfa, 2022-02-02:
In addition to that, the `User.Drawings` and `User.Comments` layers do not get added to the JSON Gerber job file `*.gbrjob`, althouth the option `create_gerber_job_file` is set to `true`.

The `layers` specification of the `gerber` type output section looks like this:

```yaml
    layers:
      - copper
      - technical
      - Edge.Cuts
      - User.Drawings
      - User.Comments
```

Interestingly enough, the `Edge.Cuts` layer also specified separately *does* get included in the Gerber job file.

Any hints what might happen here?
Thanks!

set-soft, 2022-02-09:
Hi @frfa !

Both things are made by KiCad. We just say to KiCad "generate the gerber for this layer" and "generate a gerber job file".

The first problem seems to be KiCad 6 trying to be "smart" by recreating the dimensions, I tried KiCad 5 and it doesn't do it.

About the second: Do you get the drawings layer added when using KiCad GUI? I tried KiCad 5 and it doesn't.

I'm investigating how to tell KiCad 6 which units we want.

set-soft, 2022-02-09:
> Notice that `gr_text` correctly describes the unit as "mm", and it seems the `units` and/or `units_format` expressions describe which unit and format to use.

`units 3` means automatic, if you change it to `units 2` (`DIM_UNITS_MODE_MILLIMETRES`) you get millimeters.

I can't figure out what's needed to make `millimiters` the default value. I'm asking it on the [forum](https://forum.kicad.info/t/kicad-6-dimensions-and-python-api/33962)

set-soft, 2022-02-09:
According to KiCad developers you should avoid using the mode 3. Unless until they add a mechanism to give a default from Python.

I think this issue should be closed.

frfa, 2022-02-10:
Hi  @set-soft !

Thanks a lot for your investigation and elaborate answer.
No, the User.Drawings and User.Comments layer don't get added to the .gbrjob file by KiCad 6 either. So this is a KiCad
issue and we'll have to wait for a solution there.
And for the inches instead of millimeters for the default unit I understand that there will be a mechanism to set the default from Python, also a KiCad issue.
I'll close the issue for now while waiting for the next KiCad release.

closed 2022-02-10 by frfa

set-soft, 2022-02-14:
Hi @frfa !

I think I found a solution. Now you can define `units: millimeters` in the `global` section, like this:

```yaml
global:
  units: millimeters
```

This will patch every `dimension` in the loaded PCBs that uses `automatic` units. I tried the simple example I used to diagnose the problem and was able to control what `automatic` means.

frfa, 2022-02-15:
Hi  @set-soft !

Thanks a lot for the fix, KiBot now works flawlessly exporting units in millimeters to the Gerbers.