INTI-CMNB/KiBot #148: eeschema_do fails with "error 10"

neilenns, 2022-02-06:
I have two github repos using KiBot and KiCad 6 to generate output. One is working fine, the other (which was based on the working one) fails.

Working repo: https://github.com/neilenns/CJ4-MFD-Menu-Panel/
Failing repo: https://github.com/neilenns/CJ4-Electrical-Panel

Here's the failure I'm getting:

```
DEBUG:Loading sheet from /workspaces/CJ4-Electrical-Panel/CJ4-Electrical-Panel.kicad_sch (kibot - v6_sch.py:1644)
- 'PDF - print schematic' (print_sch) [pdf_sch_print]
DEBUG:Output destination: /workspaces/CJ4-Electrical-Panel/KiBotOutput/PDFs (kibot - kiplot.py:330)
DEBUG:Running: ['eeschema_do', '--version'] (kibot - kiplot.py:96)
DEBUG:Executing: ['eeschema_do', '-r', '-v', 'export', '--all_pages', '--file_format', 'pdf', '/workspaces/CJ4-Electrical-Panel/CJ4-Electrical-Panel.kicad_sch', '/workspaces/CJ4-Electrical-Panel/KiBotOutput/PDFs'] (kibot - kiplot.py:168)
DEBUG:Failed with error 10, retrying ... (kibot - kiplot.py:177)
ERROR:eeschema reported an error (eeschema_do - eeschema_do:156) (kibot - kiplot.py:149)
DEBUG:Output from command:
> INFO:Waiting for "Main eeschema window" ...
> INFO:Waiting for "^Warning$" ...
> ERROR:eeschema reported an error (eeschema_do - eeschema_do:156)
>  (kibot - kiplot.py:181)
ERROR:eeschema_do returned 10 (kibot - out_pdf_sch_print.py:73)
```

Unfortunately the screen capture recording doesn't contain anything, I think because I'm running this under WSL2 on Windows.

I've tried everything I can think of to chase down what the problem is but can't figure it out. The schematic will print to PDF just fine from within KiCad 6. There are no ERC or DRC errors in the design.

Can you think of what might be causing the failure?

set-soft, 2022-02-06:
Hi @neilenns !

Did you try using GitHub workflows?
I tried it locally without problems.
KiCad is opening a Warning dialog, are you sure you are using a docker image with KiCad 6? The `test_import` won't work, you'll need `v1_k6`.
I think you should get a valid screen capture even on the WSL2, did you convert the OGV into frames? (So you can see each frame, sometimes the important information is in just one frame)

neilenns, 2022-02-06:
Sigh, this is 100% user error 🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️ I copied from the wrong GitHub repo, I based off a KiCad 5 repo instead of KiCad 6.

I really shouldn't create repos before having morning coffee!!!

closed 2022-02-06 by neilenns