INTI-CMNB/KiBot #156: pdf_pcb_print: allow mirroring on a per-layer basis

bugadani, 2022-02-23:
I'm using pdf_pcb_print to generate manual assembly diagrams. I would like to generate a single PDF with both front and backside, rendered as separate pages. So far, so good.

What I can't do currently, is mirroring the back side only in this setup. This is useful, because:
 - by default, everything on B.Fab is mirrored
 - during manual soldering, the board is flipped, and not viewed from a top-down orientation, which would be the case of a non-mirrored rendering

set-soft, 2022-02-26:
I think you can install `poppler-utils` and use `pdfunite` to join two (or more) PDF files.
In this way you can have in the same PDF two pages: one for the TOP and another for the BOTTOM, mirrored.
In addition you can include drawings and other layers added to each  of the above pages.
A `pdfunite` output could be handy. But could take time.

set-soft, 2022-03-03:
Ok, now you can join two or more PDFs, here is an example:

```yaml
kiplot:
  version: 1

outputs:
  - name: PDF_Joined
    comment: "PDF files for top layer + mirrored bottom layer"
    type: pdfunite
    options:
      outputs:
        - from_output: PDF_top
        - from_output: PDF_bottom

  - name: PDF_top
    comment: "PDF files for top layer"
    type: pdf
    dir: PDF
    options:
      exclude_edge_layer: false
      exclude_pads_from_silkscreen: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true

      # PDF options
      drill_marks: small
      mirror_plot: false
      negative_plot: false
      line_width: 0.01
    layers:
      - layer: F.Cu
        suffix: F_Cu
      - layer: F.SilkS
        suffix: F_Silks

  - name: PDF_bottom
    comment: "PDF files for bottom layer"
    type: pdf
    dir: PDF
    options:
      exclude_edge_layer: false
      # PDF options
      drill_marks: full
      mirror_plot: true
      negative_plot: true
      line_width: 0.01
    layers:
      - layer: B.Cu
        suffix: B_Cu
      - layer: B.SilkS
        suffix: B_Silks
```

commit 0c0c6ffd (2022-03-03): New output to join PDFs.

Closes #156
closed 2022-03-03 by set-soft