INTI-CMNB/KiBot #157: `var_rename_kicost` seems to not apply correctly

bugadani, 2022-02-24:
I have the following in my config:

```yaml
filters:
  - name: 'kicost_rename'
    type: var_rename_kicost
    separator: ':'

variants:
  - name: 'MCP73811T'
    type: kicost
    file_id: _MCP73811T
    variant: ^(charger(|.mcp73811))$
    pre_transform: 'kicost_rename'

  - name: 'MCP73812T'
    type: kicost
    file_id: _MCP73812T
    variant: ^(charger(|.mcp73812))$
    pre_transform: 'kicost_rename'

outputs:
  - name: "Schematic diagram (MCP73811T)"
    type: pdf_sch_print
    dir: MCP73811T
    options:
      output: '%f%v-%i.%x'
      variant: MCP73811T

  - name: "Schematic diagram (MCP73812T)"
    type: pdf_sch_print
    dir: MCP73812T
    options:
      output: '%f%v-%i.%x'
      variant: MCP73812T
```

According to the logs, the renaming is applied successfully, here's an excerpt:
```
2022-02-24T10:32:38.7172778Z DEBUG:Parsing filter `kicost_rename` (var_rename_kicost) (kibot - config_reader.py:135)
2022-02-24T10:32:38.7173283Z DEBUG:Parsing variant `MCP73811T` (kicost) (kibot - config_reader.py:135)
2022-02-24T10:32:38.7189932Z DEBUG:Parsing variant `MCP73812T` (kicost) (kibot - config_reader.py:135)
2022-02-24T10:32:38.7190421Z DEBUG:Parsing variant `DEV MCP73811T` (kicost) (kibot - config_reader.py:135)
2022-02-24T10:32:38.7190996Z DEBUG:Parsing variant `DEV MCP73812T` (kicost) (kibot - config_reader.py:135)
2022-02-24T10:32:38.7191576Z DEBUG:Pre-parsing output options for `Schematic diagram (MCP73811T)` (pdf_sch_print) (kibot - config_reader.py:82)
2022-02-24T10:32:38.7192210Z DEBUG:Pre-parsing output options for `Schematic diagram (MCP73812T)` (pdf_sch_print) (kibot - config_reader.py:82)
2022-02-24T10:32:38.7192783Z DEBUG:Pre-parsing output options for `Interactive BoM (MCP73811T)` (ibom) (kibot - config_reader.py:82)
2022-02-24T10:32:38.7193348Z DEBUG:Pre-parsing output options for `Interactive BoM (MCP73812T)` (ibom) (kibot - config_reader.py:82)
...
DEBUG:Applying transform filter `kicost_rename` (kibot - fil_base.py:140)
DEBUG:ref: U2 value: MCP73812T-420I-OT -> MCP73811T-420I-OT (kibot - fil_var_rename_kicost.py:77)
DEBUG:ref: U2 manufacturer part number: MCP73812T-420I/OT -> MCP73811T-420I/OT (kibot - fil_var_rename_kicost.py:77)
```

However, the resulting schematics and iboms contain the non-renamed values:

[image]
[image]

Are my assumptions correct, that the part values should be updated in both the schematics, and the ibom? I haven't testes other outputs yet.

set-soft, 2022-02-24:
@bugadani : can you provide this part of the schematic? Just copy this small part so I can reproduce the situation. Also post the ibom outputs configs

bugadani, 2022-02-24:
I will create the test case for #158 so that it includes this as well, if that's okay with you.

bugadani, 2022-02-24:
Artifacts: https://github.com/bugadani/kibot_bug_repro/actions/runs/1893662537
Repro: https://github.com/bugadani/kibot_bug_repro/tree/de0a8d5685cd63c300cb57a383e70f98cf6ebe82

set-soft, 2022-02-24:
Ok, I think I found the problem. Is very KiCad 6 specific, there are many places where `Value` is defined. The problem here is that the component gets its correct value, but the *instance* doesn't. This is related to the poor design of the v6 format. I pointed out it some months ago: instances handles fields poorly and in a way that make they look as something that isn't a field.
I'll see how to fix it.

set-soft, 2022-02-24:
Now `footprint`, `value` and `reference` are copied to the instance.
The test is working

bugadani, 2022-02-24:
Awesome, thank you!

commit 26a6db4e (2022-02-24): Fixed KiCad 6 schematic component instance not updated

Fixes #157