INTI-CMNB/KiBot #158: Missing argument exception

bugadani, 2022-02-24:
I got this exception while trying to generate some artifacts for my project:

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 383, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 413, in generate_outputs
    run_output(out)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 369, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 125, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_pdf_sch_print.py", line 59, in run
    fname = GS.sch.save_variant(sch_dir)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1626, in save_variant
    self.save(fname, dest_dir)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1610, in save
    _add_items(self.symbols, sch, sep=True, cross=cross)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1462, in _add_items
    sch.append(i.write(cross=True))
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1060, in write
    d = f.write()
TypeError: write() missing 1 required positional argument: 'f'
```

https://github.com/INTI-CMNB/KiBot/blob/17911fe1361b2adcba5e32b909ddc7f5c1d151f7/kibot/kicad/v6_sch.py#L1060

The code in question iterates over the `self.fields` field, which is defined in `SchematicComponent` imported from v5, and looking at the crash, it seems to contain at least some objects of `v5.LibComponentField` type, whose `write` method has an `f` parameter.

I'm trying to figure out what triggered this error in my project.

I'm using KiCad 6 with the `INTI-CMNB/KiBot@v1_k6` github action

set-soft, 2022-02-24:
Hi @bugadani !

> The code in question iterates over the `self.fields` field, which is defined in `SchematicComponent` imported from v5, and looking at the crash, it seems to contain at least some objects of `v5.LibComponentField` type, whose `write` method has an `f` parameter.

Are you reading a KiCad 5 file using KiCad 6? This can't be done.
You must convert the files to v6 format first.
If you still using KiCad 5, use the `v1` tag.

bugadani, 2022-02-24:
@set-soft please see the last 3 lines of my post with links to the relevant code paths. `add_field` uses the v5 code even for v6. I'm not trying to do anything with KiCad v5 files, it's v6 all the way :)

set-soft, 2022-02-24:
@bugadani , I see. Do you have a small test case? I can add an implementation of `set_field`, but without a test it could have some problems.

bugadani, 2022-02-24:
Do you need a complete project with both kibot config and schematic? I'll make one

set-soft, 2022-02-24:
Yes, schematic and config file that triggers the error.

set-soft, 2022-02-24:
This patch fixes the crash, doesn't check if the result is the expected.

commit 1dbaf67c (2022-02-24): Fixed KiCad v6 field rename

- The code created v5 fields

Fixes #158
closed 2022-02-24 by set-soft