INTI-CMNB/KiBot #159: Would it be possible to export Gencad?

flaminggoat, 2022-02-25:
Kicad has the option to export Gencad files. A PCB supplier I use requests this file output for assembly. It would be helpful if Kibot could automate this.

set-soft, 2022-02-26:
This isn't currently available and is ticky.
There are no Python interface for this and the GUI can't define a hot-key for this option.
Sure this could be added to KiAuto, but will be delicated (prone to fail because of slight GUI changes)
I'll  leave it as a low priority enhancement

set-soft, 2022-03-08:
Ok @flaminggoat , time to test it ;-)

closed 2022-03-08 by set-soft

set-soft, 2023-04-11:
Hi @flaminggoat !
KiCad 7.99 is changing some text angle in GenCad files. Do you know about any tool to view these files to check if this is wrong?

commit cb47cf30 (2022-03-08): New output to export PCB in GENCAD format.

Closes #159