INTI-CMNB/KiBot #160: v1_k6 tag no longer works in GitHub Actions

richardbarlow, 2022-03-09:
As of a few moments ago, I can no longer use the KiBot github action. I was using the v1_k6 tag, but have tried master too and get the same failure:

```
The following packages have unmet dependencies:
   curl : Depends: libcurl4 (= 7.74.0-1.3+deb11u1) but 7.81.0-1~bpo11+1 is to be installed
  E: Unable to correct problems, you have held broken packages.
```

richardbarlow, 2022-03-09:
KiBot is a fantastic project by the way and I wholeheartedly support automation and CI of electronic hardware design. I'm happy to test things out as necessary :)

set-soft, 2022-03-09:
Hi @richardbarlow !
Try now, I updated the Dockerfile.
Is related to the upgrade to KiCad 6.0.2, now I'm using pure Debian again, but the package is a backport. This is forcing curl from backports.
Let me know if this is working for you.

richardbarlow, 2022-03-09:
Yep, that's fixed it!

Thank you for sorting it so quickly.

closed 2022-03-09 by richardbarlow