INTI-CMNB/KiBot #161: Support replacing project Text Variables

s-ol, 2022-03-12:
With KiCAD 6, project-level text variables can be declared and expanded in schematic and pcb text, header block fields etc using `${VARIABLE}` syntax. After being configured in either the *Board Setup* or *Schematic Setup* menus, they are added to the `PROJECT.kicad_pro` JSON file under the `text_variables` key.

The schematic file contains only the variable references, not their values. By updating the `kicad_pro` file, the schematic is completely "updated".

The PCB file caches the variable values in `(property "VARIABLE" "VALUE")` fields, but reloads the current value from the `kicad_pro` file when pcbnew is started. The PCB can be "updated" by simply opening the file and saving it immediately.

This mechanism could be used as a cleaner way of accomplishing what `sch_replace` and `pcb_replace` have been used for so far.
A `set_text_var` preflight option could look like this:

```
set_text_var:
  STATIC_TEXT:
    text: "static text. not sure why, but might be useful if yaml is generated..."

  COMMIT_HASH:
    command: "git rev-parse --short HEAD"
```

And would just need to modify the `kicad_pro` JSON document.

closed 2022-03-12 by set-soft

set-soft, 2022-03-12:
Hi @s-ol !
This is a good idea. I didn't investigate much about KiCad 6 variables (where are stored, when are applied, etc.), so you filled this gap ;-)
I implemented it in a way that is more similar to how `pcb_replace` and `sch_replace` are defined, so KiCad 5 users can easilly migrate to the new preflight.
In addition I added %X pattern expansion. This could help people using global variants.

s-ol, 2022-03-13:
This is the error/warning I meant:

```
WARNING:(W075) Trying to reformat SCH time, but not in ISO format (${DATE}) (kibot.gs - gs.py:149)
WARNING:(W075) Problem: Invalid isoformat string: '${DATE}' (kibot.gs - gs.py:150)
```

And this [`pcbnew.ExpandTextVars(str, PROJECT)`](https://docs.kicad.org/doxygen-python/namespacepcbnew.html#ad3104172fde92d0ddbb1080e47a990b7) function looks right.

set-soft, 2022-03-13:
Hi @s-ol !
The last patch (342fb40c59a05e4fb39c8180a8f17dd3acbbdb7e) should expand `DATE` before using it, please let me know if it works for you.

set-soft, 2022-03-13:
> And this [`pcbnew.ExpandTextVars(str, PROJECT)`](https://docs.kicad.org/doxygen-python/namespacepcbnew.html#ad3104172fde92d0ddbb1080e47a990b7) function looks right.

I'm doing the expansion manually to have more control.

s-ol, 2022-03-13:
Hey @set-soft, I'm still getting this issue with current master: (02a942c)

```
WARNING:(W075) Trying to reformat SCH time, but not in ISO format (${VCS_DATE}) (kibot.gs - gs.py:181)
WARNING:(W075) Problem: Invalid isoformat string: '${VCS_DATE}' (kibot.gs - gs.py:182)
```

Heres the start of my config:

```yaml
kibot:
  version: 1

preflight:
  check_zone_fills: true
  run_drc: true
  run_erc: true

  filters:
    - filter: 'Ignore R7/8 overlap with R7/8alt'
      error: 'courtyards_overlap'
      regex:  'Footprint R[78]'

  set_text_variables:
    - name: 'VCS_DATE'
      command: 'git log -1 --format="%as"'

    - name: 'VCS_REV'
      command: 'git rev-parse --short HEAD'
```

set-soft, 2022-03-15:
Ok, try now. The code tried to reformat before we expanded it.

s-ol, 2022-03-20:
Sorry, took a while before I had a chance to try it out again. It seems to work perfectly now :smiley: 

Is it intentional that the variable substitutions remain in the `*.kicad_pro` file after kibot is done? Since the schematic and pcb files aren't updated (or rather, are copied to avoid modifying them) I would maybe expect the same to happen with the project file.

set-soft, 2022-03-21:
> Is it intentional that the variable substitutions remain in the `*.kicad_pro` file after kibot is done?

Yes, is intentional. Some preflight options modify the files. This is one of the cases.

set-soft, 2022-04-23:
> I have a ${date} and ${version} variable. In the XLSX file both get expanded properly. In the ibom the ${date} variable does not expand. Might be more of an issue with the interactivebom plugin than with kibot, but just wanted to let you know.

Do you have an example? I tried it and worked as expected.

set-soft, 2022-04-23:
I used the files involved in the above commit:

```
src/kibot -vvv -b tests/board_samples/kicad_6/light_control.kicad_pcb -c tests/yaml_samples/ibom_no_ops.kibot.yaml -d pp
```

From the KiBot project root (I'm trying with `date` now)

set-soft, 2022-04-23:
Ok @ZOUDIO now I found your problem, you must be more explicit. The *date* displayed at the top of the web page isn't expanded, even when the *rev* is.

set-soft, 2022-04-23:
I reported it to upstream and patched the iBoM we have in CI/CD images (v2.4.1-2 now). It should work in some minutes (after images propagation).

commit 1824d568 (2022-03-12): Added a preflight to define KiCad 6 variables.

Closes #161