INTI-CMNB/KiBot #162: Kicad 6 variable vs %r revision from pcb/sch

stevaedrum, 2022-03-13:
Hi,

Thanks a lot for the support of KICAD 6 variable ;) 
But I have found an issue when I put a variable for the revision in KICAD (for exemple: ${git_tag} the ouput file with %r doen't have the value.

Export file:
      Test-KIBOT-ibom_${git_tag}.html

yaml synthaxe:
      output: '%f-%i_%r.%x'

KICAD Pro:
  "text_variables": {
    "git_hash": "Git Tag: <bc60fef>",
    "git_tag": "v1.0.0"
  }

KICAD Sch:
  (title_block
    (date "2022-03-02_19-32-19")
    (rev "${git_tag}")
    (comment 1 "${git_hash}")
  )

Is it possible to take in account this limitation?
The improvement of KICAD, allows more flexibility in updating cartridges.

Thanks in advance. Great tool ;)

set-soft, 2022-03-13:
Hi @stevaedrum !
I think this patch solves this situation. Now title block items are expanded after loading. Also: when expanding the %X patterns the code tries to expand text variables twice (before and after pattern expansion). This allows some mix of patterns inside variables, but with some depth limitation.
Also note that when using `set_text_variables` it should be the first preflight, in this way variables get a chance to be updated.

closed 2022-03-13 by set-soft

stevaedrum, 2022-03-15:
Hi @set-soft,

Thank very much for your improvement and quick modification.

commit 342fb40c (2022-03-13): Added text variable replace in the title block items

- There are some limitations inm the depth of vars + %X usage
Closes #162