INTI-CMNB/KiBot #170: Time out error on GitLab runner while pdf pcb print

PRidon, 2022-03-30:
I face to timeout problem while CI on runner. I have success on my local pc use the same config, docker image and *.kibot.yaml.

yaml part:
```
  # PDF PCB Print (Portable Document Format):
  # This is the main format to document your PCB.
  # This output is what you get from the 'File/Print' menu in pcbnew.
  - name: 'pdf_pcb_print'
    comment: 'Exports the PCB to the most common exhange format. Suitable for printing.'
    type: 'pdf_pcb_print'
    dir: 'AutoGen/pdf_pcb_print'
    options:
      # [string|list(string)=''] Name of the filter to mark components as not fitted.
      # A short-cut to use for simple cases where a variant is an overkill
      dnf_filter: ''
      # [string='full'] What to use to indicate the drill places, can be none, small or full (for real scale)
      drill_marks: 'none'
      # [boolean=false] Print mirrored (X axis inverted). ONLY for KiCad 6
      mirror: false
      # [boolean=false] Print in black and white
      monochrome: true
      # [string='%f-%i%v.%x'] Filename for the output PDF (%i=layers, %x=pdf). Affected by global options
      output: '%f.%x'
      # `output_name` is an alias for `output`
      # [boolean=true] Include the title-block
      plot_sheet_reference: false
      # [number=1.0] Scale factor (0 means autoscaling)
      scaling: 1.0
      # [boolean=false] Print layers in separated pages
      separated: false
      # [string=''] Board variant to apply
      variant: ''
    layers: ['Edge.Cuts']
```

<details><summary>Log with -vvv</summary>

```
$ kibot --skip-pre all -vvv -b MainBoard.kicad_pcb -c ci_cfg/neo-board-tg840.kibot.yaml pdf_pcb_print
00:58
DEBUG:Detected KiCad v6.0.2 (6.0.2+dfsg-1~bpo11+1 6000002) (kibot - __main__.py:261)
DEBUG:KiCad config path /root/.config/kicad/6.0 (kibot - __main__.py:304)
DEBUG:Importing from /usr/local/lib/python3.9/dist-packages/kibot (kibot - kiplot.py:57)
DEBUG:- Importing out_qr_lib (kibot - kiplot.py:64)
DEBUG:- Importing out_ibom (kibot - kiplot.py:64)
DEBUG:- Importing out_position (kibot - kiplot.py:64)
DEBUG:- Importing out_gencad (kibot - kiplot.py:64)
DEBUG:- Importing out_report (kibot - kiplot.py:64)
DEBUG:- Importing out_download_datasheets (kibot - kiplot.py:64)
DEBUG:- Importing out_boardview (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_gerber (kibot - kiplot.py:64)
DEBUG:- Importing out_dxf (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_any_layer (kibot - kiplot.py:64)
DEBUG:- Importing out_render_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_base (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf (kibot - kiplot.py:64)
DEBUG:- Importing out_gerb_drill (kibot - kiplot.py:64)
DEBUG:- Importing out_svg (kibot - kiplot.py:64)
DEBUG:- Importing out_pdfunite (kibot - kiplot.py:64)
DEBUG:- Importing out_pcbdraw (kibot - kiplot.py:64)
DEBUG:- Importing out_bom (kibot - kiplot.py:64)
DEBUG:- Importing out_step (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_sch_variant (kibot - kiplot.py:64)
DEBUG:- Importing out_kibom (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_base_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_compress (kibot - kiplot.py:64)
[... 435 lines cut ...]
> DEBUG:Closing pipe with 397 (pcbnew_do.kiauto.ui_automation - ui_automation.py:46)
> DEBUG:Terminating 397 (pcbnew_do.kiauto.ui_automation - ui_automation.py:58)
> Traceback (most recent call last):
>   File "/usr/bin/pcbnew_do", line 1107, in <module>
>     print_layers(cfg, id_pcbnew)
>   File "/usr/bin/pcbnew_do", line 288, in print_layers
>     wait_not_focused(id_sel_f[0])
>   File "/usr/lib/python3/dist-packages/kiauto/ui_automation.py", line 337, in wait_not_focused
>     raise RuntimeError('Timed out waiting for %s window to lose focus' % id)
> RuntimeError: Timed out waiting for 4210249 window to lose focus
> DEBUG:Restoring old KiCad common config (pcbnew_do.kiauto.file_util - file_util.py:166)
> DEBUG:Restoring old user hotkeys config (pcbnew_do.kiauto.file_util - file_util.py:166)
> DEBUG:Restoring old pcbnew config (pcbnew_do.kiauto.file_util - file_util.py:166)
> DEBUG:Checking if /tmp/tmp-kibot-pdf_pcb_print-bfzit8gg/MainBoard.kicad_pro was modified (pcbnew_do.kiauto.file_util - file_util.py:291)
> DEBUG:Checking if /tmp/tmp-kibot-pdf_pcb_print-bfzit8gg/MainBoard.kicad_prl was modified (pcbnew_do.kiauto.file_util - file_util.py:291)
>  (kibot - kiplot.py:181)
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot - kiplot.py:183)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot - kiplot.py:184)
DEBUG:Removing temporal variant dir `/tmp/tmp-kibot-pdf_pcb_print-bfzit8gg` (kibot - out_any_pcb_print.py:138)
ERROR:pcbnew_do returned 1 (kibot - out_any_pcb_print.py:141)
ERROR: Job failed: exit code 1
```
</details>

docker image setsoft/kicad_auto:dev_k6

set-soft, 2022-03-31:
Hi @PRidon !

Do you have the video (OGV) generated by this session?

It looks like the dialog missed the "Enter" and didn't continue.

Is the error always at the same step? 

I can try to add a retry there.

PRidon, 2022-03-31:
I don't understand about video. Do you need some screen capture video of the console output? It happen inside gitlab runner. I just can catch listing after that.

I tried three times. Logs are the same, except window id. Each of them spends 57 seconds on `$ kibot --skip-pre all -vvv -b MainBoard.kicad_pcb -c ci_cfg/neo-board-tg840.kibot.yaml pdf_pcb_print`

set-soft, 2022-03-31:
> I don't understand about video. Do you need some screen capture video of the console output? It happen inside gitlab runner. I just can catch listing after that.

The video is generated by `-vvv` you just need to collect the artifacts, even after a fail.

> I tried three times. Logs are the same, except window id. Each of them spends 57 seconds on `$ kibot --skip-pre all -vvv -b MainBoard.kicad_pcb -c ci_cfg/neo-board-tg840.kibot.yaml pdf_pcb_print`

Ok, GitLab machines are known to be problematic. But I need to know if the problem is just the `Enter` or is worst.

PRidon, 2022-03-31:
> The video is generated by -vvv you just need to collect the artifacts, even after a fail.

Roger. How I can get it? Some info about it, pls. Location or way to get it out?

PRidon, 2022-03-31:
`DEBUG:Recording session with: ['recordmydesktop', '--overwrite', '--no-sound', '--no-frame', '--on-the-fly-encoding', '-o', '.../pdf_pcb_print/pcbnew_export_screencast.ogv']`

Is that?

set-soft, 2022-03-31:
> `DEBUG:Recording session with: ['recordmydesktop', '--overwrite', '--no-sound', '--no-frame', '--on-the-fly-encoding', '-o', '.../pdf_pcb_print/pcbnew_export_screencast.ogv']`
> 
> Is that?

Yes, is generated in the same directory as the outputs. You must collect the outputs even when the worflow failed.

PRidon, 2022-04-01:
[artifacts (2).zip](https://github.com/INTI-CMNB/KiBot/files/8395860/artifacts.2.zip)

Main info inside
[image]

set-soft, 2022-04-02:
I think the message is quite clear, you don't have enough permissions for the working directory.

Adding some `ls -la` to the workflow could help to understand which file/folder is the problematic one.

set-soft, 2022-04-02:
I tried my [CI/CD example](https://gitlab.com/set-soft/kicad-ci-test) at GitLab using `setsoft/kicad_auto:dev_k6` and it generated the PDF [without problems](https://gitlab.com/set-soft/kicad-ci-test/-/jobs/2284272943)

set-soft, 2022-04-11:
Hi @PRidon !
Without farther information I assume this is a permissions problem that must be fixed in your setup.
If you find more information that suggests this is something on KiBot side please open a new issue with the specific problem.