INTI-CMNB/KiBot #171: kicad-packages3D Not working

KenwoodFox, 2022-03-30:
Hey all! New here~ trying out building step files with `type: step`

it builds the PCB just fine witch is awesome! but seems it cant locate any of the other 3d models, took a peek and the default 3d model link: https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/ seems broken, is there another link? Thanks!

set-soft, 2022-03-31:
Hi @KenwoodFox !

I found the problem: files created by KiCad 6 uses `KICAD6_3DMODEL_DIR`, files imported from KiCad 5 uses `KISYS3DMOD`. The former wasn't working.

Now should work. 

BTW:  https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/  works, just try soemthing like: https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x04_P2.54mm_Horizontal.wrl

closed 2022-03-31 by set-soft

KenwoodFox, 2022-03-31:
> 

Ah! I tested out the example link and you're right, it does work, just dosent show the root i guess, testing this out rn!