INTI-CMNB/KiBot #172: Question: Custom 3D model directories

KenwoodFox, 2022-03-31:
The issue with `KICAD6_3DMODEL_DIR` was fixed (thanks so much~) But now im trying to setup the ESPRESSIF_3DMODELS custom 3d models.

I added their git repo as a submodule https://github.com/espressif/kicad-libraries and populated the ESPRESSIF_3DMODELS as a custom text replacement in my yaml file
```yaml
    - name: "ESPRESSIF_3DMODELS" # 3D models for ESP devices
      command: "echo 'Hardware/BattleBot_Motherboard/Libraries/kicad-libraries/3d'"
```

My other text replacements seem to work just not this one, it errors out that it cant find the model still
```
 ERROR:Unable to expand `ESPRESSIF_3DMODELS` in `${ESPRESSIF_3DMODELS}/ESP32-S2-WROVER.STEP` (kibot - config.py:64)
WARNING:(W047) Missing 3D model for U1: `/github/workspace/${ESPRESSIF_3DMODELS}/ESP32-S2-WROVER.STEP` (kibot - out_base_3d.py:126)
```

This is not really an issue and more of an open question so feel free to close for unrelated, i could not find an irc or gitter channel for KiBot. Thanks for looking in!

set-soft, 2022-03-31:
Ok, I understand your idea, is straight-forward. But:

1. Is not implemented in KiBot, what you get expanded is what KiCad defines in `kicad_common.json` (section `environment` sub-section `vars`). But I understand this is only useful for a regular machine and quite annoying for a docker image.
2. I verified that `kicad2step` doesn't support it. I defined a `text_variable` called `MYVAR` and then got `File not found: ${MYVAR}/R_0805_2012Metric.step` from it. But it seems to work from KiCad, i.e. correct 3D View.

I'll try to see if I can make it work.

set-soft, 2022-03-31:
@KenwoodFox try it, it passes a small test.

The code is expanding the text vars and passing a PCB with expanded vars to kicad2step

KenwoodFox, 2022-04-01:
This totally works! You fixed it, i dident even have to change a single thing on my end just re-run the ci/cd. 
[image]

You guys rock, thank you!

set-soft, 2022-04-02:
> i dident even have to change a single thing on my end just re-run the ci/cd.

Yes, CI/CD concept is quite powerful ... and dangerous, if I introduce a nasty bug all workflows could fail ;-)

I should release a new KiBot version and let `dev*` images only for people needing the last updates/fixes.

KenwoodFox, 2022-04-02:
> > i dident even have to change a single thing on my end just re-run the ci/cd.
> 
> Yes, CI/CD concept is quite powerful ... and dangerous, if I introduce a nasty bug all workflows could fail ;-)
> 
> I should release a new KiBot version and let `dev*` images only for people needing the last updates/fixes.

Good idea, that might sheild us a bit, or we could use specific image hashes instead of the `v2_k6` tag?

set-soft, 2022-04-02:
I'll create tags to use the `dev` images and `v2_k6` will point to the last stable release.