INTI-CMNB/KiBot #173: Text replacement not occuring in board renders

KenwoodFox, 2022-04-02:
Looks like variable text replacement dosent happen in board renders?

[image]

It seems to work fine for the schematic view though
[image]
[image]

KenwoodFox, 2022-04-02:
I should note that this is kicad 6! And the readme does mention it being broken in this release!

KenwoodFox, 2022-04-02:
Text replacement does work for the gerber files though 
[image]

set-soft, 2022-04-02:
Ok, I fixed this problem in `render_3d` output.

Note that this problem was because we need to copy the project along with any temporal PCB file. This wasn't needed before the introduction of text variables. So you could find the same problem on other outputs.

closed 2022-04-02 by set-soft

KenwoodFox, 2022-04-02:
Confirmed! Fixed! 
[image]

This is an incredible amount of issues fixed in such a short time! I am very impressed and appreciative of your hard work!

KenwoodFox, 2022-04-27:
Im wondering if its because i use `-s all` in the gitlab ci job and maybe its not running the preflight text replacement?

set-soft, 2022-04-27:
Most probably. Is this the only call to KiBot in the workflow?

set-soft, 2022-04-27:
Try using `-s run_erc,run_drc` or similar

KenwoodFox, 2022-04-27:
Its not the only one, ive got a drc and erc check formatted like this: 

```yaml
- "kibot -e hardware/ID-4001-WIFI/ID-4001-WIFI.kicad_sch -c hardware/config.kibot.yaml -d Fabrication -s update_xml,run_drc -i"
 ```

Can i force the `-s all` to explicitily do text replacement?

set-soft, 2022-04-27:
> Can i force the `-s all` to explicitily do text replacement?

`-s run_erc,run_drc,update_xml`

KenwoodFox, 2022-04-27:
That did it, thanks!

commit bb0f86ae (2022-04-02): Fixed KiCad 6 text variables not expanded in `render_3d` output

- We must copy the project file to the temporal dir
Fixes #173