INTI-CMNB/KiBot #174: Relative paths to title block worksheets (*.wks) don't work when KiBot makes a copy of the PCB layout

richardbarlow, 2022-04-04:
When generating a PDF (`pdf_pcb_print`) and using the `title` option KiBot fails if the path to `page_layout_descr_file` in the `.kicad_pro` file is relative to the project.

All paths in my projects are relative to the project itself (otherwise it's impossible to version control). KiBot appears to make a copy of the PCB file to apply the modification to the title field (although, interestingly, the `pcb_replace` operations don't perform a copy). When it makes this copy it breaks the relative paths.

If necessary, I can probably create a minimal example, but my description might be clear enough as-is?

For reference, my `page_layout_descr_file` value is `${KIPRJMOD}/../lib/title_block.kicad_wks` and all KiCAD operations behave correctly with this. The only exception is when KiBot creates a copy.

I wonder if it's possible for KiBot to set `KIPRJMOD` to point to the source directory, rather than the copy directory?

closed 2022-04-04 by set-soft

set-soft, 2022-04-04:
Hi @richardbarlow !
Now it should work. I added two tests (one PCB and one schematic) for KiCad 5 and KiCad 6. The tests passed.
This was really complex, we can't change KIPRJMOD this is the place where the project is located, not a real variable.
The copy to a temporal folder is done to keep the file names unaltered. Otherwise you'll get a bogus filename (the name of the temporal file).
Now KiBot is scanning the copied project and looks for the WKS files. If found they are copied to the temporal folder and the project is altered to make reference to the copied files.

richardbarlow, 2022-04-05:
Oh my goodness, this is fantastic! Thank you for fixing it so quickly!

I can't guarantee anything, but is there a way that my company might be able to support the development of KiBot? It is a great project and I personally feel it is a critical piece of software to allow KiCAD to be used in a professional setting. I would like to ensure that development and support continues and that KiBot grows into something great!

set-soft, 2022-04-12:
> I can't guarantee anything, but is there a way that my company might be able to support the development of KiBot? It is a great project and I personally feel it is a critical piece of software to [...]

Any contribution is welcome, code, ideas, information (i.e. a KiBot configuration that generates gerbers for a particular manufacturer), etc. 
I also added a PayPal donate button to the main page.

commit 9e1f645c (2022-04-04): Fixed problems when using page layout files with relative paths.

Fixes #174