INTI-CMNB/KiBot #176: Error with STEP & Render 3D export

stevaedrum, 2022-04-04:
Hi,

I have detected two errors with STEP and Render 3D.
Installed on Fedora 35 and version [4b0f7a2].

Step export enable, Log with kibot -v:

------------------------------------------------------------------------------------------------------------------------------------------------
- 'Exports the PCB as a 3D model.' (step_Export) [step]
DEBUG:Output destination: /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Step (kibot - kiplot.py:347)
DEBUG:Filters reset (kibot - fil_base.py:161)
DEBUG:Applying filter `_mechanical` to exclude (kibot - fil_base.py:154)
DEBUG:Applying filter `_kibom_dnf_Config` to fitted (kibot - fil_base.py:171)
DEBUG:Applying filter `_kibom_dnc_Config` to fixed (kibot - fil_base.py:181)
DEBUG:Applying KiBoM style variants `default` (kibot - var_kibom.py:97)
DEBUG:Running: ['kicad2step_do', '--version'] (kibot - kiplot.py:96)
DEBUG:Reading KiCad config from `/home/a.stephan1/.config/kicad/6.0/kicad_common.json` (kibot - config.py:257)
DEBUG:Detected KICAD_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:270)
DEBUG:Exporting KISYS3DMOD="/usr/share/kicad/3dmodels" (kibot - config.py:295)
DEBUG:Exporting KICAD6_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:306)
DEBUG:Exporting KICAD6_3DMODEL_DIR="/usr/share/kicad/3dmodels" (kibot - config.py:314)
DEBUG:Exporting KICAD6_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:322)
DEBUG:Exporting KICAD6_FOOTPRINT_DIR="/usr/share/kicad/footprints" (kibot - config.py:330)
DEBUG:Loading symbols lib table `/home/a.stephan1/.config/kicad/6.0/sym-lib-table` (kibot - config.py:336)
DEBUG:Storing modified PCB to `/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/tmpfu87o_cg.kicad_pcb` (kibot - out_base_3d.py:188)
DEBUG:Copying project `/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Test-KIBOT.kicad_pro` to `/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/tmpfu87o_cg.kicad_pro` (kibot.gs - gs.py:336)
DEBUG:Executing: ['kicad2step_do', '-o', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Step/Test-KIBOT-3D_v1.0.0.step', '-f', '-vv', '--subst-models', '--grid-origin', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/tmpfu87o_cg.kicad_pcb'] (kibot - out_step.py:84)
ERROR:Failed to create Step file, error 255 (kibot - out_step.py:88)
DEBUG:Output from command: DEBUG:Detected KiCad v6.0.4 (6.0.4-1.fc35 6000004) (kicad2step_do - misc.py:142)
DEBUG:Config path /home/a.stephan1/.config/kicad/6.0 (kicad2step_do - misc.py:157)
DEBUG:KiCad environment: {'KICAD6_3DMODEL_DIR': '/mnt/Data-Raid1/data/VirtualMachines/Kicad_Library/Libraries/kicad-packages3D', 'KICAD6_FOOTPRINT_DIR': '/mnt/Data-Raid1/data/VirtualMachines/Kicad_Library/Libraries/kicad-footprints', 'KICAD6_SYMBOL_DIR': '/mnt/Data-Raid1/data/VirtualMachines/Kicad_Library/Libraries/kicad-symbols', 'KICAD_TEMPLATE_DIR': '/usr/share/kicad/template', 'KICAD_USER_TEMPLATE_DIR': '/home/a.stephan1/kicad/template', 'KIGITHUB': 'https://github.com/KiCad', 'SNAPEDA_3D_MOD': '/mnt/Data-Raid1/data/VirtualMachines/Kicad_Library/SamacSys_PCB_Library/KiCad/SamacSys_Parts.3dshapes'} (kicad2step_do - misc.py:237)
INFO:KiCad to STEP wrapper
DEBUG:Command: ['kicad2step', '-o', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Step/Test-KIBOT-3D_v1.0.0.step', '-f', '--grid-origin', '--subst-models', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/tmpfu87o_cg.kicad_pcb'] (kicad2step_do - kicad2step_do:136)
DEBUG:Waiting for virtual X server ... (kicad2step_do.kiauto.ui_automation - ui_automation.py:88)
DEBUG:Current DISPLAY is :983960540 (kicad2step_do.kiauto.ui_automation - ui_automation.py:90)
DEBUG:Checking using ['setxkbmap', '-query'] (kicad2step_do.kiauto.ui_automation - ui_automation.py:101)
INFO:Waiting for "KiCad to STEP" ...
DEBUG:get_windows 1294  (kicad2step_do.kiauto.ui_automation - ui_automation.py:273)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:275)
DEBUG:get_windows 1294  (kicad2step_do.kiauto.ui_automation - ui_automation.py:273)
DEBUG:get_windows 2097159 Kicad2step (kicad2step_do.kiauto.ui_automation - ui_automation.py:273)
DEBUG:get_windows 2097160  (kicad2step_do.kiauto.ui_automation - ui_automation.py:273)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:275)
DEBUG:Found KiCad to STEP window (1) (kicad2step_do.kiauto.ui_automation - ui_automation.py:371)
DEBUG:Window id: 2097159 (kicad2step_do.kiauto.ui_automation - ui_automation.py:376)
DEBUG:Waiting for 2097159 window to get focus... (kicad2step_do.kiauto.ui_automation - ui_automation.py:307)
DEBUG:Currently focused id: 2097159 (kicad2step_do.kiauto.ui_automation - ui_automation.py:310)
DEBUG:Clipboard retrieve "xclip: Error: There is no owner for the CLIPBOARD selection
" (kicad2step_do.kiauto.ui_automation - ui_automation.py:255)
INFO:xclip: Error: There is no owner for the CLIPBOARD selection

ERROR:Still running (kicad2step_do - kicad2step_do:61)
DEBUG:Closing pipe with 572818 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Killing 572818 (kicad2step_do.kiauto.ui_automation - ui_automation.py:77)
DEBUG:Error level -1 (kicad2step_do - kicad2step_do:144)
 (kibot - out_step.py:90)
[a.stephan1@frtn01s Test-KIBOT]$
------------------------------------------------------------------------------------------------------------------------------------------------

Render 3D export enable, Log with kibot -v:

------------------------------------------------------------------------------------------------------------------------------------------------
- 'Exports the image generated by KiCad's 3D viewer.' (render_3d_example) [render_3d]
DEBUG:Output destination: /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Render_3D (kibot - kiplot.py:347)
DEBUG:Reading KiCad config from `/home/a.stephan1/.config/kicad/6.0/kicad_common.json` (kibot - config.py:257)
DEBUG:Detected KICAD_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:270)
DEBUG:Exporting KISYS3DMOD="/usr/share/kicad/3dmodels" (kibot - config.py:295)
DEBUG:Exporting KICAD6_SYMBOL_DIR="/usr/share/kicad/symbols" (kibot - config.py:306)
DEBUG:Exporting KICAD6_3DMODEL_DIR="/usr/share/kicad/3dmodels" (kibot - config.py:314)
DEBUG:Exporting KICAD6_TEMPLATE_DIR="/usr/share/kicad/template" (kibot - config.py:322)
[... 21 lines cut ...]
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> INFO:Waiting for "Options dialog" ...
> ERROR:Time-out waiting for 3D viewer to be responsive (pcbnew_do - pcbnew_do:421)
>  (kibot - kiplot.py:181)
ERROR:pcbnew_do returned 9 (kibot - out_render_3d.py:187)
[a.stephan1@frtn01s Test-KIBOT]$
------------------------------------------------------------------------------------------------------------------------------------------------

Thanks in advance for your support and help

set-soft, 2022-04-04:
Hi @stevaedrum !

Joining two reports doesn't help much.
Anyways, this information isn't enough to diagnose the problem. You must:

1. Raise the debug level to at least 3 (-vvv)
2. Collect the generated video files (OGV) to see what's going on.

At first sight this looks like an extremely slow (or overloaded) machine. The first error is like this:

1. KiAuto runs kicad2step (a GUI tool in KiCad 6, only god knows why)
2. KiAuto detects the window is there
3. KiAuto sends a click to select the text area
4. KiAuto sends Ctrl+A to select all the text and then Ctrl+C to copy it to the clipboard
5. KiAuto runs xclip to get the clipboard content
6. xclip says nothing is in the clipboard

Looking at the video we could see if the window was actually selected and the text selected.
Without it I must assume the machine was so loaded that the window was there, but the events went lost.

stevaedrum, 2022-04-06:
Hi,

Please find the log file with -vvvv for the render 3D export.

[Log-render3D.zip](https://github.com/INTI-CMNB/KiBot/files/8430443/Log-render3D.zip)
[kibot-vvv.log](https://github.com/INTI-CMNB/KiBot/files/8430444/kibot-vvv.log)

Thanks in advance for your support.

Best Regards

set-soft, 2022-04-07:
Hi @stevaedrum !
Which KiAuto are you using? The traceback doesn't match with the current code. Looks like an older release that doesn't retry if the 3D Viewer window isn't opened.
Last release is 1.6.8.

stevaedrum, 2022-04-07:
HI @set-soft ,

Thanks for your answer and feedback I have twice install of kiauto. Now I have installed the correct version.
the export render 3D is working now.

But for the STEP export, the step file is generate correctly but I have and error !!!
Please you can see below the log with -vvvv

Thanks in advance.

[a.stephan1@frtn01s Test-KIBOT]$ kibot -vvv
DEBUG:Detected KiCad v6.0.4 (6.0.4-1.fc35 6000004) (kibot - __main__.py:261)
DEBUG:KiCad config path /home/a.stephan1/.config/kicad/6.0 (kibot - __main__.py:304)
DEBUG:Importing from /home/a.stephan1/ASCIE/Tools/KiBot/kibot (kibot - kiplot.py:57)
DEBUG:- Importing out_bom (kibot - kiplot.py:64)
DEBUG:- Importing out_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_download_datasheets (kibot - kiplot.py:64)
DEBUG:- Importing out_hpgl (kibot - kiplot.py:64)
DEBUG:- Importing out_gerber (kibot - kiplot.py:64)
DEBUG:- Importing out_compress (kibot - kiplot.py:64)
DEBUG:- Importing out_gerb_drill (kibot - kiplot.py:64)
DEBUG:- Importing out_dxf (kibot - kiplot.py:64)
DEBUG:- Importing out_base_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_any_layer (kibot - kiplot.py:64)
DEBUG:- Importing out_gencad (kibot - kiplot.py:64)
DEBUG:- Importing out_position (kibot - kiplot.py:64)
DEBUG:- Importing out_render_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_any_drill (kibot - kiplot.py:64)
DEBUG:- Importing out_ibom (kibot - kiplot.py:64)
DEBUG:- Importing out_any_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_report (kibot - kiplot.py:64)
DEBUG:- Importing out_pdfunite (kibot - kiplot.py:64)
DEBUG:- Importing out_step (kibot - kiplot.py:64)
DEBUG:- Importing out_any_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_ps (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_boardview (kibot - kiplot.py:64)
DEBUG:- Importing out_pcbdraw (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_sch_variant (kibot - kiplot.py:64)
DEBUG:- Importing out_qr_lib (kibot - kiplot.py:64)
DEBUG:- Importing out_kibom (kibot - kiplot.py:64)
DEBUG:- Importing out_svg (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_excellon (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf (kibot - kiplot.py:64)
DEBUG:- Importing out_kicost (kibot - kiplot.py:64)
DEBUG:- Importing out_base (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing pre_annotate_power (kibot - kiplot.py:64)
DEBUG:- Importing pre_filters (kibot - kiplot.py:64)
DEBUG:- Importing pre_update_xml (kibot - kiplot.py:64)
DEBUG:- Importing pre_sch_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_annotate_pcb (kibot - kiplot.py:64)
DEBUG:- Importing pre_erc_warnings (kibot - kiplot.py:64)
DEBUG:- Importing pre_set_text_variables (kibot - kiplot.py:64)
DEBUG:- Importing pre_check_zone_fills (kibot - kiplot.py:64)
DEBUG:- Importing pre_erc (kibot - kiplot.py:64)
DEBUG:- Importing pre_fill_zones (kibot - kiplot.py:64)
DEBUG:- Importing pre_update_qr (kibot - kiplot.py:64)
DEBUG:- Importing pre_any_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_pcb_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_base (kibot - kiplot.py:64)
DEBUG:- Importing pre_ignore_unconnected (kibot - kiplot.py:64)
DEBUG:- Importing pre_drc (kibot - kiplot.py:64)
DEBUG:- Importing var_ibom (kibot - kiplot.py:64)
DEBUG:- Importing var_base (kibot - kiplot.py:64)
DEBUG:- Importing var_kicost (kibot - kiplot.py:64)
DEBUG:- Importing var_kibom (kibot - kiplot.py:64)
DEBUG:- Importing fil_var_rename (kibot - kiplot.py:64)
[... 637 lines cut ...]
DEBUG:Command: ['kicad2step', '-o', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Step/Test-KIBOT-3D_v1.0.0.step', '-f', '--grid-origin', '--subst-models', '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/tmpt6osmnlf.kicad_pcb'] (kicad2step_do - kicad2step_do:136)
DEBUG:Waiting for virtual X server ... (kicad2step_do.kiauto.ui_automation - ui_automation.py:88)
DEBUG:Current DISPLAY is :1525771280 (kicad2step_do.kiauto.ui_automation - ui_automation.py:90)
DEBUG:Checking using ['setxkbmap', '-query'] (kicad2step_do.kiauto.ui_automation - ui_automation.py:101)
INFO:Waiting for "KiCad to STEP" ...
DEBUG:get_windows 1294  (kicad2step_do.kiauto.ui_automation - ui_automation.py:276)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:278)
DEBUG:get_windows 1294  (kicad2step_do.kiauto.ui_automation - ui_automation.py:276)
DEBUG:get_windows 2097159 Kicad2step (kicad2step_do.kiauto.ui_automation - ui_automation.py:276)
DEBUG:get_windows 2097160  (kicad2step_do.kiauto.ui_automation - ui_automation.py:276)
DEBUG:get_windows end of list (kicad2step_do.kiauto.ui_automation - ui_automation.py:278)
DEBUG:Found KiCad to STEP window (1) (kicad2step_do.kiauto.ui_automation - ui_automation.py:374)
DEBUG:Window id: 2097159 (kicad2step_do.kiauto.ui_automation - ui_automation.py:379)
DEBUG:Waiting for 2097159 window to get focus... (kicad2step_do.kiauto.ui_automation - ui_automation.py:310)
DEBUG:Currently focused id: 2097159 (kicad2step_do.kiauto.ui_automation - ui_automation.py:313)
DEBUG:Clipboard retrieve "xclip: Error: There is no owner for the CLIPBOARD selection
" (kicad2step_do.kiauto.ui_automation - ui_automation.py:258)
INFO:xclip: Error: There is no owner for the CLIPBOARD selection

ERROR:Still running (kicad2step_do - kicad2step_do:61)
DEBUG:Closing pipe with 1811623 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Killing 1811623 (kicad2step_do.kiauto.ui_automation - ui_automation.py:77)
DEBUG:Error level -1 (kicad2step_do - kicad2step_do:144)
 (kibot - out_step.py:88)
[a.stephan1@frtn01s Test-KIBOT]$

set-soft, 2022-04-07:
Hi @stevaedrum !
I added video recording to the `step` output, update KiBot and post the video so we can see if the text gets selected.

stevaedrum, 2022-04-09:
Hi @set-soft,

Thank for your support, please see the video concerning this issue.
[kicad2step_screencast.zip](https://github.com/INTI-CMNB/KiBot/files/8456958/kicad2step_screencast.zip)

set-soft, 2022-04-09:
Hi @stevaedrum !
The video shows that the text isn't selected, but the STEP file was generated.
Looking closer to the log and the code looks like the message generated by `xclip` changed in the version included on Fedora 35. 
Instead od just saying *STRING not available* it explains that *There is no owner for the CLIPBOARD selection*. I added this message to the conditional that detects that we must wait and try again.
Can you test the git code? (you can clone the repo and use pip to install the code from the repo, is something like `pip3 install .` (usually with `--user`)

stevaedrum, 2022-04-10:
Hi @set-soft ,

Please could you find the log after the update of KiAuto. Now there are no error :)

[a.stephan1@frtn01s Test-KIBOT]$ kibot -vvv
DEBUG:Detected KiCad v6.0.4 (6.0.4-1.fc35 6000004) (kibot - __main__.py:261)
DEBUG:KiCad config path /home/a.stephan1/.config/kicad/6.0 (kibot - __main__.py:304)
DEBUG:Importing from /home/a.stephan1/ASCIE/Tools/KiBot/kibot (kibot - kiplot.py:57)
DEBUG:- Importing out_bom (kibot - kiplot.py:64)
DEBUG:- Importing out_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_download_datasheets (kibot - kiplot.py:64)
DEBUG:- Importing out_hpgl (kibot - kiplot.py:64)
DEBUG:- Importing out_gerber (kibot - kiplot.py:64)
DEBUG:- Importing out_compress (kibot - kiplot.py:64)
DEBUG:- Importing out_gerb_drill (kibot - kiplot.py:64)
DEBUG:- Importing out_dxf (kibot - kiplot.py:64)
DEBUG:- Importing out_base_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_any_layer (kibot - kiplot.py:64)
DEBUG:- Importing out_gencad (kibot - kiplot.py:64)
DEBUG:- Importing out_position (kibot - kiplot.py:64)
DEBUG:- Importing out_render_3d (kibot - kiplot.py:64)
DEBUG:- Importing out_any_drill (kibot - kiplot.py:64)
DEBUG:- Importing out_ibom (kibot - kiplot.py:64)
DEBUG:- Importing out_any_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_report (kibot - kiplot.py:64)
DEBUG:- Importing out_pdfunite (kibot - kiplot.py:64)
DEBUG:- Importing out_step (kibot - kiplot.py:64)
DEBUG:- Importing out_any_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_ps (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing out_boardview (kibot - kiplot.py:64)
DEBUG:- Importing out_pcbdraw (kibot - kiplot.py:64)
DEBUG:- Importing out_svg_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_sch_variant (kibot - kiplot.py:64)
DEBUG:- Importing out_qr_lib (kibot - kiplot.py:64)
DEBUG:- Importing out_kibom (kibot - kiplot.py:64)
DEBUG:- Importing out_svg (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_sch_print (kibot - kiplot.py:64)
DEBUG:- Importing out_excellon (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf (kibot - kiplot.py:64)
DEBUG:- Importing out_kicost (kibot - kiplot.py:64)
DEBUG:- Importing out_base (kibot - kiplot.py:64)
DEBUG:- Importing out_pdf_pcb_print (kibot - kiplot.py:64)
DEBUG:- Importing pre_annotate_power (kibot - kiplot.py:64)
DEBUG:- Importing pre_filters (kibot - kiplot.py:64)
DEBUG:- Importing pre_update_xml (kibot - kiplot.py:64)
DEBUG:- Importing pre_sch_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_annotate_pcb (kibot - kiplot.py:64)
DEBUG:- Importing pre_erc_warnings (kibot - kiplot.py:64)
DEBUG:- Importing pre_set_text_variables (kibot - kiplot.py:64)
DEBUG:- Importing pre_check_zone_fills (kibot - kiplot.py:64)
DEBUG:- Importing pre_erc (kibot - kiplot.py:64)
DEBUG:- Importing pre_fill_zones (kibot - kiplot.py:64)
DEBUG:- Importing pre_update_qr (kibot - kiplot.py:64)
DEBUG:- Importing pre_any_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_pcb_replace (kibot - kiplot.py:64)
DEBUG:- Importing pre_base (kibot - kiplot.py:64)
DEBUG:- Importing pre_ignore_unconnected (kibot - kiplot.py:64)
DEBUG:- Importing pre_drc (kibot - kiplot.py:64)
DEBUG:- Importing var_ibom (kibot - kiplot.py:64)
DEBUG:- Importing var_base (kibot - kiplot.py:64)
DEBUG:- Importing var_kicost (kibot - kiplot.py:64)
DEBUG:- Importing var_kibom (kibot - kiplot.py:64)
DEBUG:- Importing fil_var_rename (kibot - kiplot.py:64)
DEBUG:- Importing fil_field_rename (kibot - kiplot.py:64)
DEBUG:- Importing fil_rot_footprint (kibot - kiplot.py:64)
DEBUG:- Importing fil_base (kibot - kiplot.py:64)
DEBUG:- Importing fil_subparts (kibot - kiplot.py:64)
DEBUG:- Importing fil_generic (kibot - kiplot.py:64)
DEBUG:- Importing fil_var_rename_kicost (kibot - kiplot.py:64)
[... 343 lines cut ...]
Create PCB solid model
Build board outline (3 items).
Build board cutouts and holes (2 holes).

Generate board full shape.
Write STEP file

*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 0                      ******
** WorkSession : Sending all data
 Step File Name : $tempfile$.step(25536 ents)  Write  Done

STEP file '/mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-KIBOT/Export/Step/Test-KIBOT-3D_v1.0.0.step' created.

DEBUG:Closing pipe with 3423947 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Terminating the session recorder (kicad2step_do.kiauto.ui_automation - ui_automation.py:169)
DEBUG:Closing pipe with 3423946 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Error level 0 (kicad2step_do - kicad2step_do:144)
 (kibot - out_step.py:101)
Found 5 unique warning/s (6 total)
[a.stephan1@frtn01s Test-KIBOT]$

closed 2022-04-10 by set-soft