INTI-CMNB/KiBot #178: KiCad 6: hierarhical projects doesn't work when sheet imported from other project

WisniewskiP, 2022-04-07:
When processing projects that has imported sheets KiBot throws exception:
```
DEBUG:Loading sheet from /workspaces/test/top/top.kicad_sch (kibot - v6_sch.py:1707)
DEBUG:Loading sheet from /workspaces/test/top/test.kicad_sch (kibot - v6_sch.py:1707)
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 16, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 398, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 415, in generate_outputs
    config_output(out)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 361, in config_output
    load_sch()
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 275, in load_sch
    GS.sch = load_any_sch(GS.sch_file, GS.sch_basename)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 248, in load_any_sch
    sch.load(file, project)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1820, in load
    sch.sch = sch.load_sheet(project, fname, self)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1368, in load_sheet
    sheet.load(os.path.join(parent_dir, self.file), project, parent_obj)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kicad/v6_sch.py", line 1835, in load
    comp = sheet.symbol_uuids[comp_uuid]
KeyError: '38a501e2-0ee8-439d-bd02-e9e90e7503e9'
```

**Steps to reproduce:**
1. Create first project with one sheet containing one component & annotate it. (test.kicad_sch)
2. Create second project and import sheet from first one as Hierarchical sheet
3. try to generate any output with KiBot for second project (top.kicad_sch)
I have attached example project: [top.zip](https://github.com/INTI-CMNB/KiBot/files/8442307/top.zip)

**Quick investigation**
in both schematics _sheet_instances_ & _symbol_instances_ exists:
- test.kicad_sch:
```
  (sheet_instances
    (path "/" (page "1"))
  )
  (symbol_instances
    (path "/38a501e2-0ee8-439d-bd02-e9e90e7503e9"
      (reference "R101") (unit 1) (value "R") (footprint "")
    )
  )
```
- top.kicad_sch:
```
  (sheet_instances
    (path "/" (page "1"))
    (path "/22c28634-55a5-4f76-9217-6b70ddd108b8" (page "2"))
  )
  (symbol_instances
    (path "/22c28634-55a5-4f76-9217-6b70ddd108b8/38a501e2-0ee8-439d-bd02-e9e90e7503e9"
      (reference "R1") (unit 1) (value "R") (footprint "")
    )
  )
```
when processing test.kicad_sch as sub-sheet:
 - path is set to /
 - sheet.symbol_uuids is {}
 - s.path is /38a501e2-0ee8-439d-bd02-e9e90e7503e9

sheet.symbol_uuids is coming from top.kicad_sch when I add a component to the top, then it is listed inside sheet.symbol_uuids

**Comparison to working sub-sheets**

I have some sub-sheets that are working. They don't have sheet_instances and symbol_instances defined.
This problem happens only when one sheet is imported/copied from other project that didn't has hierarchy.

WisniewskiP, 2022-04-07:
one solution could be parsing symbol_instances only on parent.
```
elif e_type == 'symbol_instances':
    if parent is None:
        self.symbol_instances = SymbolInstance.parse(e)
```
but I don't know if there are other edge cases for this.

set-soft, 2022-04-07:
Hi @WisniewskiP !
Thanks for the information. I think skipping symbol instances is a good idea, but could generate problems for outputs that generates an schematic file that is supposed to be edited by the user. This isn't frequent, but could happend. So I take a deeper look at this issue.

set-soft, 2022-04-07:
I'm trying to skip the extra processing done for sheet/symbol instances on sub-sheets. All the relevant processing applies only to the top-level. But the data is parsed so we can include it in saved schematics.

commit 73b4196e (2022-04-07): Fixed problems with imported sub-sheets

- We must avoid processing the sheet and symbol instances of them
Fixes #178