INTI-CMNB/KiBot #181: question about importing preflights

nerdyscout, 2022-04-09:
I am currently trying to set up imports my first time. really a cool feature! but could it be the preflight section is not imported? At least it feels like this is my current problem... I set up a drc.kibot.yaml with drc filters and update_xml and so on. importing this file in a bom.kibot.yaml is possible, but updating the xml is still neccessary.

set-soft, 2022-04-09:
Hi @nerdyscout !
The regular import mechanism  imports just outputs.
The extended import mechanism allows to import filters, variants and globals.
Currently no mechanism imports preflights

nerdyscout, 2022-04-10:
hi @set-soft, thank you for clarification!
I was hoping the extended import mechanism would include preflights as well, that would have allowed me to have a common kibot config file set used for all kind of projects together with a lean project specific config. but I will try to set up my project differently.

closed 2022-04-10 by nerdyscout

set-soft, 2022-07-12:
Now should be possible.