INTI-CMNB/KiBot #182: dev_k6: ERROR: Missing library signature

SukuWc, 2022-04-13:
Hi, 

I just migrated a project from v5 to v6 (KiBot github action was working perfectly on the v5 project)

I updated the workflow file to use container: "setsoft/kicad_auto:dev_k6"

Unfortunately, now I get the following error:

`ERROR:At line 1 of '/usr/share/kicad/symbols/power.kicad_sym': Missing library signature (kibot - kiplot.py:255)
ERROR:Line content: '(kicad_symbol_lib (version 20211014) (generator kicad_symbol_editor)' (kibot - kiplot.py:256)`

What could be the cause? Is there something else I need to setup in order to use v6? 

Thank you

set-soft, 2022-04-13:
Are you sure you removed all KiCad 5 files from your project?
Run KiBot using `-vvv` and post the full output

SukuWc, 2022-04-14:
Based on your comment I edited the Makefile to point to the correct v6 schematic (SCH=xxx.kicad_sch).  I also removed the all of the old project file (.pro) and the old schematic files (.sch) and kibot now fails at the DRC stage. If I disable the DRC in the preflight settings then the everything is built correctly. 

I check and I only have warnings in the mfg-bot/xxx-drc.txt most of them are silkscreen related. I also checked in KiCAD and I got the same exact warnings in the PCB editor's DRC checker (0 Errors).  

How can I setup kibot to ingore DRC warnings and only throw and error on actual DRC errors?

set-soft, 2022-04-14:
Without logs I can't say much, what you say are warnings are most probably reported as *DRC violations* so they are DRC errors.
You can [filter DRC errors](https://github.com/INTI-CMNB/KiBot#filtering-drc-and-erc-errors)

closed 2022-04-14 by set-soft

SukuWc, 2022-04-19:
In kicad I only have warnings
[image]

kibot generates a drc report file with only warnings in it (Severity: warning) still it fails with the following message:

```
make[1]: *** [kibot.mak:64: mfg-bot/PBF4-drc.txt] Error 5
make[1]: Leaving directory '/__w/grid-hardware/grid-hardware/Electrical/Design/PBF4/PBF4'
make: *** [Makefile:33: drc_and_fab] Error 2
make: Leaving directory '/__w/grid-hardware/grid-hardware/Electrical/Design/PBF4/PBF4'
Error: Process completed with exit code 2.
```
This is part of the the generated PBF4-drc.txt. I checked the whole file, there are only warnings in it. (courtyards_overlap, silk_over_copper & silk_overlap)
```
** Drc report for /__w/grid-hardware/grid-hardware/Electrical/Design/PBF4/PBF4/PBF4.kicad_pcb **
** Created on 2022-04-19 09:10:01 **
** Found 129 DRC violations **
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(98.0000 mm, 104.0000 mm): Footprint C40
    @(98.0000 mm, 102.6000 mm): Footprint TP13
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(103.0000 mm, 63.0000 mm): Footprint C50
    @(102.5000 mm, 60.5000 mm): Footprint TP19
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(98.0000 mm, 98.0000 mm): Footprint C11
    @(98.0000 mm, 96.6000 mm): Footprint TP18
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(102.0000 mm, 104.0000 mm): Footprint C12
    @(102.0000 mm, 102.6000 mm): Footprint TP17
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(97.2500 mm, 54.6500 mm): Footprint C37
    @(97.7500 mm, 52.0000 mm): Footprint TP9
[courtyards_overlap]: Courtyards overlap
    Local override; Severity: warning
    @(100.0000 mm, 62.5000 mm): Footprint R2
    @(97.5000 mm, 62.5000 mm): Footprint TP14

.....
.....
.....

```
I tried to add the following drc filter to the preflight section, but did not make any difference:
```
preflight:
  run_erc: true
  run_drc: true
  check_zone_fills: false
  ignore_unconnected: true  
  filters:
    - filter: 'Filter to ignore courtyards_overlap'
      error: 'courtyards_overlap'
      regex: 'Footprint'
    - filter: 'Filter to ignore silk_over_copper'
      error: 'silk_over_copper'
      regex: 'Silkscreen'
    - filter: 'Filter to ignore silk_overlap'
      error: 'silk_overlap'
      regex: 'Silkscreen'
```
How can I setup kibot to ignore all of the warnings?

set-soft, 2022-04-19:
I tried using `dev_k6` in my machine using this example: 
[test.zip](https://github.com/INTI-CMNB/KiBot/files/8512402/test.zip)

```
salvador@a2ff4bdd49ca:~/1$ kibot -b courtyard_overlaps.kicad_pcb -c drc_filter_crtyd.kibot.yaml ; echo $?
- Running the DRC
WARNING:(W058) Ignoring (courtyards_overlap) Courtyards overlap
    Local override; Severity: warning
    @(108.0000 mm, 80.0000 mm): Footprint C2
    @(114.8900 mm, 79.9100 mm): Footprint C1 (pcbnew_do.kiauto.file_util - file_util.py:104) (kibot - kiplot.py:152)
Found 1 unique warning/s (1 total)
0
salvador@a2ff4bdd49ca:~/1$ kibot -b courtyard_overlaps.kicad_pcb -c drc.kibot.yaml ; echo $?
- Running the DRC
ERROR:Found 1 DRC errors and 0 unconnected pads (pcbnew_do - pcbnew_do:820) (kibot - kiplot.py:149)
WARNING:(W058) (courtyards_overlap) Courtyards overlap
    Local override
    @(108.0000 mm, 80.0000 mm): Footprint C2
    @(114.8900 mm, 79.9100 mm): Footprint C1 (pcbnew_do.kiauto.file_util - file_util.py:130) (kibot - kiplot.py:152)
ERROR:DRC errors: 1 (kibot.pre_drc - pre_drc.py:61)
5
```

I forced `courtyards_overlap` to be a warning in the project and KiBot is reporting it as a warning.
And I can filter it with the filter line you shown.

You posted:

```
make[1]: *** [kibot.mak:64: mfg-bot/PBF4-drc.txt] Error 5
make[1]: Leaving directory '/__w/grid-hardware/grid-hardware/Electrical/Design/PBF4/PBF4'
make: *** [Makefile:33: drc_and_fab] Error 2
make: Leaving directory '/__w/grid-hardware/grid-hardware/Electrical/Design/PBF4/PBF4'
Error: Process completed with exit code 2.
```

But this says nothing but "you got an error", you must post the whole log.

Downloading https://github.com/intechstudio/grid-hardware is really annoying, advice: put the heavy files in a separated sub-project, you have more than 1.2 GiB in Mechanical, and also a lot of back-up files.

Looking at your last workflow run it says:

```
WARNING:(W058) (holes_co_located) Drilled holes co-located
    Local override
    @(99.2500 mm, 54.7500 mm): Via [/MCU/USB_DP] on F.Cu - B.Cu
    @(99.2500 mm, 54.7500 mm): Via [/MCU/USB_DP] on F.Cu - B.Cu (pcbnew_do.kiauto.file_util - file_util.py:130) (kibot - kiplot.py:152)
ERROR:DRC errors: 1 (kibot.pre_drc - pre_drc.py:61)
```

This isn't filtered. Take a look at: https://github.com/intechstudio/grid-hardware/suites/6170671535/artifacts/216407983