INTI-CMNB/KiBot #183: [question] How to use more recent commit in GitHub Actions with KiCAD 6

richardbarlow, 2022-04-14:
I see from a few commits on 25th March that you tried to unify the KiCAD 5/6 Docker image stuff. Since then the default Docker image build has KiCAD v5, not v6 in it. Currently I use the v2_k6 tag in CI and would like to switch to using the hash of the most recent commit on master to gain access to the recent improvements. However, that does not work in GitHub Actions as the container has v5, not v6 in it. How would you recommend I go about using a more up to date commit of KiBot with v6 in GitHub Actions?

set-soft, 2022-04-14:
> I see from a few commits on 25th March that you tried to unify the KiCAD 5/6 Docker image stuff. 

Yes, but the mechanism used by GiHub to create the docker images for the actions is too limited.

> Since then the default Docker image build has KiCAD v5, not v6 in it. 

This doesn't really matter, you use a tag and the tag creates an image using another image as base, not the KiBot code in the tag.

> Currently I use the v2_k6 tag in CI and would like to switch to using the hash of the most recent commit on master to gain access to the recent improvements.

`v2_k6` docker image starts with: `FROM setsoft/kicad_auto:dev_k6` and `dev_k6` is exactly what you want: the last commit that passed all regression tests.

The tag just selects how the action is built, but currently GitHub actions use the last code available. It will change in the future,I'll most probably introduce `v2_dev_*`

closed 2022-04-14 by set-soft

richardbarlow, 2022-04-14:
Ah, OK. Thank you. I hadn't realised that the KiBot `v2_k6` tag actually resulted in the GitHub action using the latest KiBot code. That's helpful right now, but not fantastic for reproducibility of builds.